esphome/issues #3565: Lolin S2 Mini ADC reding capped at 2.59v

AndreasBVB, 2022-09-08:
### The problem

With any attenuation, the ADC reading is 2.59v max even though it's actually beyond. Not scaled to 2.59 but capped.

### Which version of ESPHome has the issue?

2022.8.1

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.3.5

### What platform are you using?

ESP32-IDF

### Board

Lolin S2 Mini

### Component causing the issue

adc

drapergeek, 2023-01-09:
I'm seeing a similar issue on my device. Regardless of the device, I can't get the reading to work properly on the s2 mini, but it works fine on other devices.

jazzmonger, 2023-03-07:
My S2 is capped at 0.70v on ALL ADC inputs. So incredibly strange.