esphome/issues #3566: how should the bluetoot_proxy work?

evlo, 2022-09-08:
### The problem

I don't understand how esphome bluetooth proxy_should work.

i have added
```
esp32_ble_tracker:
  scan_parameters:
    active: true
bluetooth_proxy:
```

to one of my existing esphome devices (climate)

I have tried removing it and adding it into home assistant, but no new entries in esphome integration, bluetooth integration says there are no unconfigured adapters. 

Will the bluetooth proxy discovered devices with with homekit controller if they are actually homekit compatible?

### Which version of ESPHome has the issue?

2022.8.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.9

### What platform are you using?

ESP32

### Board

nodemcu

### Component causing the issue

bluettoth_proxy

### Example YAML snippet

```yaml
esp32_ble_tracker:
  scan_parameters:
    active: true
bluetooth_proxy:
```

### Additional information

no logs, nothing from bluetooth proxy in debug log, also compile is not complaining about bluetooth_proxy being invalid line etc.

austwhite, 2022-09-10:
The bluetooth proxy will only discover Passive BLE devices currently.   
Whar are you trying to discover?    If they are locks or similar devices that need secure or active bluetooth connections, then currently bluetooth proxy will not detet these.   If it is devices like Mi FLora, you the bluetooth proxy will detect these, with the exception of battery that needs an active or paired bluetooth connection.

To asnwer your question, you need to specifify what devices you want to connect to.

syphernl, 2022-09-10:
I'm seeing the same, nothing appears to be happening. I have many MiFloras which I expected to see in the logs at least but nothing is showing up.

Same for other devices such as my Switchbot Curtain, Mi Band or even my Nest smoke detectors. Opening a Bluetooth (LE) scanner on my phone shows quite a list, yet the bluetooth proxy seems to not detect any..

austwhite, 2022-09-10:
You wont see it other than an ESPHome device.     The logs do not show the connections.  It just acts as it says, a bluetooth proxy.
The devices connect to it like it was just a bluetooth adapter, with the exception of only being passive.
If your devices are working, then it is working.   
I believe you could turn verbose logging on if you want to see it all in the logs, but by default there is no logging of the ble broadcasts

syphernl, 2022-09-12:
With `verbose` logging I indeed see a lot of traffic being proxied, a small excerpt:

```
[09:07:57][V][bluetooth_proxy:018]: Proxying packet from Flower care - XX:XX:XX:XX:XX:23. RSSI: -71 dB
[09:07:57][V][bluetooth_proxy:018]: Proxying packet from  - XX:XX:XX:XX:XX:2E. RSSI: -91 dB
[09:07:57][V][bluetooth_proxy:018]: Proxying packet from  - XX:XX:XX:XX:XX:2E. RSSI: -90 dB
[09:07:57][V][bluetooth_proxy:018]: Proxying packet from  - XX:XX:XX:XX:XX:74. RSSI: -61 dB
[09:07:57][V][bluetooth_proxy:018]: Proxying packet from  - XX:XX:XX:XX:XX:69. RSSI: -75 dB
```

Nothing ends up in HASS though. 
E.g. if I enable the "Xiaomi BLE" integration I would expect it to detect all of the Flower Care (MiFlora) devices yet it shows "No devices found on the network".

zigomatichub, 2022-09-17:
Hello, I'm testing also the BT proxy in esphome and I'm not able to have good result.
Example: A LED BLE  is discovered by the BT internal component into raspberry4 without the esphome proxy. It works good.
I put a esp32 with ble32 and proxy enabled (with parameters given into the github generic esp32) into another room and put also the LED BLE near that place. I can't have the control of the Led as it still kept unavailable. 
If I remove bluetooth_proxy, I can see into the logs, LED BLE information coming. As soon as I put the proxy line into the yaml, I only see "Starting scan..."   So till now, I'm not able to have the proxy running.

austwhite, 2022-09-17:
@zigomatichub 
Please read the documentation for ESPHome Bluetooth Proxy.   It very clearly states it is PASSIVE only.
Your LED Lights require an ACTIVE connection to secure the connection via Bluetooth.      

What you are seeing is expected behaviour.    Things like Sensors that just send the data out for anything to see will work with ESPHome BT Proxy as that is passive.   Anything that needs a secured or paired connection, like Lights, Locks, Switches, will NOT work with a passive BT Proxy, which ESPHome is at this time

zigomatichub, 2022-09-17:
@austwhite  Thank you to explain me this.

I read the documentation and it says: "Home Assistant can expand it’s Bluetooth reach by communicating through the Bluetooth proxy component in ESPHome".  Ok now I see that short notice in "Note section" that is passive only.  I was thinking that if it's linked and controled via BT in raspi4 then I was expecting that it will be also the same on the esphome BT proxy. I have to do some research to understand what is a passive connection.

zigomatichub, 2022-09-17:
I moved the esp32 in a room where there is a BT flower sensor and I get directly the information ! @austwhite Thank you for your fast answer. The proxy bt is working then perfectly as it should. I hope to see an extension of the active mode too  haha :)