esphome/issues #3569: KeyError: 'target_platform' with board 'wemos_d1_uno32'

fernandolcardoso, 2022-09-09:
### The problem

[image]

### Which version of ESPHome has the issue?

v2022.8.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.9.1

### What platform are you using?

ESP32

### Board

WEMOS D1 R32

### Component causing the issue

board

### Example YAML snippet

```yaml
esp32:
  board: wemos_d1_uno32
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/yaml.yaml...
ERROR Unexpected exception while reading configuration:
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
  File "/esphome/esphome/__main__.py", line 931, in main
    return run_esphome(sys.argv)
  File "/esphome/esphome/__main__.py", line 909, in run_esphome
    config = read_config(dict(args.substitution) if args.substitution else {})
  File "/esphome/esphome/config.py", line 973, in read_config
    res = load_config(command_line_substitutions)
  File "/esphome/esphome/config.py", line 828, in load_config
    return _load_config(command_line_substitutions)
  File "/esphome/esphome/config.py", line 816, in _load_config
    result = validate_config(config, command_line_substitutions)
  File "/esphome/esphome/config.py", line 743, in validate_config
    result.run_validation_steps()
  File "/esphome/esphome/config.py", line 124, in run_validation_steps
    task.step.run(self)
  File "/esphome/esphome/config.py", line 489, in run
    validated = schema(input_conf)
  File "/esphome/esphome/voluptuous_schema.py", line 34, in __call__
    res = super().__call__(data)
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/schema_builder.py", line 818, in validate_callable
    return schema(data)
  File "/esphome/esphome/voluptuous_schema.py", line 34, in __call__
    res = super().__call__(data)
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/schema_builder.py", line 595, in validate_dict
    return base_validate(path, iteritems(data), out)
  File "/esphome/esphome/voluptuous_schema.py", line 148, in validate_mapping
    cval = cvalue(key_path, value)
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/schema_builder.py", line 818, in validate_callable
    return schema(data)
  File "/esphome/esphome/pins.py", line 61, in validator
    return validator({CONF_NUMBER: value})
  File "/esphome/esphome/pins.py", line 67, in validator
    return PIN_SCHEMA_REGISTRY[CORE.target_platform][1](value)
  File "/esphome/esphome/core/__init__.py", line 587, in target_platform
    return self.data[KEY_CORE][KEY_TARGET_PLATFORM]
KeyError: 'target_platform'
```

### Additional information

Regardless of the settings, the board option does not accept 'wemos_d1_uno32'

dclobato, 2022-09-10:
Same here, but using a Sonoff Pow R2

```yaml
esphome:
  name: ${nomeDoNodo}

esp8266:
  board: esp8285
  esp8266_restore_from_flash : True

preferences:
  flash_write_interval: 1h
  
logger:
  baud_rate: 0

uart:
  rx_pin: RX
  baud_rate: 4800

# Enable Home Assistant API
api:
  password: !secret API_PASSWORD

ota:
  password: !secret OTA_PASSWORD

wifi:
  ssid: ${WIFI_SSID}
  password: !secret wifi_password
  fast_connect: true
  output_power: 20

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${nomeDoNodo} FBH"
    password: "BbpKayY8SAy5"

  manual_ip:
    static_ip: ${IP}
    gateway: ${GATEWAY}
    subnet: 255.255.255.0
    dns1: ${GATEWAY}

captive_portal:

time:
  - platform: sntp
    timezone: America/Sao_Paulo
    servers:
      - 10.0.1.17
      
sensor:
  - platform: total_daily_energy
    name: "Energia total consumida ${emPrep} ${Nome}"
    power_id: potencia
    filters:
        - multiply: 0.001
    unit_of_measurement: kWh
    device_class: "energy"

  - platform: cse7766
    update_interval: 2s
    current:
      name: "Corrente ${emPrep} ${Nome}"
      filters:
        - sliding_window_moving_average:
            window_size: 15
            send_every: 10
    voltage:
      name: "Tensão ${emPrep} ${Nome}"
      filters:
        - sliding_window_moving_average:
            window_size: 15
            send_every: 10
    power:
      name: "Potência ${emPrep} ${Nome}"
      id: potencia
      filters:
        - sliding_window_moving_average:
            window_size: 15
            send_every: 10
    energy:
      name: "Energia ${emPrep} ${Nome}"
      filters:
        - sliding_window_moving_average:
            window_size: 15
            send_every: 10
```

The error is the following

```bash
PS D:\ESPHome\HomeAssistant> esphome.exe compile tomada3d.yaml
INFO Reading configuration tomada3d.yaml...
ERROR Unexpected exception while reading configuration:
Traceback (most recent call last):
  File "C:\Program Files\Python310\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Program Files\Python310\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "C:\Program Files\Python310\Scripts\esphome.exe\__main__.py", line 7, in <module>
  File "C:\Users\danie\AppData\Roaming\Python\Python310\site-packages\esphome\__main__.py", line 931, in main
    return run_esphome(sys.argv)
  File "C:\Users\danie\AppData\Roaming\Python\Python310\site-packages\esphome\__main__.py", line 909, in run_esphome
    config = read_config(dict(args.substitution) if args.substitution else {})
  File "C:\Users\danie\AppData\Roaming\Python\Python310\site-packages\esphome\config.py", line 973, in read_config
    res = load_config(command_line_substitutions)
  File "C:\Users\danie\AppData\Roaming\Python\Python310\site-packages\esphome\config.py", line 828, in load_config
    return _load_config(command_line_substitutions)
  File "C:\Users\danie\AppData\Roaming\Python\Python310\site-packages\esphome\config.py", line 816, in _load_config
    result = validate_config(config, command_line_substitutions)
  File "C:\Users\danie\AppData\Roaming\Python\Python310\site-packages\esphome\config.py", line 743, in validate_config
    result.run_validation_steps()
  File "C:\Users\danie\AppData\Roaming\Python\Python310\site-packages\esphome\config.py", line 124, in run_validation_steps
    task.step.run(self)
  File "C:\Users\danie\AppData\Roaming\Python\Python310\site-packages\esphome\config.py", line 498, in run
    validated = schema(self.conf)
  File "C:\Users\danie\AppData\Roaming\Python\Python310\site-packages\esphome\voluptuous_schema.py", line 34, in __call__
    res = super().__call__(data)
  File "C:\Users\danie\AppData\Roaming\Python\Python310\site-packages\voluptuous\schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "C:\Users\danie\AppData\Roaming\Python\Python310\site-packages\voluptuous\validators.py", line 229, in _run
    return self._exec(self._compiled, value, path)
  File "C:\Users\danie\AppData\Roaming\Python\Python310\site-packages\voluptuous\validators.py", line 353, in _exec
    v = func(path, v)
  File "C:\Users\danie\AppData\Roaming\Python\Python310\site-packages\voluptuous\schema_builder.py", line 818, in validate_callable
    return schema(data)
  File "C:\Users\danie\AppData\Roaming\Python\Python310\site-packages\esphome\voluptuous_schema.py", line 34, in __call__
    res = super().__call__(data)
  File "C:\Users\danie\AppData\Roaming\Python\Python310\site-packages\voluptuous\schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "C:\Users\danie\AppData\Roaming\Python\Python310\site-packages\voluptuous\schema_builder.py", line 595, in validate_dict
    return base_validate(path, iteritems(data), out)
  File "C:\Users\danie\AppData\Roaming\Python\Python310\site-packages\esphome\voluptuous_schema.py", line 118, in validate_mapping
    not isinstance(key.default, vol.Undefined)
  File "C:\Users\danie\AppData\Roaming\Python\Python310\site-packages\esphome\config_validation.py", line 1455, in default
    if CORE.is_esp8266:
  File "C:\Users\danie\AppData\Roaming\Python\Python310\site-packages\esphome\core\__init__.py", line 591, in is_esp8266
    return self.target_platform == "esp8266"
  File "C:\Users\danie\AppData\Roaming\Python\Python310\site-packages\esphome\core\__init__.py", line 587, in target_platform
    return self.data[KEY_CORE][KEY_TARGET_PLATFORM]
KeyError: 'target_platform'
```

fernandolcardoso, 2023-01-09:
Any solution?