esphome/issues #3571: interlock doesn't actually interlock

ccrome, 2022-09-11:
### The problem

Interlock does not reliably interlock. 

With the following 3-speed fan snippet, there are 3 speeds and 3 separate relays.  Each relay must not be on a the same time.

When I use an interlock, the gpios are most definitely not interlocked -- i.e. it's easy to get multiple pins on at once, and for them to stay that way. 

How to reproduce:
  * load up an esp32 module with the yaml settings below
  * go to the user interface, which should show: 
  [image]
  * Nicely click relay 1, then relay 2, then relay 3.  The interlock give the illusion of working correctly. 
  * However, if you click the "Switch" switch to flip all three at once, you get rather unexpected behavior and relay 1 comes on for 20ms, then without any interlock wait time, relay 3 turns on.  Then relay 2 turns on after the interlock time, leaving both relay 3 and relay 2 on continuously
  
You can see the behavior with this logic analyzer capture, and also in the graphical UI.

[image]

[image]

### Which version of ESPHome has the issue?

2022.8.3

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2022.9.2

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

switch

### Example YAML snippet

```yaml
fan:
  - platform: speed
    id: speed_fan
    output: fan_output_value
    name: "2 Speed Fan"
    speed_count: 3
    on_turn_on:
    - logger.log: "Fan Turned On"
    on_speed_set:
    - logger.log: "Fan Speed was changed"
    on_turn_off:
    - logger.log: "Fan Turned Off"

#relay outputs for 3-speed fan motor 
switch:
  - platform: gpio
    id: relay1
    pin: 16
    restore_mode: ALWAYS_OFF
    #only one should be on
    interlock: &interlock_group [relay1, relay2]
    interlock_wait_time: 1s
    name: "relay1"
  - platform: gpio
    id: relay2
    pin: 17
    restore_mode: ALWAYS_OFF
    interlock: *interlock_group
    interlock_wait_time: 1s
    name: "relay2"
  - platform: gpio
    id: relay3
    pin: 18
    restore_mode: ALWAYS_OFF
    interlock: *interlock_group
    interlock_wait_time: 1s
    name: "relay3"
    
output:
# 3-speed fan output_value    
  - platform: template
    id: fan_output_value
    type: float
    write_action:
      - if:
          condition:
            lambda: return ((state == 0));
          then:
            - switch.turn_off: relay1
            - switch.turn_off: relay2
            - switch.turn_off: relay3
      - if:
          condition:
            lambda: return ((state > 0) && (state <= .34));
          then:
            - switch.turn_on: relay1
            - switch.turn_off: relay2
            - switch.turn_off: relay3
      - if:
          condition:
            lambda: return ((state > .35) && (state <= .67));
          then:
            - switch.turn_on: relay2
            - switch.turn_off: relay1
            - switch.turn_off: relay3
      - if:
          condition:
            lambda: return ((state > .67));
          then:
            - switch.turn_on: relay3
            - switch.turn_off: relay1
            - switch.turn_off: relay2
```

ssieb, 2022-09-12:
`relay3` isn't in the interlock group.

ccrome, 2022-09-12:
> 

  :facepalm: I swear I had it in before, and still witnessed the problem.  But now I can't reproduce.

Thanks for the keen eye.  

Closing... sorry for the spurious noise.

closed 2022-09-12 by ccrome