esphome/issues #3572: interlock wait time does not apply properly when quickly switching

ccrome, 2022-09-11:
### The problem

The switch interlock should use the interlock_wait_time as a means of preventing *any* changes to GPIOs in an interlock group within a certain amount of time since the last change.

That is, if there are multiple requests to the the interlock group within a short amount of time, they should either be queued, or the last one should take precedence, but in no case should anything change within the interlock time.

This is slightly different from 'interlock doesn't work' issues I just filed.  (https://github.com/esphome/issues/issues/3571)

Since the reason we want a wait time for interlocking is that relays shouldn't be switched on rapidly, this should be prevented at all costs.  

An example YAML and capture is attached, using the fan settings to set the relays.  

The switch.turn_on() and switch.turn_off() seems to take effect immediately, rather than respecting the interlock, whicih causes a rapid switching on the relays, which would be horrible on a 3 speed motor.  

By using the fan slider (and using the keyboard to quickly switch between states)
[image]

You can get rapid relay switches, even when the interlock_wait_time is set to 1 second.
[image]

I don't believe that should be able to happen.

### Which version of ESPHome has the issue?

2022.8.3

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2022.9.2

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

switch

### Example YAML snippet

```yaml
fan:
  - platform: speed
    id: speed_fan
    output: fan_output_value
    name: "2 Speed Fan"
    speed_count: 3
    on_turn_on:
    - logger.log: "Fan Turned On"
    on_speed_set:
    - logger.log: "Fan Speed was changed"
    on_turn_off:
    - logger.log: "Fan Turned Off"

#relay outputs for 3-speed fan motor 
switch:
  - platform: gpio
    id: relay1
    pin: 16
    restore_mode: ALWAYS_OFF
    #only one should be on
    interlock: &interlock_group [relay1, relay2]
    interlock_wait_time: 1s
    name: "relay1"
  - platform: gpio
    id: relay2
    pin: 17
    restore_mode: ALWAYS_OFF
    interlock: *interlock_group
    interlock_wait_time: 1s
    name: "relay2"
  - platform: gpio
    id: relay3
    pin: 18
    restore_mode: ALWAYS_OFF
    interlock: *interlock_group
    interlock_wait_time: 1s
    name: "relay3"
    
output:
# 3-speed fan output_value    
  - platform: template
    id: fan_output_value
    type: float
    write_action:
      - if:
          condition:
            lambda: return ((state == 0));
          then:
            - switch.turn_off: relay1
            - switch.turn_off: relay2
            - switch.turn_off: relay3
      - if:
          condition:
            lambda: return ((state > 0) && (state <= .34));
          then:
            - switch.turn_on: relay1
            - switch.turn_off: relay2
            - switch.turn_off: relay3
      - if:
          condition:
            lambda: return ((state > .35) && (state <= .67));
          then:
            - switch.turn_on: relay2
            - switch.turn_off: relay1
            - switch.turn_off: relay3
      - if:
          condition:
            lambda: return ((state > .67));
          then:
            - switch.turn_on: relay3
            - switch.turn_off: relay1
            - switch.turn_off: relay2
```

ssieb, 2022-09-12:
You're also missing `relay3` from the interlock group here.  I would expect that quickly switching the relays would be much less hard on the motor than shutting it off and then changing speeds.  The intention of interlock is to avoid having multiple relays active at the same time to avoid the chance of slow relays causing a short circuit.

ccrome, 2022-09-12:
> You're also missing `relay3` from the interlock group here.
Doh!  Yes, but it actually has no effect on the issue I'm reporting.
Here's an example of just relay 1 and relay 2 -- yes they are interlocking.  However, they are not respecting interlock_wait_time
I'm able to switch them many times in a second.  I actually put relay 3 into the group for this latest test.

[image]

> I would expect that quickly switching the relays would be much less hard on the motor than shutting it off and then changing speeds. 

I'm not at all sure of that.  I have a whole house attic fan that has 3 speeds.  This is a big motor relative to little desktop oscillating fans.  When the stock system controller switches speeds, it turns off for about 5 seconds, then switches on to the next speed.  I'm actually an electrical engineer, but I never took a rotating electrical machinery class -- so I really don't know the ins and outs and details of mutliple-speed AC motors.   I just want to make sure I can replicate the behavior of the stock system before I go and start breaking things.  To that end -- I want to protect the system from within the ESP32 and ensure that no network message spamming can violate the law of:  off for 5 seconds -- then next speed on.

> The intention of interlock is to avoid having multiple relays active at the same time to avoid the chance of slow relays causing a short circuit.

Yeah, I'm worried about more than just a short circuit.  The momentum of the fan will cause lots of back EMF, and I really don't know how that will interact when one speed is swapped for the next.  Might be nothing.  Might be something.  I assume it's something, because the stock controller essentially waits long enough for the fan to spin down completely.

I just wrote up custom exclusive output that will be guaranteed to ignore any spurious inputs from the server, and always wait the right amount of time before switching outputs. 

I'll upload that after a little cleanup, if anybody wants it.

ssieb, 2022-09-12:
> Yeah, I'm worried about more than just a short circuit. The momentum of the fan will cause lots of back EMF, and I really don't know how that will interact when one speed is swapped for the next. [...]

Letting it spin down would be creating a lot more whatever than switching to another speed would do.  But 5s is enough to spin down the fan completely?  Is it using some form of active braking?

(I think you're confusing back EMF with something else.)

But about this issue, how are you generating that output that you're measuring.  Also, since you're using the interlock, why are you manually turning off the other switches as well?

ccrome, 2022-09-12:
> Letting it spin down would be creating a lot more whatever than switching to another speed would do. But 5s is enough to spin down the fan completely? Is it using some form of active braking?

Honestly, I've been burned too many times -- big inductive loads are no joke.  I really just need to guarantee the behavior is similar to the stock controller for my own piece of mind.  It's a new, expensive fan :-)

> how are you generating that output that you're measuring

Saleae logic analyzer, looking at the GPIOs that would control the relays.

> Also, since you're using the interlock, why are you manually turning off the other switches as well?

Because I want to make 100% sure, that no server spamming can cause the relays to misbehave.

If you're interested, here's my interlocked with time delay/guaranteed solution: https://github.com/ccrome/home-assistant-relay-fan-output

ssieb, 2022-09-13:
> > how are you generating that output that you're measuring
> 
> Saleae logic analyzer, looking at the GPIOs that would control the relays.

I mean what are you doing that is creating that output on the GPIOs?

ccrome, 2022-09-13:
Ohhhh.  I click with the mouse on the fan slider to give the slider focus.  Then use my keyboard left and right keys to switch quickly back and forth between speeds.

kadamski, 2023-01-11:
@ccrome I'm not entirely sure I get your problem (not entierly sure how to read your graphs) but looking at the esphome code, the interlock_wait_time is **only applied** when you **turn on** your switch **and** there is at least one switch other than the one you are changing right now **that is already on** in the group.
This means, that if you would first turn off all the switches then immediately after that turn on one of them, the interlock_wait_time will not be respected.

Is this maybe what you see in your case?

If I look at the output from your logic analyzer, it seems to me that after 10th second, when red one is going on, both others are off, which would explain why it can be enabled immediately. I am puzzled, however, by the first transition at around 8th second, here,  the orange seems to be still on while red goes on. This should not happen.

If you look at the code, what is does that is you turn on the switch, it iterates over the whole group, and for every switch other than the one that is being changed right now, it checks if it is on. If it is, it turns it off. Next, if it did turn off at least one switch, it and the interlock_wait_time is not 0, it starts a timer that switches on this switch after interlock_wait_time, and exists.

Note that the logs might be misleading as the code **does not** log "Turning ON/OFF" to the console at the moment the state of the GPIO is actually changed but when the function is called.

I also think that if you would call switch_on() twice in a row quickly, then here is what would happen:
1. At the first call, it would find that there is some other relay on, would turn it off and schedule writing state 1 in 1s from now.
2. At the 2nd call, it would see that all other relays are actually off (because they were turned off at the first call) so it would turn on **immediately**, regardless of the fact that this timer was scheduled by the 1st call and the 1s delay did not happen. I don't see the GPIOSwitch::write_state() checking if the timer is pending anywhere.

ccrome, 2023-01-16:
> @ccrome I'm not entirely sure I get your problem (not entierly sure how to read your graphs) but looking at the esphome code, the interlock_wait_time is **only applied** when you **turn on** your [...]

Yes that's it.  In my opinion, the interlock wait time should be a minimum-off-time.  This allows all magnetcs and spinning parts in motors to spin down before jolting on a new voltage source/coil.  It may not be necessary all the time, but there should be a minimum off and and a minimum on time for safest operation.

> If I look at the output from your logic analyzer, it seems to me that after 10th second, when red one is going on, both others are off, which would explain why it can be enabled immediately. I am [...]

 
> If you look at the code, what is does that is you turn on the switch, it iterates over the whole group, and for every switch other than the one that is being changed right now, it checks if it is [...]

I believe this isn't really the correct behavior.  I think there should be a timer that excludes any transitions on any outputs closer than interlock_wait_time.  I understand it works as designed,

> 
> Note that the logs might be misleading as the code **does not** log "Turning ON/OFF" to the console at the moment the state of the GPIO is actually changed but when the function is called.

That's why I was capturing from a GPIO with a logic analyzer just to make sure.

silentnyte, 2023-06-04:
I am having a similar issue and I agree that ideally the interlock function would work in software the same way that this diagram would work electrically. This ensures that R1 cannot turn on until R2 has been off for a set period and whether R2 is turned on by S2 or through HA it will still prevent R1 from turning on.
https://i.imgur.com/iaT5uHl.png