esphome/issues #3575: TCS34725 does overexposure on auto (which leads to negative color temperatures)

RubenKelevra, 2022-09-12:
### The problem

I'm using a TCS34725 with `integration_time: auto` on a D1 mini. 

Around noon the color temperature is either NaN or is reported negative:

[image]

After 14:30 for the rest of the day (with obviously less intense sun hitting the sensor) the values are reported correctly:

[image]

[image]

[image]

[image]

### Which version of ESPHome has the issue?

2022.8.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.9.2

### What platform are you using?

ESP8266

### Board

D1_mini

### Component causing the issue

TCS34725

### Example YAML snippet

```yaml
esphome:
  name: $devicename
  platform: ESP8266
  board: d1_mini

substitutions:
  devicename: ambient-light-sensor
  friendly_name: "Ambient Light Sensor"

logger:
  esp8266_store_log_strings_in_flash: false

mdns:
  disabled: true
  

api:
  encryption: 
    key: !secret api_key-ambient-light-sensor

ota:
  password: !secret ota_password_ambient-light-sensor

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.8.27
    gateway: !secret wifi_gateway
    subnet: !secret wifi_subnet
    dns1: !secret wifi_dns1
    dns2: !secret wifi_dns2
  fast_connect: true
  reboot_timeout: 5min

i2c:
  sda: D1
  scl: D2
  scan: true
  id: bus_a

sensor:
  - platform: tcs34725
    i2c_id: bus_a
    red_channel:
      name: "$friendly_name Red Channel relative"
      accuracy_decimals: 2
      filters:
        - exponential_moving_average:
            alpha: 0.05
            send_every: 2
            send_first_at: 2
        - or:
          - throttle: 10s
          - delta: 0.5
    green_channel:
      name: "$friendly_name Green Channel relative"
      accuracy_decimals: 2
      filters:
        - exponential_moving_average:
            alpha: 0.05
            send_every: 2
            send_first_at: 2
        - or:
          - throttle: 10s
          - delta: 0.5
    blue_channel:
      name: "$friendly_name Blue Channel relative"
      accuracy_decimals: 2
      filters:
        - exponential_moving_average:
            alpha: 0.05
            send_every: 2
            send_first_at: 2
        - or:
          - throttle: 10s
          - delta: 0.5
    clear_channel:
      name: "$friendly_name Clear Channel"
      accuracy_decimals: 5
      filters:
        - exponential_moving_average:
            alpha: 0.05
            send_every: 2
            send_first_at: 2
        - or:
          - throttle: 10s
          - delta: 0.5
    illuminance:
      name: "$friendly_name Illuminance"
      accuracy_decimals: 2
      filters:
        - exponential_moving_average:
            alpha: 0.05
            send_every: 2
            send_first_at: 2
        - or:
          - throttle: 10s
          - delta: 0.2
    color_temperature:
      name: "$friendly_name Color Temperature"
      accuracy_decimals: 2
      filters:
        - exponential_moving_average:
            alpha: 0.05
            send_every: 2
            send_first_at: 2
        - or:
          - throttle: 10s
          - delta: 0.5
    integration_time: auto
    glass_attenuation_factor: 1.0
    address: 0x29
    update_interval: 1s    

output:
  - platform: gpio
    pin: D3
    id: white_led
    inverted: false
  
light:
  - platform: binary
    output: white_led
    name: "$friendly_name White LED"
```

RubenKelevra, 2022-09-13:
Example from today:

[image]

RubenKelevra, 2022-09-16:
@jesserockz @dcousens @ @swifty99 maybe you can have a look at this? Thanks! :)

swifty99, 2022-09-19:
that's not good. 
could you provide more information? the raw values of all 
```
    red_channel:
      name: "TCS34725 Red Channel"
    green_channel:
      name: "TCS34725 Green Channel"
    blue_channel:
      name: "TCS34725 Blue Channel"
    clear_channel:
      name: "TCS34725 Clear Channel"
    illuminance:
      name: "TCS34725 Illuminance"

```
would be helpful.

RubenKelevra, 2022-09-27:
@swifty99 yeah sure - sorry for the delay:

[image]
[image]
[image]
[image]
[image]
[image]

RubenKelevra, 2022-09-27:
@swifty99 It looks like the negative values in illuminance triggers the negative values in color temperature.

swifty99, 2022-09-27:
no worries,

do you have a chance to log the Esphome logs while this happens? It will look something like this:
```
[21:13:10][D][tcs34725:251]: Got Red=62.5%,Green=25.0%,Blue=17.5%,Clear=0.0% Illuminance=0.1lx Color Temperature=2343.5K
[21:13:10][D][tcs34725:292]: Integration time: 614.4ms, ideal: 614.4ms regval_new 0 Gain: 60 Clear channel raw: 40  gain reg: 3

```
Probably the sensor ist oversaturated. The only thing that really helps would be a grey filter. I use the light from a reflected surface. Not linear perfectly but good enough.

cheers
Jan

swifty99, 2022-12-04:
Maybe we can close this issue.

Probable take away: have a look in the logs for the overasaturation message.

RubenKelevra, 2023-02-26:
Hey @swifty99,

so I can confirm issues with the latest ESPHome version as well: While there's an output of the chip reported which includes the Color temperature, HA will receive an "inf K" reporting:

<details>

```
INFO Reading configuration /config/esphome/ambient-light-sensor.yaml...
INFO Starting log output from 192.168.8.27 using esphome API
INFO Successfully connected to 192.168.8.27
[17:23:50][I][app:102]: ESPHome version 2023.2.3 compiled on Feb 20 2023, 08:23:51
[17:23:50][C][wifi:504]: WiFi:
[17:23:50][C][wifi:362]:   Local MAC: [redacted]
[17:23:50][C][wifi:363]:   SSID: [redacted]
[17:23:50][C][wifi:364]:   IP Address: 192.168.8.27
[17:23:50][C][wifi:365]:   BSSID: [redacted]
[17:23:50][C][wifi:367]:   Hostname: 'ambient-light-sensor'
[17:23:50][C][wifi:369]:   Signal strength: -70 dB ▂▄▆█
[17:23:50][C][wifi:373]:   Channel: 1
[17:23:50][C][wifi:374]:   Subnet: 255.255.254.0
[17:23:50][C][wifi:375]:   Gateway: 192.168.8.1
[17:23:50][C][wifi:376]:   DNS1: 192.168.8.2
[17:23:50][C][wifi:377]:   DNS2: 0.0.0.0
[17:23:50][C][logger:293]: Logger:
[17:23:50][C][logger:294]:   Level: DEBUG
[17:23:50][C][logger:295]:   Log Baud Rate: 115200
[17:23:50][C][logger:296]:   Hardware UART: UART0
[17:23:50][C][i2c.arduino:052]: I2C Bus:
[17:23:50][C][i2c.arduino:053]:   SDA Pin: GPIO5
[17:23:50][C][i2c.arduino:054]:   SCL Pin: GPIO4
[17:23:50][C][i2c.arduino:055]:   Frequency: 50000 Hz
[17:23:50][C][i2c.arduino:058]:   Recovery: bus successfully recovered
[17:23:50][I][i2c.arduino:068]: Results from i2c bus scan:
[17:23:50][I][i2c.arduino:074]: Found i2c device at address 0x29
[17:23:50][C][uptime.sensor:031]: Uptime Sensor 'Uptime Sensor'
[17:23:50][C][uptime.sensor:031]:   Device Class: 'duration'
[17:23:50][C][uptime.sensor:031]:   State Class: ''
[17:23:50][C][uptime.sensor:031]:   Unit of Measurement: 's'
[17:23:50][C][uptime.sensor:031]:   Accuracy Decimals: 0
[17:23:50][C][uptime.sensor:031]:   Icon: 'mdi:timer-outline'
[17:23:50][D][tcs34725:248]: Got Red=57.3%,Green=49.6%,Blue=47.8%,Clear=66.9% Illuminance=15220.0lx Color Temperature=3994.0K
[17:23:50][D][tcs34725:291]: Integration time: 154.5ms, ideal: 154.0ms regval_new 191 Gain: 1 Clear channel raw: 44095  gain reg: 0
[17:23:50][C][gpio.output:010]: GPIO Binary Output:
[17:23:50][C][gpio.output:011]:   Pin: GPIO0
[17:23:50][C][light:104]: Light 'Ambient Light Sensor White LED'
[17:23:50][C][adc:087]: ADC Sensor 'VCC Voltage'
[17:23:50][C][adc:087]:   Device Class: 'voltage'
[17:23:50][C][adc:087]:   State Class: 'measurement'
[17:23:50][C][adc:087]:   Unit of Measurement: 'V'
[17:23:50][C][adc:087]:   Accuracy Decimals: 2
[17:23:50][C][adc:090]:   Pin: VCC
[17:23:50][C][adc:126]:   Update Interval: 60.0s
[17:23:50][C][tcs34725:047]: TCS34725:
[17:23:50][C][tcs34725:048]:   Address: 0x29
[17:23:50][C][tcs34725:052]:   Update Interval: 1.0s
[17:23:50][C][tcs34725:054]:   Clear Channel 'Ambient Light Sensor Clear Channel'
[17:23:50][C][tcs34725:054]:     State Class: 'measurement'
[17:23:50][C][tcs34725:054]:     Unit of Measurement: '%'
[17:23:50][C][tcs34725:054]:     Accuracy Decimals: 5
[17:23:50][C][tcs34725:054]:     Icon: 'mdi:lightbulb'
[17:23:50][C][tcs34725:055]:   Red Channel 'Ambient Light Sensor Red Channel relative'
[17:23:50][C][tcs34725:055]:     State Class: 'measurement'
[17:23:50][C][tcs34725:055]:     Unit of Measurement: '%'
[17:23:50][C][tcs34725:055]:     Accuracy Decimals: 2
[17:23:50][C][tcs34725:055]:     Icon: 'mdi:lightbulb'
[17:23:50][C][tcs34725:056]:   Green Channel 'Ambient Light Sensor Green Channel relative'
[17:23:50][C][tcs34725:056]:     State Class: 'measurement'
[17:23:50][C][tcs34725:056]:     Unit of Measurement: '%'
[17:23:50][C][tcs34725:056]:     Accuracy Decimals: 2
[17:23:50][C][tcs34725:056]:     Icon: 'mdi:lightbulb'
[... 270 lines cut ...]
[17:25:09][D][sensor:126]: 'Ambient Light Sensor Red Channel relative': Sending state 58.03018 % with 2 decimals of accuracy
[17:25:09][D][sensor:126]: 'Ambient Light Sensor Green Channel relative': Sending state 49.83602 % with 2 decimals of accuracy
[17:25:09][D][sensor:126]: 'Ambient Light Sensor Blue Channel relative': Sending state 48.14580 % with 2 decimals of accuracy
[17:25:09][D][sensor:126]: 'Ambient Light Sensor Illuminance': Sending state 14915.60840 lx with 2 decimals of accuracy
[17:25:09][D][sensor:126]: 'Ambient Light Sensor Illuminance': Sending state 14915.60840 lx with 2 decimals of accuracy
[17:25:09][D][sensor:126]: 'Ambient Light Sensor Color Temperature': Sending state inf K with 2 decimals of accuracy
[17:25:09][D][tcs34725:248]: Got Red=58.3%,Green=49.9%,Blue=48.3%,Clear=66.4% Illuminance=14846.9lx Color Temperature=3930.4K
[17:25:09][D][tcs34725:291]: Integration time: 154.5ms, ideal: 154.0ms regval_new 191 Gain: 1 Clear channel raw: 43778  gain reg: 0
[17:25:10][D][tcs34725:248]: Got Red=58.3%,Green=49.9%,Blue=48.3%,Clear=66.4% Illuminance=14831.3lx Color Temperature=3930.3K
[17:25:10][D][tcs34725:291]: Integration time: 154.5ms, ideal: 154.0ms regval_new 191 Gain: 1 Clear channel raw: 43735  gain reg: 0
[17:25:11][D][sensor:126]: 'Ambient Light Sensor Blue Channel relative': Sending state 48.15688 % with 2 decimals of accuracy
[17:25:11][D][sensor:126]: 'Ambient Light Sensor Illuminance': Sending state 14907.65039 lx with 2 decimals of accuracy
[17:25:11][D][tcs34725:248]: Got Red=58.3%,Green=49.9%,Blue=48.3%,Clear=66.4% Illuminance=14836.5lx Color Temperature=3929.8K
[... cut ...]
[17:25:13][D][tcs34725:291]: Integration time: 154.5ms, ideal: 154.0ms regval_new 191 Gain: 1 Clear channel raw: 43433  gain reg: 0
[17:25:13][D][sensor:126]: 'VCC Voltage': Sending state 3.00098 V with 2 decimals of accuracy
[17:25:14][D][tcs34725:248]: Got Red=58.3%,Green=49.9%,Blue=48.3%,Clear=65.9% Illuminance=14720.0lx Color Temperature=3933.0K
[17:25:14][D][tcs34725:291]: Integration time: 154.5ms, ideal: 154.0ms regval_new 191 Gain: 1 Clear channel raw: 43421  gain reg: 0
```

</details>

swifty99, 2023-02-27:
The internal calculation is OK as:
`[17:25:13][D][tcs34725:248]: Got Red=58.3%,Green=49.9%,Blue=48.3%,Clear=65.9% Illuminance=14722.4lx Color Temperature=3933.1K`

with this code:
```
  ESP_LOGD(TAG,
           "Got Red=%.1f%%,Green=%.1f%%,Blue=%.1f%%,Clear=%.1f%% Illuminance=%.1flx Color "
           "Temperature=%.1fK",
           channel_r, channel_g, channel_b, channel_c, this->illuminance_, this->color_temperature_);
```

this value will be given to sensor properties:
```
   if (this->color_temperature_sensor_ != nullptr)
      this->color_temperature_sensor_->publish_state(this->color_temperature_);
```

At the moment I think I cannot do anything, as the value was calcualted correctly (3933.1K ). And it is not overexposed, none of the raw values reaches 100%.
What you could try is take a different Node and give it different name and try once more. Another try would be to remove the filters temporary.

RubenKelevra, 2023-03-02:
Alright, will do that.

So I think what happens is, that some calculations runs towards infinity and then the value gets stuck there...

That's what it looks like here in this HA-graph:

[image]

RubenKelevra, 2023-03-02:
@swifty99 wrote:
> The internal calculation is OK as:
> `[17:25:13][D][tcs34725:248]: Got Red=58.3%,Green=49.9%,Blue=48.3%,Clear=65.9% Illuminance=14722.4lx Color Temperature=3933.1K`
>
> At the moment I think I cannot do anything, as the value was calcualted correctly (3933.1K ). And it is not overexposed, none of the raw values reaches 100%.
> What you could try is take a different Node and give it different name and try once more. Another try would be to remove the filters temporary.

Well, actually the sensor does get overexposed: clear channel hits (nearly 100%). Guess the "nearly" 100% is not detected and thus the calculations runs off the rails.

[image]

This is then leading to those absurd numbers I guess:

[image]

swifty99, 2023-03-02:
ok. the logging did not indicate that.

Overexposure shall be avoided. Use a filter or a different location. There is nothing the hard or software could fix.

RubenKelevra, 2023-03-19:
@swifty99 so I dug in the code today, and I think I found the issue – and at least one other.

I'll tackle them and send PRs :)

closed 2024-09-30 by jesserockz