esphome/issues #3576: Display component not compiling

declan-wade, 2022-09-13:
### The problem

The issue is that the attached YAML only compiles using the arduino framework, the compiler flags the same warnings but the ESP-IDF version has another line `cc1plus: some warnings being treated as errors` (see log). 
The only difference is that this is now an S2 board. 
Is this a bug or issue with the ESP-IDF framework? 

### Which version of ESPHome has the issue?

2022.8.3

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.9

### What platform are you using?

ESP32

### Board

LOLIN S2 Mini

### Component causing the issue

Display

### Example YAML snippet

```yaml
esphome:
  name: mini-s2

esp32:
  board: esp32-s2-saola-1
  framework:
    type: esp-idf

# Enable logging
logger:
  level: VERY_VERBOSE

# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxx"

ota:
  password: "xxxx"

wifi:
  ssid: "ssid"
  password: "password"

font:
  - file: 'arial.ttf'
    id: font1
    size: 18
  - file: 'arial.ttf'
    id: font2
    size: 14

globals:
  - id: daytime
    type: bool
    initial_value: 'true'

[... 241 lines cut ...]
Compiling .pioenvs/mini-s2/bootloader_support/src/bootloader_flash_config_esp32s2.o
Compiling .pioenvs/mini-s2/bootloader_support/src/bootloader_efuse_esp32s2.o
Compiling .pioenvs/mini-s2/bootloader_support/src/idf/bootloader_sha.o
Compiling .pioenvs/mini-s2/cbor/tinycbor/src/cborencoder_close_container_checked.o
Compiling .pioenvs/mini-s2/cbor/tinycbor/src/cborencoder.o
Compiling .pioenvs/mini-s2/cbor/tinycbor/src/cborerrorstrings.o
Compiling .pioenvs/mini-s2/cbor/tinycbor/src/cborparser_dup_string.o
Compiling .pioenvs/mini-s2/cbor/tinycbor/src/cborparser.o
Archiving .pioenvs/mini-s2/esp-idf/bootloader_support/libbootloader_support.a
Compiling .pioenvs/mini-s2/cbor/tinycbor/src/cborpretty_stdio.o
Compiling .pioenvs/mini-s2/cbor/tinycbor/src/cborpretty.o
Compiling .pioenvs/mini-s2/cbor/tinycbor/src/cbortojson.o
Compiling .pioenvs/mini-s2/cbor/tinycbor/src/cborvalidation.o
Compiling .pioenvs/mini-s2/cbor/tinycbor/src/open_memstream.o
Compiling .pioenvs/mini-s2/cmock/CMock/src/cmock.o
Compiling .pioenvs/mini-s2/coap/libcoap/src/address.o
Archiving .pioenvs/mini-s2/esp-idf/cmock/libcmock.a
Compiling .pioenvs/mini-s2/coap/libcoap/src/async.o
Compiling .pioenvs/mini-s2/coap/libcoap/src/block.o
Compiling .pioenvs/mini-s2/coap/libcoap/src/coap_event.o
cc1plus: some warnings being treated as errors
Compiling .pioenvs/mini-s2/coap/libcoap/src/coap_hashkey.o
*** [.pioenvs/mini-s2/src/main.o] Error 1
========================= [FAILED] Took 18.66 seconds =========================
```

declan-wade, 2022-09-13:
As a workaround, adding a compile flag (technically an _unflag_) makes it work:
'''
esphome:
  name: mini-s2
  platformio_options:
    build_unflags: -Werror=all
'''
But I will leave this issue open as it seems to be a bug with the ESP-IDF framework.

ssieb, 2022-09-13:
The bug is in your code, just like the warning message says.  Why are you doing that?

closed 2022-09-13 by ssieb