esphome/issues #3578: Device unable to connect to api - home assistant reports 2 different devices

workingmanrob, 2022-09-13:
### The problem

I have had this happen a few times recently and it is just bizarre.

For example today my workbench remote stopped working.  It was connected to wifi but not the api.  Checked home assistant frontend and workbench remote reported 2 devices.

[image]

When I click on 2 devices I get a list - one is obviously not my workbench remote.

[image]

This is an esp32 board and an esp32 camera but I am pretty sure I have had this with esp8266 devices as well.

### Which version of ESPHome has the issue?

2022.8.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant 2022.9.2

### What platform are you using?

ESP32

### Board

esp32doit-devkit-v1

### Component causing the issue

api

### Anything in the logs that might be useful for us?

```txt
There was a message about api not connecting but I have removed, updated and added the device back to home assistant.
```

ssieb, 2022-09-15:
You need to remove both of those devices from the integration list to fix this.

workingmanrob, 2022-09-15:
Ya thanks - I did that already - I have done it several times now actually cause I had to do it every time it happened.

The reason I reported it was the hope of having it fixed so it doesn't happen again. :beers:

ssieb, 2022-09-16:
It shouldn't be happening unless you're either changing the name of a device or somehow two devices have the same IP address at different times.

workingmanrob, 2022-09-16:
One is an esp32 camera in my living room the other is an esp32 board with a touchscreen attached to it so they have never switched places.

I use static DHCP mappings for all my IoT goodies so they should never come up with the same IP.

I do agree that it should not be happening.

closed 2023-01-29 by github-actions[bot]