esphome/issues #3580: BLE Scanner - ESP32C3 - undefined reference to `esp_ble_gap_stop_scanning'

diginfo, 2022-09-14:
### The problem

See error below.

Crashed during build / install process.

### Which version of ESPHome has the issue?

2022.8.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.9.3

### What platform are you using?

ESP32-IDF

### Board

QT-PI ESP32C3

### Component causing the issue

esp32_ble_tracker

### Example YAML snippet

```yaml
esphome:
  name: c3test
  platformio_options:
    board_build.flash_mode: dio
    
esp32:
  board: esp32-c3-devkitm-1
  variant: ESP32C3
  framework:
    type: esp-idf

esp32_ble_tracker:
text_sensor:
  - platform: ble_scanner
    name: "BLE Scanner"
```
```

### Anything in the logs that might be useful for us?

```txt
Linking /data/c3test/.pioenvs/c3test/firmware.elf
/data/cache/platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /data/c3test/.pioenvs/c3test/src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.o: in function `esphome::esp32_ble_tracker::ESP32BLETracker::start_scan_(bool)':
/config/esphome/.esphome/build/c3test/src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.cpp:231: undefined reference to `esp_ble_gap_set_scan_params'
/data/cache/platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /config/esphome/.esphome/build/c3test/src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.cpp:234: undefined reference to `esp_ble_gap_start_scanning'
/data/cache/platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /data/c3test/.pioenvs/c3test/src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.o: in function `esphome::esp32_ble_tracker::ESP32BLETracker::loop()':
/config/esphome/.esphome/build/c3test/src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.cpp:98: undefined reference to `esp_ble_gap_stop_scanning'
collect2: error: ld returned 1 exit status
*** [/data/c3test/.pioenvs/c3test/firmware.elf] Error 1
========================= [FAILED] Took 92.54 seconds =========================
```
```

stumpylog, 2022-10-06:
I am also seeing this.  With an updated framework/platformio, it still happens:
```
  framework:
    type: esp-idf
    version: "4.4.2"
    platform_version: "5.2.0"
```

stumpylog, 2022-10-06:
#2648 and #2968

```
esp32:
# At some point the actual board might be supported.
#  board: adafruit_qt_py_esp32-c3
  board: esp32-c3-devkitm-1
  variant: esp32
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_BT_BLE_42_FEATURES_SUPPORTED: y
```

That compiled at least for me.

rhoades-brown, 2023-03-28:
Brilliant, thanks for this.  It allowed me to compile for my Seeed XIAO ESP32C3's:

```yaml
esphome:
  name: esp32-micro1
  friendly_name: esp32-micro1

esp32:
  board: seeed_xiao_esp32c3
  variant: ESP32C3
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_BT_BLE_42_FEATURES_SUPPORTED: y

# Enable logging
logger:
```