esphome/issues #3583: Wifi-NodeM unable to connect to Wi-Fi after upgrade

travnick, 2022-09-16:
### The problem

I have the Wifi-NodeM (a NodeMcu like board - some strange clone with 1MB flash like this one https://allegro.pl/oferta/wifi-nodem-esp8266-16mb-wifi-arduino-wemos-nodemcu-11740898258)

The board is not able to connect to Wi-Fi any more after upgrading from 2022.05 to 2022.8.3 (Home Assistant addon)

There are some random glitches like connected to wifi and external mqtt but not to HA, connected to wifi but not mqtt, not connected to wifi at all.

I have second board, true `NodeMCU v2` and this one works correctly with the same config.

### Which version of ESPHome has the issue?

2022.8.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.9.2, 2022.9.4

### What platform are you using?

ESP8266

### Board

WiFi-NodeM

### Component causing the issue

wifi mqtt

### Example YAML snippet

```yaml
esphome:
  name: robot-dyn
  
esp8266:
  board: esp01_1m
  framework:
    version: recommended

# Enable logging
logger:

# Enable Home Assistant API
api:
  reboot_timeout: 0s

ota:
  safe_mode: true
  password: !secter robot_dyn_ota

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  reboot_timeout: 5min

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Robot-Dyn Fallback Hotspot"
    password: !secret wifi_fallback_password
    channel: 11

captive_portal:

i2c:

sensor:
  - platform: hdc1080
    temperature:
      name: "Temperature"
      force_update: true
    humidity:
      name: "Humidity"
      force_update: true
    update_interval: 10s

mqtt:
  broker: 192.168.1.2
  port: 1883
  username: !secret mqtt_user
  password: !secret mqtt_password
```

### Anything in the logs that might be useful for us?

```txt
[14:13:30][D][sensor:126]: 'Temperature': Sending state 24.17113 °C with 1 decimals of accuracy
[14:13:30][D][sensor:126]: 'Humidity': Sending state 67.30347 % with 0 decimals of accuracy
[14:13:30][D][hdc1080:069]: Got temperature=24.2°C humidity=67.3%
[14:13:35][W][wifi_esp8266:482]: Event: Disconnected ssid='Co Ja Pacze' bssid=[redacted] reason='Beacon Timeout'
[14:13:35][W][wifi:120]: WiFi Connection lost... Reconnecting...
[14:13:35][W][wifi:572]: Restarting WiFi adapter...
[14:13:35][I][wifi:248]: WiFi Connecting to 'Co Ja Pacze'...
[14:13:35][W][wifi_esp8266:212]: wifi_apply_hostname_(robot-dyn): lwIP error -16 on interface st (index 0)
[14:13:35][W][mqtt:264]: MQTT Disconnected: WiFi disconnected.
[14:13:40][D][sensor:126]: 'Temperature': Sending state 24.20135 °C with 1 decimals of accuracy
[14:13:40][D][sensor:126]: 'Humidity': Sending state 67.30347 % with 0 decimals of accuracy
[14:13:40][D][hdc1080:069]: Got temperature=24.2°C humidity=67.3%
[14:13:41][W][wifi_esp8266:479]: Event: Disconnected ssid='Co Ja Pacze' reason='Probe Request Unsuccessful'
[14:13:41][W][wifi:536]: Error while connecting to network.
[14:13:41][W][wifi:572]: Restarting WiFi adapter...
[14:13:46][D][wifi:373]: Starting scan...
[14:13:50][D][sensor:126]: 'Temperature': Sending state 24.17113 °C with 1 decimals of accuracy
[14:13:50][D][sensor:126]: 'Humidity': Sending state 67.30347 % with 0 decimals of accuracy
[14:13:50][D][hdc1080:069]: Got temperature=24.2°C humidity=67.3%
[14:13:52][D][wifi:388]: Found networks:
[14:13:52][I][wifi:431]: - 'Co Ja Pacze' [redacted]▂▄▆█
[14:13:52][D][wifi:433]:     Channel: 1
[14:13:52][D][wifi:434]:     RSSI: -71 dB
[14:13:52][I][wifi:248]: WiFi Connecting to 'Co Ja Pacze'...
[14:13:52][W][wifi_esp8266:212]: wifi_apply_hostname_(robot-dyn): lwIP error -16 on interface st (index 0)
[14:13:58][W][wifi_esp8266:482]: Event: Disconnected ssid='Co Ja Pacze' bssid=[redacted] reason='Association Expired'
[14:13:58][W][wifi:536]: Error while connecting to network.
[14:13:58][W][wifi:572]: Restarting WiFi adapter...
[14:13:58][I][wifi:248]: WiFi Connecting to 'Co Ja Pacze'...
[14:13:58][W][wifi_esp8266:212]: wifi_apply_hostname_(robot-dyn): lwIP error -16 on interface st (index 0)
[14:14:00][D][sensor:126]: 'Temperature': Sending state 24.16106 °C with 1 decimals of accuracy
[14:14:00][D][sensor:126]: 'Humidity': Sending state 67.49879 % with 0 decimals of accuracy
[14:14:00][D][hdc1080:069]: Got temperature=24.2°C humidity=67.5%
[14:14:01][W][wifi_esp8266:482]: Event: Disconnected ssid='Co Ja Pacze' bssid=[redacted] reason='Association Expired'
[14:14:01][W][wifi:536]: Error while connecting to network.
[14:14:01][W][wifi:572]: Restarting WiFi adapter...
```
```
[11:54:41][D][wifi:388]: Found networks:
[11:54:41][I][wifi:431]: - 'Co Ja Pacze' [redacted]▂▄▆█
[11:54:41][D][wifi:433]:     Channel: 1
[11:54:41][D][wifi:434]:     RSSI: -70 dB
[11:54:41][I][wifi:248]: WiFi Connecting to 'Co Ja Pacze'...
[11:54:41][W][wifi_esp8266:212]: wifi_apply_hostname_(robot-dyn): lwIP error -16 on interface st (index 0)
[11:54:43][W][wifi_esp8266:482]: Event: Disconnected ssid='Co Ja Pacze' bssid=[redacted] reason='Association Expired'
[11:54:43][W][wifi:536]: Error while connecting to network.
[11:54:43][I][wifi:248]: WiFi Connecting to 'Co Ja Pacze'...
[11:54:43][W][wifi_esp8266:212]: wifi_apply_hostname_(robot-dyn): lwIP error -16 on interface st (index 0)
[11:54:45][W][wifi_esp8266:482]: Event: Disconnected ssid='Co Ja Pacze' bssid=[redacted] reason='Association Expired'
[11:54:45][W][wifi:536]: Error while connecting to network.
[11:54:48][D][sensor:126]: 'Temperature': Sending state 23.62731 °C with 1 decimals of accuracy
[11:54:48][D][sensor:126]: 'Humidity': Sending state 62.28638 % with 0 decimals of accuracy
[11:54:48][D][hdc1080:069]: Got temperature=23.6°C humidity=62.3%
[11:54:50][D][wifi:373]: Starting scan...
[11:54:56][D][wifi:388]: Found networks:
[11:54:56][I][wifi:431]: - 'Co Ja Pacze' [redacted]▂▄▆█
[11:54:56][D][wifi:433]:     Channel: 1
[11:54:56][D][wifi:434]:     RSSI: -71 dB
[11:54:56][I][wifi:248]: WiFi Connecting to 'Co Ja Pacze'...
[11:54:56][W][wifi_esp8266:212]: wifi_apply_hostname_(robot-dyn): lwIP error -16 on interface st (index 0)
[11:54:58][D][sensor:126]: 'Temperature': Sending state 23.64745 °C with 1 decimals of accuracy
[11:54:58][D][sensor:126]: 'Humidity': Sending state 62.38404 % with 0 decimals of accuracy
[11:54:58][D][hdc1080:069]: Got temperature=23.6°C humidity=62.4%
[11:54:58][W][wifi_esp8266:482]: Event: Disconnected ssid='Co Ja Pacze' bssid=[redacted] reason='Association Expired'
[11:54:58][W][wifi:536]: Error while connecting to network.
[11:54:58][I][wifi:248]: WiFi Connecting to 'Co Ja Pacze'...
[11:54:58][W][wifi_esp8266:212]: wifi_apply_hostname_(robot-dyn): lwIP error -16 on interface st (index 0)
```
```

filmkorn, 2022-10-27:
> lwIP error -16 on interface st

Having the same issue with all my D1 minis for about a month now. They'll eventually connect but it usually takes about 10+ minutes or a router reboot.

filmkorn, 2022-10-28:
I found I had to disable 802.11ax / WiFi 6 mode on the 2.4G network to get the Nodes (Wemos D1 mini) to reliably connect.

bartowl, 2022-11-16:
one thing get me thinking... why does tasmota work without issues when esphome does not. I looked at the available settings and have found out, that power saving is deactivated at both, yet tx power level on tasmota is set to 17dbm, while in esphome it is set to 20dbm (0-20.5 range). In addition tasmota **strongly** disadvises to set it to more than 17dbm [here](https://tasmota.github.io/docs/WifiPower/). I tried with `output_power: 15` and static IP configuration (dhcp has got longer problems connecting) and surprise surprise - wifi connected and device seems to be working... Will check for how long, but It might be that setting too high tx power has somehow damaged/degraded the chip and what forked for some time, does not work now. This is just my suspection. I do not have wifi6 and this change did make a difference. I wonder now stable will it stay...

travnick, 2022-11-27:
I've changed logging level from (default) debug to info, then it magically started to work as expected.

There is also other magic here, since enabling/disabling some modules, makes the thing working or not.