esphome/issues #3586: Native WLED UDP Protocol (wled effect) isn't implemented

digitydogs, 2022-09-17:
### The problem

When attempting to broadcast from a WLED device to an ESPHome device using the wled effect you receive a Frame Invalid error and your lights do not respond.  

The wled effect does not actually implement the native WLED UDP protocol, Wled notifier, instead it uses the generic udp protocols. 

### Which version of ESPHome has the issue?

v2022.8.3

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

All Versions

### What platform are you using?

ESP8266

### Board

Wemos D1 Mini

### Component causing the issue

wled_light_effect.cpp

### Example YAML snippet

```yaml
Any Light using the Wled effect. 

light:
  - platform: neopixelbus
    num_leds: 22
    name: "Front Room Side Window Leds"                
    type: GRB
    variant: WS2812x
    pin: D4
    restore_mode: RESTORE_DEFAULT_OFF    
    default_transition_length: 0s
    effects:
    - e131:
        universe: 1
        channels: RGB
    - wled:
       port: 21324
```

### Anything in the logs that might be useful for us?

```txt
[05:06:30][D][wled_light_effect:071]: Frame: Invalid (size=489, first=0x00).
[05:06:33][D][wled_light_effect:071]: Frame: Invalid (size=489, first=0x00).
[05:06:33][D][wled_light_effect:071]: Frame: Invalid (size=489, first=0x00).
[05:06:33][D][wled_light_effect:071]: Frame: Invalid (size=489, first=0x00).
[05:06:34][D][wled_light_effect:071]: Frame: Invalid (size=489, first=0x00).
[05:06:34][D][wled_light_effect:071]: Frame: Invalid (size=489, first=0x00)
```

### Additional information

After looking over wled_light_effect.cpp from the wled component of ESPHome, and comparing it to the actual udp sync protocol used by WLED @ https://github.com/Aircoookie/WLED/blob/main/wled00/udp.cpp, it appears that none of the wled udp notifier protocol has actually been implemented.  

Wled sends a good bit of information such as rgb colors, dim levels, transition times etc in its udp stream.  You can see from the errors in the log wled is sending a data size of 489 based off of what 11 lights on that controller are doing. 

When i used LedFx to test the udp control with an led total of 22, I would receive the same error but with a much smaller data size. 
[05:06:34][D][wled_light_effect:071]: Frame: Invalid (size=66, first=0x00)

This was because I missed a note in LedFx stating the byte prefix needed to be set to 0201, which is the DGRB protocol. Once that was set the error vanished and LedFx was able to properly control the lights. 

I would like to propose removing the UDP control options from the Wled effect component and moving them into their own effect components like was done with E131. This would not only help reduce user confusion but should make the features simpler to maintain and document in the future, as a change to the way one protocol works would no longer potentially affect the others. 

The Wled effect could then focus on adding in the various effects and palettes from Wled and implementing its native version of UDP control. (Wled notifier)

P3RF3CTION, 2022-10-12:
I can confirm the Error.

Same happens on my current HAS Installation with ESPHome latest release 2022.9.4.
Using a ESP32 D1 Mini.
I tried switching the LED Types, and the Port already:

`[08:17:22][D][light:108]:   Effect: 'WLED'
[08:17:27][D][wled_light_effect:071]: Frame: Invalid (size=489, first=0x00).
[08:17:27][D][wled_light_effect:071]: Frame: Invalid (size=489, first=0x00).`

With a switched port:
`[12:27:51][D][wled_light_effect:071]: Frame: Invalid (size=44, first=0xFF)`

lplassman, 2022-11-27:
Any updates to this? I am experiencing the same issue using the latest ESPHome.

kristof-dragon, 2023-02-15:
Any chance this could be fixed as it is still not working as of 02/2023

digitydogs, 2023-02-16:
I do not believe it has been integrated officially yet, but this add on component will allow you to have proper ddp control over your leds.  I have been using this for about 2 months now, and it was used in my christmas light show on 19 devices 8 hours a day without a single flaw. 
https://github.com/KaufHA/common/blob/main/components/README.md#ddp

closed 2024-02-21 by jesserockz