esphome/issues #3590: HMC5883L <---> QMC5883L doc need improvement

renarena, 2022-09-18:
### The problem

There are 2 different sensors, HMC5883L and QMC5883L.
If its named GY-271 and looks like HMC5883L that doesnt mean you have a HMC5883L!
But you dont know this and esphome cannot read anything, if you have the one mimicking the other.

If you get an error about could not read anything and  your esp log says.
```
[i2c.arduino:053]: Results from i2c bus scan:
[i2c.arduino:059]: Found i2c device at address 0x0D
```
you actually have the QMC5883L, which has other address and need other values.

### Which version of ESPHome has the issue?

each / website documentation

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Component causing the issue

HMC5883L

### Example YAML snippet

```yaml
sensor:
#doesnt work if you have a qmc5883l mimicking hmc5883l
  - platform: hmc5883l
    address: 0x68
    oversampling: 1x
    range: 130uT
    update_interval: 60s

works, but maybe yo dont know there is another one you need to use
  - platform: qmc5883l
    address: 0x0D
    oversampling: 64x      # 512x (default), 256x, 128x, 64x
    range: 800uT           # 200 uT, 800 uT
```

### Anything in the logs that might be useful for us?

```txt
[i2c.arduino:053]: Results from i2c bus scan:
[i2c.arduino:059]: Found i2c device at address 0x0D
```

### Additional information

On https://esphome.io/components/sensor/hmc5883l.html add a warning about breakout boards using another sensor then in technical data and add a  link to https://esphome.io/components/sensor/qmc5883l.html (and maybe vice versa)

renarena, 2023-01-17:
I think a note about / link to the other sensor would prevent lots of hassle for someone having the wrong sensor.

scottbob09, 2025-04-06:
wait, hold on.  this is broken.  still broken.   
I cannot get ANY, yes ANY i2c to function any more.
I have tried to use templates, uninstalled, etc.  as in my previous posts.  I2c is 100% broken the second you declare it.
Speaking of....
The tag line of esphome is......
"Build your own smart home devices using ESPHome, no programming experience required.."

So don't rocket strike on on how I don't know how to code...

Followed standard instructions for a standard device, that's supposed to compatible, and using the instructions given, it ALWAYS fails.

  File "/esphome/esphome/components/qmc5883l/sensor.py", line 29, in <module>
    "QMC5883LComponent", cg.PollingComponent, i2c.I2CDevice
                                              ^^^^^^^^^^^^^
AttributeError: module 'esphome.components.i2c' has no attribute 'I2CDevice'
Failed config

sensor.qmc5883l: [source /config/esphome/esphome-web-55b9c4.yaml:26]
  
  Platform not found: 'sensor.qmc5883l'.
  platform: qmc5883l

Or I2C not found, or whatever.  If I set it to scan, it shouldn't matter if I have a sensor or _something_ to tie it to.
Which seems like what ESPhome is waiting for.  despite not giving that direction.   Instead, we get... I2CDevice not found?
Why?!?!?