esphome/issues #3591: ST7789V2 TFT LCD

AlexPotterOnGit, 2022-09-18:
### The problem

ST7789V2 240 (H)x280 (V) display doesn't work with Adafruit RR 280x240 config option.

### Which version of ESPHome has the issue?

2022.8.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.9.4

### What platform are you using?

ESP32

### Board

ESP-WROOM-32

### Example YAML snippet

```yaml
esphome:
  name: displaytest

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "hIWOHuD9rOU3rKEejX3ILNoqCzkSc/Xv2kMbPfIATr8="

ota:
  password: "b9d5afca7fb4627cb9ddcc2c81e5996b"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Displaytest Fallback Hotspot"
    password: "lYa7jM7Zks23"

captive_portal:

spi:
  clk_pin: GPIO18
  mosi_pin: GPIO23

color:
  - id: my_red
    red: 100%
    green: 0%
    blue: 0%
  - id: my_yellow
    red: 100%
    green: 100%
    blue: 0%
  - id: my_green
    red: 0%
    green: 100%
    blue: 0%
  - id: my_blue
    red: 0%
    green: 0%
    blue: 100%
  - id: my_gray
    red: 50%
    green: 50%
    blue: 50%

font:
  - file: "Helvetica.ttf"
    id: helvetica_48
    size: 48
  - file: "Helvetica.ttf"
    id: helvetica_24
    size: 24
  - file: "Helvetica.ttf"
    id: helvetica_12
    size: 12
  - file: "gfonts://Roboto"
    id: roboto16
    size: 16
  - file: "gfonts://Roboto"
    id: roboto26
    size: 26  

switch:
  - platform: gpio
    pin: GPIO5
    name: "Backlight"
    id: backlight

time:
  - platform: homeassistant
    id: esptime

display:
  - platform: st7789v
    model: Adafruit RR 280x240
    backlight_pin: GPIO5
    cs_pin: GPIO15
    dc_pin: GPIO2
    reset_pin: GPIO4
    rotation: 180
    lambda: |-
      it.print(0, 0, id(roboto26), "Hello World!");
```

### Anything in the logs that might be useful for us?

```txt
not really
```

AlexPotterOnGit, 2022-09-18:
https://www.aliexpress.com/item/1005003754253714.html?spm=a2g0o.order_list.0.0.71e71802uUTrin
Exactly the display model

nochlab1, 2022-12-03:
Did you ever get this to work?

nielsnl68, 2022-12-08:
I'm currently working on a new display platform that will use the Arduino_GFX library. 
As soon i finish it i ping you guys to test it out.

AlexPotterOnGit, 2022-12-08:
Great news, appreciated for support and update!

nielsnl68, 2023-03-28:
Sadly not yet sorry, am doing a big house renovation atm, creating my own mancave so i can do more development etc.
And i have been overthinking how to go next with that project because the idea i had before was not what i expected.

Papadoma, 2023-05-30:
Any news on this? Is it working now?
I have a screen on order and would like to know if it works or not. If it doesn't, then I will skip ESPhome altogether.

michelinostreet, 2024-02-17:
I'm having exactly the same problem with ESP8266.
Latest tft ST7789 1.33" , SPI interface, model GMT130 v1.0 and model YY-PS130 v4.0 dated 2023 and 2024 do not work.
On the contrary previous tft ST7789 1.33" model FP-133H010 dated 2021 worked very well.
Below the configuration for ESP8266 working fine:
Library used:        Adafruit_GFX.h 
                            Adafruit_ST7789.h
Initialized as:       tft.init(240, 240, SPI_MODE2)
Pinout connection for ESP8266 nodemcu V3 to tft ST7789 1.33".
TFT_RST      4   (  D2 )  ----> RES  pin of ST7789
TFT_DC       5    ( D1 )  ----> DC   pin of ST7789
TFT_CS       -1    witout CS pin piu -1
TFT_MOSI  13 ( D7 )   ---->  SDA   pin of ST7789
TFT_SCLK   14  ( D5 )  ---->  SCL   pin of 7789
G                                -----> GND pin of ST7789
3v                              -----> VCC   pin of ST7789
3v                             -----> BLK     pin of ST7789

nielsnl68, 2024-02-17:
Please show the yaml file and the test image created with the yaml example in: https://esphome.io/components/display/#color-test-pattern

michelinostreet, 2024-02-17:
Solved just now. For this new tft 1.3 display, when init use spi mode 3
rarher than spi mode 2 and all works fine. Thank for interest.

JGAguado, 2024-02-24:
Dear @michelinostreet, can you share your config.yaml? I am retaking the project, but so far I'm still having issues. I'm using this display: https://www.aliexpress.com/item/1005005238299349.html?spm=a2g0o.order_list.order_list_main.75.5a5c1802mGSiGr, but should be compatible...

Thanks and best regards.

michelinostreet, 2024-02-25:
Hi,
I do not know that particular item.
Anyway for my experience ST7789 display tft  7 pins  240*240 bought
before year 2022 were using SPI_MODE2
items after 2022 must be configured as SPI_MODE3
https://it.aliexpress.com/item/1005002287169427.html?spm=a2g0o.order_list.order_list_main.61.7df73696u6uCoA&gatewayAdapt=glo2ita

#include <Adafruit_GFX.h>    // Core graphics library
#include <Adafruit_ST7789.h> // Hardware-specific library for ST7789

#define TFT_RST   4    // D2
#define TFT_DC    5    // D1
#define TFT_CS    -1   // without CS pin piu -1
#define TFT_MOSI  13   // D7 fixed pin
#define TFT_SCLK  14   // D5 fixed pin
  //    VCC        3.3v
  //    GND        ground
  //    BLK        3.3v
#define ST7789_DRIVER     // Configure all registers
#define TFT_WIDTH           240
#define TFT_HEIGHT          240
#define SPI_FREQUENCY       40000000
#define SPI_READ_FREQUENCY  20000000
Adafruit_ST7789 tft = Adafruit_ST7789(TFT_CS, TFT_DC, TFT_RST);
tft.init(240, 240, SPI_MODE3);          // Init ST7789 240x240

step135, 2024-08-16:
What about ST7789V2 from WaveShare. How to make it work?

nvtkaszpir, 2025-01-03:
ST7789V2 works with ESPHome version 2024.12.2 pretty well

```yaml
display:
  - platform: ili9xxx
    model: ST7789V
    color_order: rgb
    auto_clear_enabled: false
    update_interval: never
    id: touch_display
    cs_pin: GPIO5
    dc_pin: GPIO2
    reset_pin: GPIO4
    invert_colors: true
    dimensions:
      height: 320
      width: 240
    #show_test_card: true
    data_rate: 80MHz
```

closed 2025-01-06 by clydebarrow