esphome/issues #3597: ttgo t-display st7789v wont display on 2022.9.0

landonr, 2022-09-21:
### The problem

I get a black screen after updating to 2022.9.0 on my TTGo Tdisplay st7789v. The backlight turns on and the display lambda gets called but in my test yaml a rectangle doesn't get drawn

### Which version of ESPHome has the issue?

2022.9.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

featheresp32

### Example YAML snippet

```yaml
esphome:
  name: tdisplaytest
  platform: ESP32
  board: featheresp32

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Tdisplaytest Fallback Hotspot"
    password: "W3EzIrtpixFo"

captive_portal:

color:
  - id: color_red
    red: 1
    green: 0
    blue: 0

spi:
  clk_pin: GPIO18
  mosi_pin: GPIO19 

# TODO brightness
display:
  - platform: st7789v
    model: TTGO_TDISPLAY_135X240
    backlight_pin: GPIO4
    cs_pin: GPIO5
    dc_pin: GPIO16
    reset_pin: GPIO23
    rotation: 90°
    #brightness: 0.5
    update_interval: 5s
    lambda: |-
      ESP_LOGD("TEST", "display");
      it.rectangle(0, 0, 40, 40, id(color_red));
```

### Additional information

it works on 2022.8.3

xavitronator, 2022-09-21:
Same problem. TTGO t-display doesn't show anything with 2022.9.0

accik, 2022-09-21:
Is yours online? My Lilygo T5 is bootlooping after update to 9.0 and thus not displaying anything.
It connects to HA api, syncs time and then resets.

xavitronator, 2022-09-21:
Mine is online. I have connected a CO2 sensor (MH-Z19) to it and the graph in HomeAssistant shows continous values last hours. 
I can see only one error in the HAS log, maybe from the actualization time:

`2022-09-21 13:44:38.987 WARNING (MainThread) [aioesphomeapi.reconnect_logic] Can't connect to ESPHome API for ttgo1 @ 192.168.1.96: Hello timed out`

zigomatichub, 2022-09-21:
May be link to this introduction of change in font: https://github.com/esphome/esphome/pull/3573

closed 2022-09-21 by jesserockz