esphome/issues #3598: Eastron SDM Energy Monitor multiple devices not working

frank8the9tank, 2022-09-21:
### The problem

I have 4 eastron sdm energy meters.
1 is working fine, but the other 3 are not communicating with the esp32.
They are connected in parallel over a rs485 bus.
Tried many things with address and parity and baudrates, but don't get the other 3 to works.
So my thinking is now is the SDM platform even capable for more that 1 device?
Or is this not possible?

### Which version of ESPHome has the issue?

2022.8.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.9.5

### What platform are you using?

ESP32

### Board

nodemcu + 485 extension

### Component causing the issue

Eastron SDM Energy Monitor

### Example YAML snippet

```yaml
esphome:
  name: modbus-meter

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: arduino

# Enable logging
logger:
  baud_rate: 0
#  level: VERBOSE

# Enable Home Assistant API
api:

ota:
  password: "a025c6db8b1b7f034820e823580d4a3c"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.X.XX
    gateway: 192.168.X.X
    subnet: 255.255.255.0
    dns1: 192.168.X.X
    dns2: 192.168.X.X
  

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Modbus-Meter Fallback Hotspot"
    password: "gvMzyGNlhFpB"

captive_portal:

uart:
  rx_pin: 3
  tx_pin: 1
  baud_rate: 9600
  stop_bits: 1
  parity: EVEN

modbus:
  flow_control_pin: 5
  id: modbus1

#modbus_controller:
#  - id: epever
#    ## the Modbus device addr
#    address: 0x2
#    modbus_id: modbus1
#    setup_priority: -10

sensor:
  - platform: sdm_meter
    address: 10
    phase_a:
      current:
        name: "Car charger current"
        accuracy_decimals: 3
        id: car_current
      active_power:
        name: "Car charger power"
        accuracy_decimals: 0
        id: car_power
    import_active_energy:
      name: "Car charger kwh"
      accuracy_decimals: 3
    update_interval: 60s
    
  - platform: template
    name: "Car charger current corrected"
    unit_of_measurement: "A"
    accuracy_decimals: 3
    lambda: !lambda |-
      float car_current_corrected;
      if (id(car_power).state < 1) {
        car_current_corrected = 0;
      } else {
        car_current_corrected = (id(car_current).state);
      }
      return car_current_corrected;

  - platform: sdm_meter
    address: 2
    phase_a:
      current:
        name: "Nibe heatpump f2120 phase 1 current"
        id: current1
      active_power:
        name: "Nibe heatpump f2120 phase 1 power"
        id: power1
    phase_b:
      current:
        name: "Nibe heatpump f2120 phase 2 current"
        id: current2
      active_power:
        name: "Nibe heatpump f2120 phase 2 power"
        id: power2
    phase_c:
      current:
        name: "Nibe heatpump f2120 phase 3 current"
        id: current3
      active_power:
        name: "Nibe heatpump f2120 phase 3 power"
        id: power3
    export_active_energy:
      name: "Nibe heatpump f2120 kwh"
      accuracy_decimals: 1
    update_interval: 60s
    
  - platform: template
    name: "Nibe heatpump f2120 power"
    unit_of_measurement: "W"
    accuracy_decimals: 1
    id: totalpower
    lambda: !lambda |-
      float totalpower;
      totalpower = abs( (id(power1).state) + (id(power2).state) + (id(power3).state) );
      return totalpower;
      
  - platform: template
    name: "Nibe heatpump f2120 current"
    unit_of_measurement: "A"
    accuracy_decimals: 3
    id: current
    lambda: !lambda |-
      float current;
      current = ( (id(current1).state) + (id(current2).state) + (id(current3).state) ) / 3;
      return current;

  - platform: sdm_meter
    address: 30
    phase_a:
      current:
        name: "Daikin airco stairs current"
        accuracy_decimals: 3
      active_power:
        name: "Daikin airco stairs power"
        accuracy_decimals: 0
    import_active_energy:
      name: "Daikin airco stairs kwh"
      accuracy_decimals: 3
    update_interval: 60s

  - platform: sdm_meter
    address: 40
    phase_a:
      current:
        name: "Zehnder q450 current"
        accuracy_decimals: 3
      active_power:
        name: "Zehnder q450 power"
        accuracy_decimals: 0
    import_active_energy:
      name: "Zehnder q450 kwh"
      accuracy_decimals: 3
    update_interval: 60s
```

### Anything in the logs that might be useful for us?

```txt
I've replaced the IP address with X.X
```

### Additional information

I don't think its a hardware issue, as all settings are the same for the devices, cabling is the same,
I like to rule out the software options, so i want to make sure that the platform can handle more devices at the same time.

SeByDocKy, 2022-09-21:
I guess you need to change the modbus adress not only in ESPhome but the more important into the device itself... Maybe there is a software to do that...

frank8the9tank, 2022-09-21:
The addres in de device is set correctly. So the address matches with the yaml file.

sknsean, 2022-10-29:
@frank8the9tank any updates on this one?

frank8the9tank, 2022-11-03:
Yes i found the problem,
2 things at the same time, when moving form 2 to 4 devices i had to communication.
turns out a bad max485 chip, working enough for 2 but not for 4 devices. 
and one eastron meter did not save it communication setting spitting out random 485 messages.
bought a m5 stack nano with 485 that solved the com problems, and a new eastron meter.
So turned out no software issue on the esphome side.

closed 2022-11-03 by ssieb