esphome/issues #3599: Config editor doesn't scroll when accessed via Android app

cvwillegen, 2022-09-21:
### The problem

The latest version, 2022.9.0, includes a new config editor. When I click "Edit" in the interface, the file is opened on screen, but I can't scroll the text. 

This happens when I start the editor via the Esphome dashboard in the Android Companion App.

I tried grabbing the scroll thumb, but that didn't work either. 

### Which version of ESPHome has the issue?

2022.9.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.9.5

### What platform are you using?

ESP8266

### Additional information

[image]

ElVit, 2022-09-21:
Same issue on the iOS companion app.

3ative, 2022-09-24:
It's an issue with the new editor...
From Microsoft - **"The Monaco editor is not supported in mobile browsers or mobile web frameworks."**

However, I believe there is work being done on this.

ssieb, 2022-09-24:
Duplicate of https://github.com/esphome/dashboard/issues/304

closed 2022-09-24 by ssieb