esphome/issues #3600: "Total Power" reporting zero on Eastron SDM230m

mprowe, 2022-09-21:
### The problem

I am using the sample code from here: [SDM Meter](https://esphome.io/components/sensor/sdm_meter.html).
My modified code is included (in full) below.
When run, the displayed value for "Total Power" is shown as zero.
[image]

### Which version of ESPHome has the issue?

2022.9.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.9.5

### What platform are you using?

ESP32

### Board

doit-devkit-v1

### Component causing the issue

MODBUS SDM_METER total_power

### Example YAML snippet

```yaml
esphome:
  name: "esp32-sdm230m"
  platform: esp32
  board: esp32doit-devkit-v1

#The source of the "Healthy" LED IS: https://community.home-assistant.io/t/blink-how/307243/13
# output:
#   - platform: gpio
#     pin:
#       number: GPIO2
#       mode: output
#     id: Heathy
# interval:
#   - interval: 2s
#     then:
#       - output.turn_on: Heathy
#       - delay: 10ms
#       - output.turn_off: Heathy

#The source of "status LED" is: https://community.home-assistant.io/t/status-led-status-led-light-binary-light-light-components/443263/2
#Healthy = 0.2Hz (once every 5sec)
#Warning = 1Hz
#Error   = 2Hz
light:
  - platform: status_led
    pin: GPIO2
    id: led_status_light
    restore_mode: ALWAYS_ON
    internal: true
    effects:
    - strobe:
        name: "Slow Blink" # 0.05s on, 5s off
        colors:
          - state: true
            brightness: 100%
            duration: 5ms
          - state: false
            duration: 5s
    on_turn_on:
      - light.turn_on:
          id: led_status_light
          effect: "Slow Blink"
          
  #framework:
  #  type: arduino

# Enable logging
logger:
  baud_rate: 0
  level: VERY_VERBOSE

api:
#  encryption:
#    key: "yDpM1UDh5Pg/y2MxkbJ3sA7ZUrP7LXLR8fSMtMS9Eic="

ota:
  password: "a4169742d0654537e5068529e6545c51"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32-Sdm230M Fallback Hotspot"
    password: "1FMrHaHV7m7F"

captive_portal:

uart:
  - id: uart1
    # RX/TX are swopped to make PCB layout easer.
    tx_pin: GPIO16 #marked as RX2 on DOIT DEVKIT V1
    rx_pin: GPIO17 #marked as TX2.
    baud_rate: 9600
    stop_bits: 1
    parity: none

modbus:
  uart_id: uart1
  id: modbus1
  #No flow control required because using UART/RS485 module
  #flow_control_pin: GPIO18 #marked as D18 on DOIT DEVKIT V1

modbus_controller:
  - id: sdm230m
    address: 0x01
    modbus_id: modbus1
    update_interval: 10s

sensor:
  - platform: sdm_meter
  
    phase_a:
      current:
        name: "ESP32.SDM230M Current"
      voltage:
        name: "ESP32.SDM230M Voltage"
      active_power:
        name: "ESP32.SDM230M Active Power (P)"
      apparent_power:
        name: "ESP32.SDM230M Apparent Power (S)"
      reactive_power:
        name: "ESP32.SDM230M Reactive Power (Q)"
      power_factor:
        name: "ESP32.SDM230M Power Factor"
      phase_angle:
        name: "ESP32.SDM230M Phase Angle"
        
    frequency:
      name: "ESP32.SDM230M Frequency"
    total_power:
      name: "ESP32.SDM230M Total Power"
    import_active_energy:
      name: "ESP32.SDM230M Import Active Energy"
    export_active_energy:
      name: "ESP32.SDM230M Export Active Energy"
    import_reactive_energy:
      name: "ESP32.SDM230M Import Reactive Energy"
    export_reactive_energy:
      name: "ESP32.SDM230M Export Reactive Energy"
    update_interval: 10s
```

### Anything in the logs that might be useful for us?

```txt
## I don't think that this is significant(?)
## I think it is just telling me that it has read what it was told to read....
## My uninformed suspicion is that we are looking at the wrong address?

[14:47:29][D][sdm_meter:069]: SDMMeter: F=49.903 Hz, Im.A.E=51.130 Wh, Ex.A.E=0.042 Wh, Im.R.E=1.152 VARh, Ex.R.E=30.087 VARh, T.P=0.000 W
[14:47:29][V][sensor:076]: 'ESP32.SDM230M Total Power': Received new state 0.000000
[14:47:29][D][sensor:127]: 'ESP32.SDM230M Total Power': Sending state 0.00000 W with 2 decimals of accuracy
[14:47:29][VV][api.service:140]: send_sensor_state_response: SensorStateResponse {
  key: 3161811911
  state: 0
  missing_state: NO
}

## Is this just because we are reading the whole block of 80 address...
## ... rather than just the ones we want? 
[14:47:29][V][component:199]: Component modbus took a long time for an operation (1.46 s).
[14:47:29][V][component:200]: Components should block for at most 20-30ms.
```

### Additional information

I can't find a datasheet or address map for the Eastron SDM230m V2.0.
This is my device: [SDM230M[Datasheet]May22.pdf](https://github.com/esphome/issues/files/9617425/SDM230M.Datasheet.May22.pdf)
This is my schematic:
[image]

Fontelek, 2023-03-02:
a mi me ocurre lo mismo, haber si tenemos suerte y alguien que sepa nos da una lección

henrykuijpers, 2025-04-25:
I have the same thing going on with SDM120M. 

See here, the modbus protocol for:
* SDM120: https://www.eastroneurope.com/images/uploads/products/protocol/SDM120-MODBUS_Protocol.pdf
* SDM230: https://midsummerwholesale.co.uk/pdfs/Solax-meter-chint-Datasheet.pdf (couldn't find an Eastron link)
* SDM630: https://www.eastroneurope.com/images/uploads/products/protocol/SDM630_MODBUS_Protocol.pdf

As you can see in the code, we are obtaining the "Total Power" measurement here: 
https://github.com/esphome/esphome/blob/4c8f5275f9a82ad47c5687b9bc7393e6cd36e0ed/esphome/components/sdm_meter/sdm_meter.cpp#L60

It's this value that is being used (which refers to the modbus register): 
https://github.com/esphome/esphome/blob/4c8f5275f9a82ad47c5687b9bc7393e6cd36e0ed/esphome/components/sdm_meter/sdm_meter_registers.h#L32

If you scroll through the PDFs, you will notice that the SDM630 has this register, however, the SDM120 & SDM230 do not have this register. The "Total Power" for SDM120 & SDM230 is stored in register `0x000C` instead of `0x0034` for SDM630.

henrykuijpers, 2025-04-25:
@jesserockz Any desired way to fix this? It looks like we need to differentiate between the 3-phase meters and the 1-phase meters; Probably an additional configuration flag would be needed for that. 

It's not possible to check the configured phases, as one could define a SDM630 and reading only 1 phase. So a configuration flag would really be the way to go.

Probably further analysis of the modbus protocol for the various SDM meters is also needed, to figure out if more adjustments need to be made.

jesserockz, 2025-05-22:
I have an SDM230, and I was testing it with the PR before it was originally merged and it must have been all working correctly for me, otherwise I would not have merged it. I don't know where it is now, and it is not in use so it's not easy for me to test again.

If there is a register difference though, yes we can add a `model` config for this component and handle it that way. Could also use that to block people from trying to configure phase 2 and 3 for SDM230

henrykuijpers, 2025-05-22:
@jesserockz I was not saying that you made a bug or anything, in fact, I see in the sheet for the sdm230 that the register is matching with what is used in the code. 

It really looks like Eastron did something a bit less expected, as they are in fact using the same registers throughout their product, but for this specific one they differentiate. 

Indeed, it would be good to add this and it would also be good to block people from adding phase 2 & 3 for their non-sdm630 devices (or not, I mean, they would totally realize they are doing something wrong anyway). 

Can I help with adding the switch for the register address? I don't know how we should really do this in the code. I mainly work on Java open source projects. 
I also don't have a sdm230. :P Only the sdm630 & sdm120.