esphome/issues #3601: Default tab length is 4 instead of 2

Protoncek, 2022-09-21:
### The problem

As title says: default tab length is now 4 and thus many times unusable. Can it be reverted back to 2 ?

### Which version of ESPHome has the issue?

2022.9.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.9

### What platform are you using?

ESP8266

ssieb, 2022-09-21:
I assume you're referring to the dashboard editor.  That's an issue for https://github.com/esphome/dashboard/issues

closed 2022-09-21 by ssieb

Protoncek, 2022-09-22:
It concerns new "monaco" editor. But, it's pretty odd that no-one noticed that yet..??? Since yaml is notoriously picky regarding indentations this is pretty big bug, don't you think?

ssieb, 2022-09-22:
Normally people don't use tabs, you should be using spaces.  But this is still the wrong repo to be filing the issue in.  The editor is part of the dashboard.  This one is for the esphome device code.

3ative, 2022-09-24:
@ssieb If that's the case, why then does the editor add a Tab (and not two 'Spaces') when you add a new line?

3ative, 2022-09-24:
Not here, it adds 2 x 'Space' when I hit Return

3ative, 2022-09-24:
it is done in https://github.com/esphome/dashboard/pull/305.