esphome/issues #3603: xtensa-lx106-elf-g++: fatal error

breimann, 2022-09-22:
### The problem

I can no longer update or create a new device. This started happening about 2 weeks ago.

It only produces this error with esp8266 devices (of which most of mine are). I have one esp32 and this updates completely fine.

### Which version of ESPHome has the issue?

2022.9.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.9.6

### What platform are you using?

ESP8266

### Board

nodemcu

### Component causing the issue

components don't seem to fix it

### Example YAML snippet

```yaml
esphome:
  name: bathroom-sensor

esp8266:
  board: nodemcuv2

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "D7YAFzIYkcjZP8kL26xuFLKPcVI8g0EPY0gjGBS0Mus="

ota:
  password: "57b623f17f329506494254afdc12e58f"
  
sensor:
  - platform: dht
    pin: D7
    model: dht22
    temperature:
      name: "Bathroom Temp"
    humidity:
      name: "Bathroom Humidity"
    update_interval: 30s

binary_sensor:
  - platform: gpio
    pin: D6
    name: "Bathroom Motion"
    device_class: motion

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Bathroom-Sensor Fallback Hotspot"
    password: "zXQna82DcjrH"

captive_portal:
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/kitchen-sensor.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing kitchen-sensor (board: nodemcuv2; framework: arduino; platform: platformio/espressif8266 @ 3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- ESPAsyncTCP-esphome @ 1.2.3
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- ESPAsyncTCP-esphome @ 1.2.3
|   |-- Hash @ 1.0
|   |-- ESP8266WiFi @ 1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
|-- NeoPixelBus @ 2.6.9
|   |-- SPI @ 1.0
Compiling /data/kitchen-sensor/.pioenvs/kitchen-sensor/src/esphome/components/api/api_frame_helper.cpp.o
Compiling /data/kitchen-sensor/.pioenvs/kitchen-sensor/src/esphome/components/binary_sensor/automation.cpp.o
Compiling /data/kitchen-sensor/.pioenvs/kitchen-sensor/src/esphome/components/binary_sensor/filter.cpp.o
Compiling /data/kitchen-sensor/.pioenvs/kitchen-sensor/src/esphome/components/captive_portal/captive_portal.cpp.o
xtensa-lx106-elf-g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
*** [/data/kitchen-sensor/.pioenvs/kitchen-sensor/src/esphome/components/api/api_frame_helper.cpp.o] Error 1
xtensa-lx106-elf-g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
*** [/data/kitchen-sensor/.pioenvs/kitchen-sensor/src/esphome/components/binary_sensor/automation.cpp.o] Error 1
========================= [FAILED] Took 35.63 seconds =========================
```

Izikd, 2022-09-28:
Same issue here.
Tried also to use "Clean build files", didn't help.

```
INFO Reading configuration /config/esphome/sonoff-s26-3.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing sonoff-s26-3 (board: esp01_1m; framework: arduino; platform: platformio/espressif8266 @ 3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- ESPAsyncTCP-esphome @ 1.2.3
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- ESPAsyncTCP-esphome @ 1.2.3
|   |-- Hash @ 1.0
|   |-- ESP8266WiFi @ 1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
Compiling /data/sonoff-s26-3/.pioenvs/sonoff-s26-3/src/esphome/components/api/api_frame_helper.cpp.o
Compiling /data/sonoff-s26-3/.pioenvs/sonoff-s26-3/src/esphome/components/api/api_server.cpp.o
Compiling /data/sonoff-s26-3/.pioenvs/sonoff-s26-3/src/esphome/components/api/list_entities.cpp.o
Compiling /data/sonoff-s26-3/.pioenvs/sonoff-s26-3/src/esphome/components/api/proto.cpp.o
Compiling /data/sonoff-s26-3/.pioenvs/sonoff-s26-3/src/esphome/components/api/subscribe_state.cpp.o
Compiling /data/sonoff-s26-3/.pioenvs/sonoff-s26-3/src/esphome/components/api/user_services.cpp.o
Compiling /data/sonoff-s26-3/.pioenvs/sonoff-s26-3/src/esphome/components/binary_sensor/automation.cpp.o
Compiling /data/sonoff-s26-3/.pioenvs/sonoff-s26-3/src/esphome/components/binary_sensor/binary_sensor.cpp.o
xtensa-lx106-elf-g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
*** [/data/sonoff-s26-3/.pioenvs/sonoff-s26-3/src/esphome/components/api/subscribe_state.cpp.o] Error 1
========================= [FAILED] Took 48.00 seconds =========================

```

ssieb, 2022-09-28:
This is most likely because you're using a pi3 which is not enough to compile esphome while running HA.

BaumannSt, 2022-09-30:
I'm having a similar problem. About two weeks ago I updated HA and ESPHome and afterwards I was not able to update the ESPHome device that was created a few weeks ago.
Also no new device could be created (ESP32 and ESP80266). Now, after updating HA(2022.9.7 with all updates) and ESPHome (v2022.9.2) the update for the old ESPHome device works fine now, but creating a new device still fails. I tried 'clean build files', reboot and restart of the system. Nothing changes on the create new device behavior.
Any ideas?
btw. flashing the same device via python3 and esphome works fine.
Below see the log of my failed attempt on HA:

INFO Reading configuration /config/esphome/test.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing test (board: d1_mini; framework: arduino; platform: platformio/espressif8266 @ 3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- ESPAsyncTCP-esphome @ 1.2.3
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- ESPAsyncTCP-esphome @ 1.2.3
|   |-- Hash @ 1.0
|   |-- ESP8266WiFi @ 1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
Compiling /data/test/.pioenvs/test/src/esphome/components/api/api_pb2.cpp.o
Compiling /data/test/.pioenvs/test/src/esphome/components/api/api_server.cpp.o
Compiling /data/test/.pioenvs/test/src/esphome/components/api/list_entities.cpp.o
Compiling /data/test/.pioenvs/test/src/esphome/components/api/proto.cpp.o
xtensa-lx106-elf-g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
*** [/data/test/.pioenvs/test/src/esphome/components/api/list_entities.cpp.o] Error 1
xtensa-lx106-elf-g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
*** [/data/test/.pioenvs/test/src/esphome/components/api/api_pb2.cpp.o] Error 1
========================= [FAILED] Took 21.56 seconds =========================

ssieb, 2022-09-30:
As I already mentioned, it won't work on a pi3 if you're also running HA on it.

BaumannSt, 2022-09-30:
Oh, excuse me I forgot to mention that I created the existing ESP32 device on a RPi4. Now with the update I could neither create a ESP32 or ESP8266 device in ESPHome. Could you give me a hint which logs I could deliver to clear up the problem.

Izikd, 2022-10-02:
@ssieb 

I didn't see any changelog regarding RPi3 not supported anymore.
I was able to overcome this issue by forcing a lower build parallelism (`-j ``nproc`` ` -> `-j 2`).
I think it worth to reconsider not to force RPi3 owners to throw their HW to the dumpster.

```
./esphome/esphome/platformio_api.py

def run_platformio_cli_run(config, verbose, *args, **kwargs) -> Union[str, int]:
-    command = ["run", "-d", CORE.build_path]
+    command = ["run", "-j", "2", "-d", CORE.build_path]	
```

ssieb, 2022-10-02:
It's not that it's not "supported" any more.  No hardware is specifically supported.  But if it doesn't have enough resources to do the task, then it's not going to work.
Thank you for finding where to add that parameter.  I searched for a long time, but couldn't find how to change the jobs number.  However, that's going to be a difficult thing to modify for most people using esphome as an HA add-on.
And sure, it's *possible* to compile on a pi3, but whether it's useful depends on how long you're willing to wait for it.

Izikd, 2022-10-11:
Well it worked with RPI3 till this version of ESPHOME.
Can it be that build process enabled parallel build? That how it feels from
the log. Can it be disabled?

Izikd, 2022-10-11:
I believe that it worth to treat this issue in at least some way.
Documenting this limitation is a direction too, as I think it's not straight forward for most of the users.

But for the long run I do believe that exposing the `-j <num>` in some way (e.g. HA add-on settings page) is the right way.
Working with the default `-j` which is actually ` -j ``nproc`` `, can fail in more scenarios.

For example, ESPHome addon is just a one dockarized addon, but if the user has few/many addons that are consuming resources, then trying to have all processors for a ESPHome compilation might be a bad idea.

Also notice that @BaumannSt reported that he has issues with RPI4, for example.

breimann, 2022-10-11:
Interestingly, I've moved my HA completely over to a windows VM, and I'm
compiling fine.

But this caused a lot of frustration for me for quite some time.

Izikd, 2022-11-12:
I believe this issue can be close after this was merged: https://github.com/esphome/esphome/pull/3952

ssieb, 2022-11-12:
Yes, I lost track of this one.

closed 2022-11-12 by ssieb