esphome/issues #3604: Captive portal causes WiFi problems on sonoff basics

pjmsullivan, 2022-09-22:
### The problem

After updating my devices to 2022.9.1, two sonoff basics couldn't connect to WiFi.  After manually connecting to the devices over serial, erasing flash, uploading esphome, and checking the logs, I discovered the following:
- With my binary_fan and captive portal enabled, the sonoff basic could not find my WiFi network whatsoever:
`esphome no matching network found`
- With default esphome sonoff basic config and captive portal enabled, the sonoff basic could locate the WiFi network, but would see the network, but could not connect
`esphome Event: Disconnected ssid='[redacted]' reason='Probe Request Unsuccessful'`
- With my binary_fan and captive portal disabled, the sonoff basic connects to WiFi as expected.

### Which version of ESPHome has the issue?

2022.9.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.9.6

### What platform are you using?

ESP8266

### Board

sonoff basic

### Component causing the issue

captive portal (I believe)

### Example YAML snippet

```yaml
esphome:
  name: bedroomfan
  comment: Controls the bedroom fan
esp8266:
  board: esp8285

wifi:
  ssid: "[redacted]"
  password: "[redacted]"

  ap:
    ssid:  "BedroomFan"
    password: "[redacted]"

  domain: .home.[redacted].com

captive_portal:

logger:

api:
  password: "[redacted]"
  reboot_timeout: 5min

ota:
  password: "[redacted]"

text_sensor:
  - platform: version
    name: "Bedroom Fan ESPHome Version"

switch:
  - platform: restart
    name: "Bedroom Fan Restart"

fan:
  - platform: binary
    output: relay
    name: "Bedroom Fan"
    id: bedroomfan

output:
  - platform: gpio
    pin: GPIO12
    id: relay

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: 
        input: true
        pullup: true
      inverted: true
    name: "Bedroom Fan Button"
    on_press:
      - fan.toggle: bedroomfan

status_led:
  pin:
    number: GPIO13
    inverted: yes
```

### Additional information

Thank you so much for this amazing project.

robertlarue, 2022-09-24:
I’m having this same issue with two sonoff basics. 

One has a DHT22 and an SGP30 sensor. The other is fairly basic with just the internal relay. 

```yaml
esphome:
  name: air-quality

esp8266:
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: [redacted]

ota:
  password: [redacted]

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Air-Quality Fallback Hotspot"
    password: [redacted]

captive_portal:
    
i2c:
  scl: GPIO3
  sda: GPIO1

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode:
        input: true
        pullup: true
      inverted: true
    name: Button
    on_press:
      - switch.toggle: relay

switch:
  - platform: gpio
    name: Switch
    pin: GPIO12
    id: relay
    
status_led:
  pin:
    number: GPIO13
    inverted: true

sensor:
  - platform: sgp30
    eco2:
      name: CO2
      accuracy_decimals: 1
    tvoc:
      name: TVOC
      accuracy_decimals: 1
    compensation:
      temperature_source: temperature
      humidity_source: humidity
    update_interval: 60s
  - platform: dht
    pin: GPIO14
    temperature:
      name: Temperature
      id: temperature
    humidity:
      name: Humidity
      id: humidity
    update_interval: 60s

mspinolo, 2022-09-24:
I have same issue with a sonoff basic, more details here
[Link to home assistant forum thread](https://community.home-assistant.io/t/sonoff-basic-r1-code-change-cant-connect-to-wifi/225414/5)

closed 2022-12-06 by pjmsullivan