esphome/issues #3605: ESPHome 2022.9.1 VL53 issue

REHolt, 2022-09-22:
### The problem

After installing the 2022.9.1 update my VL53 TOF sensor stopped working.  All data is shown as "Unavailable". All I get is:

Level 1': Sending state nan in with 5 decimals of accuracy

Looks like there may be a I2C bus problem?

### Which version of ESPHome has the issue?

2022.9.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.9.6

### What platform are you using?

ESP32

### Board

nodemcu

### Component causing the issue

VL53 

### Example YAML snippet

```yaml
switch:
  - platform: restart
    name: "TOF-Test Device Restart"

  # The relay control pin (local only)
  - platform: gpio
    name: "relay1"
    pin: GPIO17
    inverted: false
    id: relay1

i2c:
  - id: i2cbus
    sda: GPIO21 
    scl: GPIO22
    scan: True

sensor:    
  - platform: vl53l0x
    name: "Sensor 1"
    id: sens_1
    i2c_id: i2cbus
    address: 0x41
    enable_pin: GPIO05
    timeout: 200us
    update_interval: 500ms
    unit_of_measurement: "m"  
    accuracy_decimals: 5
    long_range: true
    signal_rate_limit: 0.3
    filters:
      # - calibrate_linear:
      #     # Map 2" (from sensor) to 34.5" (true value)
      #     - 0.0508 -> 0.0635
      #     - 0.2032 -> 0.227584
      #     - 0.8763 -> 0.75184
      - sliding_window_moving_average:
          window_size: 10
          send_every: 1

  - platform: vl53l0x
    name: "Sensor 2"
    id: sens_2
    i2c_id: i2cbus
    address: 0x42
    enable_pin: GPIO19
    timeout: 200us
    update_interval: 500ms
    unit_of_measurement: "m"
    accuracy_decimals: 5
    long_range: true
    signal_rate_limit: 0.3
    filters:
      # - calibrate_linear:
      #     # Map 2" (from sensor) to 34.5" (true value)
      #     - 0.0508 -> 0.0635
      #     - 0.2032 -> 0.227584
      #     - 0.8763 -> 0.75184
      - sliding_window_moving_average:
          window_size: 10
          send_every: 1

  - platform: template
    name: "Level 1"
    id: level1
    unit_of_measurement: "in"
    lambda: |-
      return ((id(sens_1).state * 39.370));
    accuracy_decimals: 5
    update_interval: 10s
    on_value_range:
        - below: 3
          then:
          - delay: .5min
          - switch.turn_off: relay1
        - above: 14
          then:
          - switch.turn_on: relay1

  - platform: template
    name: "Available Water"
    id: available_water
    unit_of_measurement: "gal"
    lambda: |-
      return (((32.75-(id(level1).state-2.14)) * 1.7));
    accuracy_decimals: 1
    update_interval: 10s
```

ssieb, 2022-09-22:
Does it show up in the i2c scan?  Watch the serial logs from boot to see if there are any messages about why the sensor isn't working.

REHolt, 2022-09-23:
Sorry for the delay, just saw this.  Yes it does show up in the scan. Below is a snippet of the serial log:

[06:58:45][I][app:102]: ESPHome version 2022.9.1 compiled on Sep 22 2022, 12:09:27
[06:58:45][C][wifi:502]: WiFi:
[06:58:45][C][wifi:360]:   Local MAC: 58:BF:25:9E:D4:40
[06:58:45][C][wifi:361]:   SSID: [redacted]
[06:58:45][C][wifi:362]:   IP Address: 192.168.1.143
[06:58:45][C][wifi:364]:   BSSID: [redacted]
[06:58:45][C][wifi:365]:   Hostname: 'tof-test'
[06:58:45][C][wifi:367]:   Signal strength: -62 dB ▂▄▆█
[06:58:45][C][wifi:371]:   Channel: 1
[06:58:45][C][wifi:372]:   Subnet: 255.255.255.0
[06:58:45][C][wifi:373]:   Gateway: 192.168.1.254
[06:58:45][C][wifi:374]:   DNS1: 192.168.1.254
[06:58:45][C][wifi:375]:   DNS2: 0.0.0.0
[06:58:45][C][logger:275]: Logger:
[06:58:45][C][logger:276]:   Level: DEBUG
[06:58:45][C][logger:277]:   Log Baud Rate: 115200
[06:58:45][C][logger:278]:   Hardware UART: UART0
[06:58:45][C][i2c.arduino:038]: I2C Bus:
[06:58:45][C][i2c.arduino:039]:   SDA Pin: GPIO21
[06:58:45][C][i2c.arduino:040]:   SCL Pin: GPIO22
[06:58:45][C][i2c.arduino:041]:   Frequency: 50000 Hz
[06:58:45][C][i2c.arduino:044]:   Recovery: bus successfully recovered
[06:58:45][I][i2c.arduino:054]: Results from i2c bus scan:
[06:58:45][I][i2c.arduino:060]: Found i2c device at address 0x41
[06:58:45][C][switch.gpio:050]: GPIO Switch 'relay1'
[06:58:45][C][switch.gpio:051]:   Pin: GPIO17
[06:58:45][C][switch.gpio:073]:   Restore Mode: Restore (Defaults to OFF)
[06:58:45][C][template.sensor:023]: Template Sensor 'Level 1'
[06:58:45][C][template.sensor:023]:   State Class: ''
[06:58:45][C][template.sensor:023]:   Unit of Measurement: 'in'
[06:58:45][C][template.sensor:023]:   Accuracy Decimals: 5
[06:58:45][C][template.sensor:024]:   Update Interval: 10.0s
[06:58:45][C][template.sensor:023]: Template Sensor 'Available Water'
[06:58:45][C][template.sensor:023]:   State Class: ''
[06:58:45][C][template.sensor:023]:   Unit of Measurement: 'gal'
[06:58:45][C][template.sensor:023]:   Accuracy Decimals: 1
[06:58:45][C][template.sensor:024]:   Update Interval: 10.0s
[06:58:45][C][restart:022]: Restart Switch 'TOF-Test Device Restart'
[06:58:45][C][restart:022]:   Icon: 'mdi:restart'
[06:58:45][C][vl53l0x:024]: VL53L0X 'Sensor 1'
[06:58:45][C][vl53l0x:024]:   State Class: 'measurement'
[06:58:45][C][vl53l0x:024]:   Unit of Measurement: 'm'
[06:58:45][C][vl53l0x:024]:   Accuracy Decimals: 5
[06:58:45][C][vl53l0x:024]:   Icon: 'mdi:arrow-expand-vertical'
[06:58:45][C][vl53l0x:025]:   Update Interval: 0.5s
[06:58:45][C][vl53l0x:026]:   Address: 0x29
[06:58:45][C][vl53l0x:028]:   Enable Pin: GPIO5
[06:58:45][C][vl53l0x:030]:   Timeout: 200us
[06:58:45][C][vl53l0x:024]: VL53L0X 'Sensor 2'
[06:58:45][C][vl53l0x:024]:   State Class: 'measurement'
[06:58:45][C][vl53l0x:024]:   Unit of Measurement: 'm'
[06:58:45][C][vl53l0x:024]:   Accuracy Decimals: 5
[06:58:45][C][vl53l0x:024]:   Icon: 'mdi:arrow-expand-vertical'
[06:58:45][C][vl53l0x:025]:   Update Interval: 0.5s
[06:58:45][C][vl53l0x:026]:   Address: 0x29
[06:58:45][C][vl53l0x:028]:   Enable Pin: GPIO19
[06:58:45][C][vl53l0x:030]:   Timeout: 200us
[06:58:45][C][captive_portal:088]: Captive Portal:
[06:58:45][C][mdns:094]: mDNS:
[06:58:45][C][mdns:095]:   Hostname: tof-test
[06:58:45][C][ota:089]: Over-The-Air Updates:
[06:58:45][C][ota:090]:   Address: tof-test.local:3232
[06:58:45][C][ota:093]:   Using Password.
[06:58:45][C][api:138]: API Server:
[06:58:45][C][api:139]:   Address: tof-test.local:6053
[06:58:45][C][api:141]:   Using noise encryption: YES
[06:58:48][D][sensor:127]: 'Level 1': Sending state nan in with 5 decimals of accuracy
[06:58:54][D][sensor:127]: 'Available Water': Sending state nan gal with 1 decimals of accuracy
[06:58:58][D][sensor:127]: 'Level 1': Sending state nan in with 5 decimals of accuracy

ssieb, 2022-09-23:
I'm not sure how re-addressing works with the scan.  Do you actually have two sensors?  I don't see any attempt to update the sensors.  Reduce the update interval and increase the logging to VERBOSE.

REHolt, 2022-09-23:
Thanks for the reply. 
 The scan will go out and try to verify each device is present. I usually have 2 devices ox41 and 0x42. each is accompanied by a chip select.  currently only have 1 sensor attached, but that was working just prior to the update; however, when I ran the update on this device it stopped posting data and everything came back as unavailable. I will change the update interval and logging and post the results back here.

REHolt, 2022-09-23:
here is the serial log with Verbose logging(looks the same to me):  

INFO Reading configuration /config/esphome/tof-test.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing tof-test (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 1.0
|-- FS @ 1.0
|-- Update @ 1.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 1.1.0
|-- ESPmDNS @ 1.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
|-- Wire @ 1.0.1
Compiling /data/tof-test/.pioenvs/tof-test/src/main.cpp.o
Linking /data/tof-test/.pioenvs/tof-test/firmware.elf
RAM:   [=         ]  13.0% (used 42572 bytes from 327680 bytes)
Flash: [=====     ]  54.9% (used 1008314 bytes from 1835008 bytes)
Building /data/tof-test/.pioenvs/tof-test/firmware.bin
esp32_create_combined_bin(["/data/tof-test/.pioenvs/tof-test/firmware.bin"], ["/data/tof-test/.pioenvs/tof-test/firmware.elf"])
Wrote 0x106320 bytes to file /data/tof-test/.pioenvs/tof-test/firmware-factory.bin, ready to flash to offset 0x0
========================= [SUCCESS] Took 25.43 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of tof-test.local
INFO  -> 192.168.1.143
INFO Uploading /data/tof-test/.pioenvs/tof-test/firmware.bin (1008416 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from tof-test.local using esphome API
INFO Successfully connected to tof-test.local
[16:55:37][I][app:102]: ESPHome version 2022.9.1 compiled on Sep 23 2022, 16:54:51
[16:55:37][C][wifi:502]: WiFi:
[16:55:37][C][wifi:360]:   Local MAC: 58:BF:25:9E:D4:40
[16:55:37][C][wifi:361]:   SSID: [redacted]
[16:55:37][C][wifi:362]:   IP Address: 192.168.1.143
[16:55:37][C][wifi:364]:   BSSID: [redacted]
[16:55:37][C][wifi:365]:   Hostname: 'tof-test'
[16:55:37][C][wifi:367]:   Signal strength: -67 dB ▂▄▆█
[16:55:37][C][wifi:371]:   Channel: 1
[16:55:37][C][wifi:372]:   Subnet: 255.255.255.0
[16:55:37][C][wifi:373]:   Gateway: 192.168.1.254
[16:55:37][C][wifi:374]:   DNS1: 192.168.1.254
[16:55:37][C][wifi:375]:   DNS2: 0.0.0.0
[16:55:37][C][logger:275]: Logger:
[16:55:37][C][logger:276]:   Level: DEBUG
[16:55:37][C][logger:277]:   Log Baud Rate: 115200
[16:55:37][C][logger:278]:   Hardware UART: UART0
[16:55:37][C][i2c.arduino:038]: I2C Bus:
[16:55:37][C][i2c.arduino:039]:   SDA Pin: GPIO21
[16:55:37][C][i2c.arduino:040]:   SCL Pin: GPIO22
[16:55:37][C][i2c.arduino:041]:   Frequency: 50000 Hz
[16:55:37][C][i2c.arduino:044]:   Recovery: bus successfully recovered
[16:55:37][I][i2c.arduino:054]: Results from i2c bus scan:
[16:55:37][I][i2c.arduino:060]: Found i2c device at address 0x41
[16:55:37][C][switch.gpio:050]: GPIO Switch 'relay1'
[16:55:37][C][switch.gpio:051]:   Pin: GPIO17
[16:55:37][C][switch.gpio:073]:   Restore Mode: Restore (Defaults to OFF)
[16:55:37][C][template.sensor:023]: Template Sensor 'Level 1'
[16:55:37][C][template.sensor:023]:   State Class: ''
[16:55:37][C][template.sensor:023]:   Unit of Measurement: 'in'
[16:55:37][C][template.sensor:023]:   Accuracy Decimals: 5
[16:55:37][C][template.sensor:024]:   Update Interval: 10.0s
[16:55:37][C][template.sensor:023]: Template Sensor 'Available Water'
[16:55:37][C][template.sensor:023]:   State Class: ''
[16:55:37][C][template.sensor:023]:   Unit of Measurement: 'gal'
[16:55:37][C][template.sensor:023]:   Accuracy Decimals: 1
[16:55:37][C][template.sensor:024]:   Update Interval: 10.0s
[16:55:37][C][restart:022]: Restart Switch 'TOF-Test Device Restart'
[16:55:37][C][restart:022]:   Icon: 'mdi:restart'
[16:55:37][C][vl53l0x:024]: VL53L0X 'Sensor 1'
[16:55:37][C][vl53l0x:024]:   State Class: 'measurement'
[16:55:37][C][vl53l0x:024]:   Unit of Measurement: 'm'
[16:55:37][C][vl53l0x:024]:   Accuracy Decimals: 5
[16:55:37][C][vl53l0x:024]:   Icon: 'mdi:arrow-expand-vertical'
[16:55:37][C][vl53l0x:025]:   Update Interval: 0.3s
[16:55:37][C][vl53l0x:026]:   Address: 0x29
[16:55:37][C][vl53l0x:028]:   Enable Pin: GPIO5
[16:55:37][C][vl53l0x:030]:   Timeout: 200us
[16:55:37][C][vl53l0x:024]: VL53L0X 'Sensor 2'
[16:55:37][C][vl53l0x:024]:   State Class: 'measurement'
[16:55:38][C][vl53l0x:024]:   Unit of Measurement: 'm'
[16:55:38][C][vl53l0x:024]:   Accuracy Decimals: 5
[16:55:38][C][vl53l0x:024]:   Icon: 'mdi:arrow-expand-vertical'
[16:55:38][C][vl53l0x:025]:   Update Interval: 0.5s
[16:55:38][C][vl53l0x:026]:   Address: 0x29
[16:55:38][C][vl53l0x:028]:   Enable Pin: GPIO19
[16:55:38][C][vl53l0x:030]:   Timeout: 200us
[16:55:38][C][captive_portal:088]: Captive Portal:
[16:55:38][C][mdns:094]: mDNS:
[16:55:38][C][mdns:095]:   Hostname: tof-test
[16:55:38][C][ota:089]: Over-The-Air Updates:
[16:55:38][C][ota:090]:   Address: tof-test.local:3232
[16:55:38][C][ota:093]:   Using Password.
[16:55:38][C][api:138]: API Server:
[16:55:38][C][api:139]:   Address: tof-test.local:6053
[16:55:38][C][api:141]:   Using noise encryption: YES
[16:55:42][D][sensor:127]: 'Level 1': Sending state nan in with 5 decimals of accuracy
[16:55:43][D][sensor:127]: 'Available Water': Sending state nan gal with 1 decimals of accuracy
[16:55:52][D][sensor:127]: 'Level 1': Sending state nan in with 5 decimals of accuracy
[16:55:53][D][sensor:127]: 'Available Water': Sending state nan gal with 1 decimals of accuracy
[16:56:02][D][sensor:127]: 'Level 1': Sending state nan in with 5 decimals of accuracy
[16:56:03][D][sensor:127]: 'Available Water': Sending state nan gal with 1 decimals of accuracy

REHolt, 2022-09-23:
here is another with the update interval changed to 800ms (the one above was set to 300ms and the original was at 500ms).
INFO Reading configuration /config/esphome/tof-test.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing tof-test (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 1.0
|-- FS @ 1.0
|-- Update @ 1.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 1.1.0
|-- ESPmDNS @ 1.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
|-- Wire @ 1.0.1
Compiling /data/tof-test/.pioenvs/tof-test/src/main.cpp.o
Linking /data/tof-test/.pioenvs/tof-test/firmware.elf
RAM:   [=         ]  13.0% (used 42572 bytes from 327680 bytes)
Flash: [=====     ]  54.9% (used 1008314 bytes from 1835008 bytes)
Building /data/tof-test/.pioenvs/tof-test/firmware.bin
esp32_create_combined_bin(["/data/tof-test/.pioenvs/tof-test/firmware.bin"], ["/data/tof-test/.pioenvs/tof-test/firmware.elf"])
Wrote 0x106320 bytes to file /data/tof-test/.pioenvs/tof-test/firmware-factory.bin, ready to flash to offset 0x0
========================= [SUCCESS] Took 25.38 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of tof-test.local
INFO  -> 192.168.1.143
INFO Uploading /data/tof-test/.pioenvs/tof-test/firmware.bin (1008416 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from tof-test.local using esphome API
INFO Successfully connected to tof-test.local
[16:59:31][I][app:102]: ESPHome version 2022.9.1 compiled on Sep 23 2022, 16:58:36
[16:59:31][C][wifi:502]: WiFi:
[16:59:31][C][wifi:360]:   Local MAC: 58:BF:25:9E:D4:40
[16:59:31][C][wifi:361]:   SSID: [redacted]
[16:59:31][C][wifi:362]:   IP Address: 192.168.1.143
[16:59:31][C][wifi:364]:   BSSID: [redacted]
[16:59:31][C][wifi:365]:   Hostname: 'tof-test'
[16:59:31][C][wifi:367]:   Signal strength: -66 dB ▂▄▆█
[16:59:31][C][wifi:371]:   Channel: 1
[16:59:31][C][wifi:372]:   Subnet: 255.255.255.0
[16:59:31][C][wifi:373]:   Gateway: 192.168.1.254
[16:59:31][C][wifi:374]:   DNS1: 192.168.1.254
[16:59:31][C][wifi:375]:   DNS2: 0.0.0.0
[16:59:31][C][logger:275]: Logger:
[16:59:31][C][logger:276]:   Level: DEBUG
[16:59:31][C][logger:277]:   Log Baud Rate: 115200
[16:59:31][C][logger:278]:   Hardware UART: UART0
[16:59:31][C][i2c.arduino:038]: I2C Bus:
[16:59:31][C][i2c.arduino:039]:   SDA Pin: GPIO21
[16:59:31][C][i2c.arduino:040]:   SCL Pin: GPIO22
[16:59:31][C][i2c.arduino:041]:   Frequency: 50000 Hz
[16:59:31][C][i2c.arduino:044]:   Recovery: bus successfully recovered
[16:59:31][I][i2c.arduino:054]: Results from i2c bus scan:
[16:59:31][I][i2c.arduino:060]: Found i2c device at address 0x41
[16:59:31][C][switch.gpio:050]: GPIO Switch 'relay1'
[16:59:31][C][switch.gpio:051]:   Pin: GPIO17
[16:59:31][C][switch.gpio:073]:   Restore Mode: Restore (Defaults to OFF)
[16:59:31][C][template.sensor:023]: Template Sensor 'Level 1'
[16:59:31][C][template.sensor:023]:   State Class: ''
[16:59:31][C][template.sensor:023]:   Unit of Measurement: 'in'
[16:59:31][C][template.sensor:023]:   Accuracy Decimals: 5
[... 26 lines cut ...]
[16:59:31][C][captive_portal:088]: Captive Portal:
[16:59:31][C][mdns:094]: mDNS:
[16:59:31][C][mdns:095]:   Hostname: tof-test
[16:59:31][C][ota:089]: Over-The-Air Updates:
[16:59:31][C][ota:090]:   Address: tof-test.local:3232
[16:59:31][C][ota:093]:   Using Password.
[16:59:31][C][api:138]: API Server:
[16:59:31][C][api:139]:   Address: tof-test.local:6053
[16:59:31][C][api:141]:   Using noise encryption: YES
[16:59:39][D][sensor:127]: 'Available Water': Sending state nan gal with 1 decimals of accuracy
[16:59:39][D][sensor:127]: 'Level 1': Sending state nan in with 5 decimals of accuracy
[16:59:49][D][sensor:127]: 'Available Water': Sending state nan gal with 1 decimals of accuracy
[16:59:49][D][sensor:127]: 'Level 1': Sending state nan in with 5 decimals of accuracy
[16:59:59][D][sensor:127]: 'Available Water': Sending state nan gal with 1 decimals of accuracy
[16:59:59][D][sensor:127]: 'Level 1': Sending state nan in with 5 decimals of accuracy
[17:00:09][D][sensor:127]: 'Available Water': Sending state nan gal with 1 decimals of accuracy
[17:00:09][D][sensor:127]: 'Level 1': Sending state nan in with 5 decimals of accuracy
[17:00:12][D][api:102]: Accepted ::FFFF:C0A8:163
[17:00:12][D][api.connection:861]: Home Assistant 2022.9.6 (::FFFF:C0A8:163): Connected successfully
[17:00:19][D][sensor:127]: 'Available Water': Sending state nan gal with 1 decimals of accuracy
[17:00:19][D][sensor:127]: 'Level 1': Sending state nan in with 5 decimals of accuracy
[17:00:29][D][sensor:127]: 'Available Water': Sending state nan gal with 1 decimals of accuracy
[17:00:29][D][sensor:127]: 'Level 1': Sending state nan in with 5 decimals of accuracy
[17:00:39][D][sensor:127]: 'Available Water': Sending state nan gal with 1 decimals of accuracy
[17:00:39][D][sensor:127]: 'Level 1': Sending state nan in with 5 decimals of accuracy

ssieb, 2022-09-23:
You didn't change the logger level.  Also, I meant to make the update interval longer, so that you won't get flooded with log messages.  Something like `10s` at least.

REHolt, 2022-09-23:
I'm not much of a software person, that's why I like ESPHome. One thing I noticed:

It appears to find the device at 0x41, but then it continues to use the default 0x29. I seem to recall that previously it remembered the address, but I don't have an old log file.

REHolt, 2022-09-23:
and one using Level: VERY_VERBOSE

========================= [SUCCESS] Took 69.37 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of tof-test.local
INFO  -> 192.168.1.143
INFO Uploading /data/tof-test/.pioenvs/tof-test/firmware.bin (1042768 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from tof-test.local using esphome API
INFO Successfully connected to tof-test.local
[17:46:58][I][app:102]: ESPHome version 2022.9.1 compiled on Sep 23 2022, 17:45:44
[17:46:58][C][wifi:502]: WiFi:
[17:46:58][C][wifi:360]:   Local MAC: 58:BF:25:9E:D4:40
[17:46:58][C][wifi:361]:   SSID: [redacted]
[17:46:58][C][wifi:362]:   IP Address: 192.168.1.143
[17:46:58][C][wifi:364]:   BSSID: [redacted]
[17:46:58][C][wifi:365]:   Hostname: 'tof-test'
[17:46:58][C][wifi:367]:   Signal strength: -64 dB ▂▄▆█
[17:46:58][V][wifi:369]:   Priority: 0.0
[17:46:58][C][wifi:371]:   Channel: 1
[17:46:58][C][wifi:372]:   Subnet: 255.255.255.0
[17:46:58][C][wifi:373]:   Gateway: 192.168.1.254
[17:46:58][C][wifi:374]:   DNS1: 192.168.1.254
[17:46:58][C][wifi:375]:   DNS2: 0.0.0.0
[17:46:58][VV][scheduler:196]: Running interval 'update' with interval=10000 last_execution=4294964269 (now=7022)
[17:46:58][V][sensor:076]: 'Available Water': Received new state nan
[17:46:58][D][sensor:127]: 'Available Water': Sending state nan gal with 1 decimals of accuracy
[17:46:58][C][logger:275]: Logger:
[17:46:58][C][logger:276]:   Level: VERY_VERBOSE
[17:46:58][C][logger:277]:   Log Baud Rate: 115200
[17:46:58][C][logger:278]:   Hardware UART: UART0
[17:46:58][C][i2c.arduino:038]: I2C Bus:
[17:46:58][C][i2c.arduino:039]:   SDA Pin: GPIO21
[17:46:58][C][i2c.arduino:040]:   SCL Pin: GPIO22
[17:46:58][C][i2c.arduino:041]:   Frequency: 50000 Hz
[17:46:58][C][i2c.arduino:044]:   Recovery: bus successfully recovered
[17:46:58][I][i2c.arduino:054]: Results from i2c bus scan:
[17:46:58][I][i2c.arduino:060]: Found i2c device at address 0x41
[17:46:58][C][switch.gpio:050]: GPIO Switch 'relay1'
[17:46:58][C][switch.gpio:051]:   Pin: GPIO17
[17:46:58][C][switch.gpio:073]:   Restore Mode: Restore (Defaults to OFF)
[17:46:58][C][template.sensor:023]: Template Sensor 'Level 1'
[17:46:58][C][template.sensor:023]:   State Class: ''
[17:46:58][C][template.sensor:023]:   Unit of Measurement: 'in'
[17:46:58][C][template.sensor:023]:   Accuracy Decimals: 5
[17:46:58][C][template.sensor:024]:   Update Interval: 10.0s
[17:46:58][C][template.sensor:023]: Template Sensor 'Available Water'
[17:46:58][C][template.sensor:023]:   State Class: ''
[17:46:58][C][template.sensor:023]:   Unit of Measurement: 'gal'
[17:46:58][C][template.sensor:023]:   Accuracy Decimals: 1
[17:46:58][C][template.sensor:024]:   Update Interval: 10.0s
[17:46:58][C][restart:022]: Restart Switch 'TOF-Test Device Restart'
[17:46:58][C][restart:022]:   Icon: 'mdi:restart'
[17:46:58][C][vl53l0x:024]: VL53L0X 'Sensor 1'
[17:46:58][C][vl53l0x:024]:   State Class: 'measurement'
[17:46:58][C][vl53l0x:024]:   Unit of Measurement: 'm'
[17:46:58][C][vl53l0x:024]:   Accuracy Decimals: 5
[17:46:58][C][vl53l0x:024]:   Icon: 'mdi:arrow-expand-vertical'
[17:46:58][C][vl53l0x:025]:   Update Interval: 0.7s
[17:46:58][C][vl53l0x:026]:   Address: 0x29
[17:46:58][C][vl53l0x:028]:   Enable Pin: GPIO5
[17:46:58][C][vl53l0x:030]:   Timeout: 200us
[17:46:58][C][vl53l0x:024]: VL53L0X 'Sensor 2'
[17:46:58][C][vl53l0x:024]:   State Class: 'measurement'
[17:46:58][C][vl53l0x:024]:   Unit of Measurement: 'm'
[17:46:58][C][vl53l0x:024]:   Accuracy Decimals: 5
[... 24 lines cut ...]
[17:47:08][V][sensor:076]: 'Available Water': Received new state nan
[17:47:08][D][sensor:127]: 'Available Water': Sending state nan gal with 1 decimals of accuracy
[17:47:13][VV][api.service:381]: on_ping_request: PingRequest {}
[17:47:13][VV][api.service:043]: send_ping_response: PingResponse {}
[17:47:17][VV][scheduler:196]: Running interval 'update' with interval=10000 last_execution=16799 (now=26801)
[17:47:17][V][sensor:076]: 'Level 1': Received new state nan
[17:47:17][D][sensor:127]: 'Level 1': Sending state nan in with 5 decimals of accuracy
[17:47:18][VV][scheduler:196]: Running interval 'update' with interval=10000 last_execution=16973 (now=26973)
[17:47:18][V][sensor:076]: 'Available Water': Received new state nan
[17:47:18][D][sensor:127]: 'Available Water': Sending state nan gal with 1 decimals of accuracy
[17:47:24][VV][scheduler:196]: Running interval '' with interval=60000 last_execution=4294940733 (now=33437)
[17:47:27][VV][scheduler:196]: Running interval 'update' with interval=10000 last_execution=26799 (now=36799)
[17:47:27][V][sensor:076]: 'Level 1': Received new state nan
[17:47:27][D][sensor:127]: 'Level 1': Sending state nan in with 5 decimals of accuracy
[17:47:28][VV][scheduler:196]: Running interval 'update' with interval=10000 last_execution=26973 (now=36973)
[17:47:28][V][sensor:076]: 'Available Water': Received new state nan
[17:47:28][D][sensor:127]: 'Available Water': Sending state nan gal with 1 decimals of accuracy
[17:47:28][VV][api.service:381]: on_ping_request: PingRequest {}
[17:47:28][VV][api.service:043]: send_ping_response: PingResponse {}
[17:47:37][VV][scheduler:196]: Running interval 'update' with interval=10000 last_execution=36799 (now=46802)
[17:47:37][V][sensor:076]: 'Level 1': Received new state nan
[17:47:37][D][sensor:127]: 'Level 1': Sending state nan in with 5 decimals of accuracy
[17:47:38][VV][scheduler:196]: Running interval 'update' with interval=10000 last_execution=36973 (now=46973)
[17:47:38][V][sensor:076]: 'Available Water': Received new state nan
[17:47:38][D][sensor:127]: 'Available Water': Sending state nan gal with 1 decimals of accuracy

ssieb, 2022-09-23:
I was wondering why so much was missing and I just noticed that this isn't the serial log.  The OTA log misses all the setup parts.

REHolt, 2022-09-23:
what does it look like and how do I get it?

REHolt, 2022-09-23:
Is this what you are looking for?

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
[I][logger:243]: Log initialized
[VV][esp32.preferences:070]: nvs_get_blob: key: 233825507, len: 4
[C][ota:465]: There have been 5 suspected unsuccessful boot attempts.
[VV][esp32.preferences:039]: s_pending_save: key: 233825507, len: 4
[D][esp32.preferences:113]: Saving 1 preferences to flash...
[VV][esp32.preferences:122]: Checking if NVS data 233825507 has changed
[V][esp32.preferences:125]: sync: key: 233825507, len: 4
[D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[VV][sensor:093]: Sensor(0x3ffb8f30)::add_filter(0x3ffb9060)
[VV][sensor.filter:029]: Filter(0x3ffb9060)::initialize(parent=0x3ffb8f30 next=0x0)
[VV][sensor:093]: Sensor(0x3ffb92d8)::add_filter(0x3ffb9468)
[VV][sensor.filter:029]: Filter(0x3ffb9468)::initialize(parent=0x3ffb92d8 next=0x0)
[I][app:029]: Running through setup()...
[V][app:030]: Sorting components by setup priority...
[I][i2c.arduino:161]: Performing I2C bus recovery
[V][esp32-hal-i2c.c:1492] i2cInit(): num=0 sda=21 scl=22 freq=0
[V][esp32-hal-i2c.c:1688] i2cSetFrequency(): freq=100000Hz
[V][esp32-hal-i2c.c:1704] i2cSetFrequency(): cpu Freq=240Mhz, i2c Freq=100000Hz
[V][esp32-hal-i2c.c:1712] i2cSetFrequency(): Fifo delta=1
[V][esp32-hal-i2c.c:1688] i2cSetFrequency(): freq=50000Hz
[V][esp32-hal-i2c.c:1704] i2cSetFrequency(): cpu Freq=240Mhz, i2c Freq=50000Hz
[V][esp32-hal-i2c.c:1712] i2cSetFrequency(): Fifo delta=1
3[VV][i2c.arduino:126]: 0x08 TX 
[VV][i2c.arduino:150]: TX failed: not acknowledged
[VV][i2c.arduino:126]: 0x09 TX 
[VV][i2c.arduino:150]: TX failed: not acknowledged
[VV][i2c.arduino:126]: 0x0A TX 
[VV][i2c.arduino:150]: TX failed: not acknowledged
[VV][i2c.arduino:126]: 0x0B TX 
[VV][i2c.arduino:150]: TX failed: not acknowledged
[VV][i2c.arduino:126]: 0x0C TX 
[VV][i2c.arduino:150]: TX failed: not acknowledged
[VV][i2c.arduino:126]: 0x0D TX 
[VV][i2c.arduino:150]: TX failed: not acknowledged
[VV][i2c.arduino:126]: 0x0E TX 
[VV][i2c.arduino:150]: TX failed: not acknowledged
[VV][i2c.arduino:126]: 0x0F TX 
[VV][i2c.arduino:150]: TX failed: not acknowledged
[VV][i2c.arduino:126]: 0x10 TX 
[VV][i2c.arduino:150]: TX failed: not acknowledged
[VV][i2c.arduino:126]: 0x11 TX 
[VV][i2c.arduino:150]: TX failed: not acknowledged
[VV][i2c.arduino:126]: 0x12 TX 
[VV][i2c.arduino:150]: TX failed: not acknowledged
[VV][i2c.arduino:126]: 0x13 TX 
[VV][i2c.arduino:150]: TX failed: not acknowledged
[VV][i2c.arduino:126]: 0x14 TX 
[VV][i2c.arduino:150]: TX failed: not acknowledged
[VV][i2c.arduino:126]: 0x15 TX 
[VV][i2c.arduino:150]: TX failed: not acknowledged
[VV][i2c.arduino:126]: 0x16 TX 
[VV][i2c.arduino:150]: TX failed: not acknowledged
[VV][i2c.arduino:126]: 0x17 TX 
[VV][i2c.arduino:150]: TX failed: not acknowledged
[VV][i2c.arduino:126]: 0x18 TX 
[VV][i2c.arduino:150]: TX failed: not acknowledged
[VV][i2c.arduino:126]: 0x19 TX 
[... 633 lines cut ...]
  unique_id: 'tof-testswitchtof-test_device_restart'
  icon: 'mdi:restart'
  assumed_state: NO
  disabled_by_default: NO
  entity_category: ENTITY_CATEGORY_CONFIG
  device_class: ''
}
[VV][api.service:148]: send_list_entities_switch_response: ListEntitiesSwitchResponse {
  object_id: 'relay1'
  key: 391537629
  name: 'relay1'
  unique_id: 'tof-testswitchrelay1'
  icon: ''
  assumed_state: NO
  disabled_by_default: NO
  entity_category: ENTITY_CATEGORY_NONE
  device_class: ''
}
[VV][api.service:055]: send_list_entities_done_response: ListEntitiesDoneResponse {}
[VV][api.service:417]: on_subscribe_states_request: SubscribeStatesRequest {}
[VV][api.service:479]: on_subscribe_homeassistant_services_request: SubscribeHomeassistantServicesRequest {}
[VV][api.service:140]: send_sensor_state_response: SensorStateResponse {
  key: 2575470155
  state: nan
  missing_state: YES

ssieb, 2022-09-23:
That's what I was looking for.  When did it last work?  Does it work if you fully power cycle the device and sensor?

REHolt, 2022-09-23:
When I loaded the yaml file with the cable connected to my laptop, it erased the device and then reloaded software (using OTA it only loaded software). It now appears to work.

Sorry to have wasted your time. I do appreciate the effort though!
Thanks for the help.
Robert

closed 2023-08-07 by ssieb