esphome/issues #3608: esphome it does not work under Windows 11 (python 3.10.7 -- pip3 install esphome)

lucora, 2022-09-23:
### The problem

Hi there,
after installing Python and running the "pip3 install esphome" installation command everything seems ok. Then I launch the command "esphome dashboard config /" and if I open the browser to the page "http: // localhost: 6052 /" the application seems to work. I try to insert a new device and when I want to generate the firmware it implants in the black screen without any reference. I also tried the prompt wizard but it doesn't work, (windows 11 my S.O.)

[image]
[image]
[image]

### Which version of ESPHome has the issue?

esphome 2022.9.1, esphome-2022.10.0.dev0

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

esphome 2022.9.1 , esphome-2022.10.0.dev0

### What platform are you using?

ESP8266

lucora, 2022-09-23:
[image]

someWhoAnon, 2023-01-11:
Same problem on Windows 11 python 3.11.0
[image]