esphome/issues #3612: Growatt Component Startup spikes

sanderlv, 2022-09-24:
### The problem

Every day the PV inverter starts up  and when there is not yet enough sunlight it switches on and off a few times. This seems normal behavior. But this has the following side effect. The sensor "today gen" is being filled/poplulated a few times with the last day today gen value. Giving:
[image]

That brings my energy dashboard in a big spike.

After the inverter is on fully (enough light) is works well:
[image]

### Which version of ESPHome has the issue?

every version

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

all versions

### What platform are you using?

ESP8266

### Board

D1 Mini

### Component causing the issue

Growatt

### Example YAML snippet

```yaml
energy_production_day:
      name: "${esp_name} - Today Gen"
```

### Additional information

I think that some filter needs to be present to not allow the previous values (last day) to be read from the inverter.

sanderlv, 2022-09-24:
for now I have added:

time:
  - platform: homeassistant
    id: homeassistant_time
    on_time:
      # Every morning on weekdays reset ID todaygen
      - seconds: 0
        minutes: 30
        hours: 1
        days_of_week: MON-SUN
        then:
          - lambda: id(todaygen).publish_state(0);

Hope this helps

ssieb, 2022-09-25:
This is not the place for support requests.  You also asked this on the discord server which is the right place.

closed 2022-09-25 by ssieb

sanderlv, 2022-09-25:
I still think this is not a support question but a bug. It should not work like this. Asking for support on github was to get a work around on this.

Please reopen my issue to have a proper solution in the component please.

ssieb, 2022-09-25:
There is no bug.  The sensor is reporting what the inverter is sending.  What possible fix could be made to the component?

sanderlv, 2022-09-25:
Make sure that no bogus data is sent out during stsrtup. This helps no single user of this component. It might not be a bug in code but sure is in functionality. 

During startup of the inverter, at sunrise, the total gen daily should alway be 0 and building up. Now it bounces a few times between 0 and the last day before total day gen 🙆‍♂️

sanderlv, 2023-01-03:
Hi Samuel, how are you doing? Did you ever come in touch with the above after the last message? I do still have the spikes every day. And I begin to question if a software update to my growatt would potentially solve things. Other people do have the same spikes,and other also do not have it. Since I have totally disconnected my growatt inverter from the internet I belong to them who might be running old(er) firmware. Any thought?

eugentib, 2023-02-07:
Hello @sanderlv ,
I'm a beginner with ESP home but I solved this problem by adding a filter to the today's generation field and not sending any data while the inverter status field is zero

The relevant code is:

globals:
  - id: cstat
    type: int
    restore_value: no
    initial_value: '0'
...
    inverter_status:
      name: "Growatt Status Code"
      filters:
      - lambda: |-
          id(cstat)=x;
          return x;
...
    energy_production_day:
      name: "Growatt Today's Generation"
      filters:
        - lambda: |-
            if (id(cstat)==0) return {};
            else return x;