esphome/issues #3614: Overflow for skip_updates in modbus sensors

fra87, 2022-09-24:
### The problem

Hello

I have an issue with the different MODBUS sensors. When I try to specify a skip_updates value greater than 255 a warning is issued:

```
src/main.cpp: In function 'void setup()':
src/main.cpp:502:176: warning: unsigned conversion from 'int' to 'uint8_t' {aka 'unsigned char'} changes value from '2500' to '196' [-Woverflow]
  502 |   cancelletto_gate_time = new modbus_controller::ModbusNumber(modbus_controller::ModbusRegisterType::HOLDING, 0, 0, 0xFFFFFFFF, modbus_controller::SensorValueType::U_WORD, 1, 2500, false);
```

(in this case the issue was with a ModbusNumber, but I also experienced it with a TextSensor).

I think that the best option would be to modify the variable type, so that greater numbers can be used there (for instance I have a system where the update cycle is pretty fast - 200ms, but there are sensors which never change, so I'd like to poll them once every 10 minutes or so, while with this limitation I'm forced to update them every 50 seconds).

If this is not possible, then maybe this shall be highlighted better, because this warning can easily go unnoticed (and this is a parameter that is hard to notice is configured badly)

### Which version of ESPHome has the issue?

v2022.8.3 and v2022.9.1

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Component causing the issue

modbus_component (and its sensors)

### Example YAML snippet

```yaml
uart:
      id: mod_bus
      tx_pin: GPIO15
      rx_pin: GPIO13
      baud_rate: 9600
      stop_bits: 1

    modbus:
      flow_control_pin: GPIO5
      id: modbus1

    modbus_controller:
      - id: modbus_cancelletto
        address: 42
        modbus_id: modbus1
        setup_priority: -10
        update_interval: 200ms

    number:
      - platform: modbus_controller
        modbus_controller_id: modbus_cancelletto
        id: cancelletto_gate_time
        name: 'PulseTime'
        address: 0
        value_type: U_WORD
        unit_of_measurement: 'ms'
        skip_updates: 250
```

### Anything in the logs that might be useful for us?

```txt
src/main.cpp: In function 'void setup()':
src/main.cpp:502:176: warning: unsigned conversion from 'int' to 'uint8_t' {aka 'unsigned char'} changes value from '2500' to '196' [-Woverflow]
  502 |   cancelletto_gate_time = new modbus_controller::ModbusNumber(modbus_controller::ModbusRegisterType::HOLDING, 0, 0, 0xFFFFFFFF, modbus_controller::SensorValueType::U_WORD, 1, 2500, false);
```

fra87, 2023-01-23:
Tested again with v2022.12.3; nothing has changed

kodiakastu, 2023-11-25:
Tested in 2023.11.4 with `skip_updates: 0xFFFF` and no issues found. After a couple of reads, you can confirm the skips are decreasing, as they're supposed to.

```
[17:00:50][V][modbus_controller:148]: Range : C Size: 8 (3) skip: 65535
[17:01:20][V][modbus_controller:148]: Range : C Size: 8 (3) skip: 65534
[17:01:50][V][modbus_controller:148]: Range : C Size: 8 (3) skip: 65533
...
[17:06:20][V][modbus_controller:148]: Range : C Size: 8 (3) skip: 65524
[17:06:50][V][modbus_controller:148]: Range : C Size: 8 (3) skip: 65523
[17:07:20][V][modbus_controller:148]: Range : C Size: 8 (3) skip: 65522
```

GitforZhangXL, 2024-02-11:
This issue has been fixed in the new version, please upgrade your ESPHOME version