esphome/issues #3615: Climate Thermostat preset_change is not triggered

trunneml, 2022-09-24:
### The problem

I'm creating an thermostat for my heating and try to trigger a script when the preset is changed to boost. But the `preset_change` is not triggered at all when I change the preset via Home-Assistant.

### Which version of ESPHome has the issue?

2021.9.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant 2022.9.6

### What platform are you using?

ESP32

### Board

Sonoff NSPanel

### Component causing the issue

thermostat

### Example YAML snippet

```yaml
climate:
  - platform: thermostat
    id: ${id}
    name: ${name}
    sensor: ${sensor}
    visual:
      min_temperature: 20 °C
      max_temperature: 25 °C
      temperature_step: 0.1 °C
    default_target_temperature_low: 20.5
    default_mode: heat
    heat_overrun: 0.0 °C
    heat_deadband: 0.2 °C
    # min_heating_off_time: 60min
    # min_heating_run_time: 5min
    # min_idle_time: 30min
    min_heating_off_time: 5sec
    min_heating_run_time: 5sec
    min_idle_time: 5sec
    heat_action:
      - switch.turn_on: relay_1
    idle_action:
      - switch.turn_off: relay_1
    preset:
      - name: eco
        default_target_temperature_low: 20.5
      - name: home
        default_target_temperature_low: 21.5
      - name: boost
        default_target_temperature_low: 24
    preset_change:
        - logger.log: Preset has been changed! Checking boost
        - if:
            condition:
              lambda: |-
                return id(${id}).preset == CLIMATE_PRESET_BOOST;
            then:
              - logger.log: "Starting Boost"
              - script.execute: ${id}_boost_stop
            else:
              - logger.log: "Stopping Boost"
              - script.stop: ${id}_boost_stop
    # target_temperature_change_action:
    #   - script.stop: ${id}_boost_stop
```

trunneml, 2022-09-24:
I also tried:

```
preset_change:
  then:
    - logger.log: Preset has been changed! Checking boost
```

But that also didn't work

glmnet, 2022-10-03:
this is esphome version 2022.9 right?

trunneml, 2022-10-03:
Yes it's 2022.09. I fixed it in the description.

closed 2022-10-12 by jesserockz