esphome/issues #3617: 1 Extra sensor causes bootloop, whether logging enabled or not (not brownout)

JeroenTuinstra, 2022-09-25:
### The problem

Not using many sensors: 28 (text and normal sensors from Home Assistant). When adding 29th it goes into a bootloop. I can still have logging. No difference if I completely switch logging off. Which seems to indicate it is not a RAM issue.

### Which version of ESPHome has the issue?

2022.9.1

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.9.6

### What platform are you using?

ESP32

### Board

ESP32-WROOM-32E (HUZZAH32 Feather board)

### Component causing the issue

spi - waveshare e-paper

### Example YAML snippet

```yaml
esphome:
  name: temperature

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "6YzTu7YvbPeV/lAk+DrrUFzY6eqez2xrxEKVlvjDbas="

ota:
  password: "f63ba4eda519c5f46a751c8b5261c8ed"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.10.14
    gateway: 192.168.10.1
    subnet: 255.255.255.0

time:
  - platform: homeassistant
    id: ha_time

font:
  - file: "fonts/AcuminPro-Medium.ttf"
    id: tm24
    size: 24
  
  - file: "fonts/AcuminPro-Medium.ttf"
    id: tm35
    size: 35
[... 776 lines cut ...]
[C][wifi:361]:   SSID: 'cheztuinstra'[redacted]
[C][wifi:362]:   IP Address: 192.168.10.14
[C][wifi:364]:   BSSID: B8:BE:F4:B1:0D:53[redacted]
[C][wifi:365]:   Hostname: 'temperature'
[C][wifi:367]:   Signal strength: -40 dB ▂▄▆█
[C][wifi:371]:   Channel: 1
[C][wifi:372]:   Subnet: 255.255.255.0
[C][wifi:373]:   Gateway: 192.168.10.1
[C][wifi:374]:   DNS1: 0.0.0.0
[C][wifi:375]:   DNS2: 0.0.0.0
[C][logger:275]: Logger:
[C][logger:276]:   Level: DEBUG
[C][logger:277]:   Log Baud Rate: 115200
[C][logger:278]:   Hardware UART: UART0
[C][mdns:094]: mDNS:
[C][mdns:095]:   Hostname: temperature
[C][ota:089]: Over-The-Air Updates:
[C][ota:090]:   Address: 192.168.10.14:3232
[C][ota:093]:   Using Password.
[W][ota:099]: Last Boot was an unhandled reset, will proceed to safe mode in 0 restarts
```

### Additional information

Thought it was something similar to this issue: #1605 but this issue went stale.

ssieb, 2022-09-25:
`Debug exception reason: Stack canary watchpoint triggered (loopTask) `
It's a stack overflow.  You need to either increase the stack size or try https://github.com/esphome/esphome/pull/3577.

JeroenTuinstra, 2022-09-26:
Thanks @ssieb. I will try #3577. In the meantime how does one increas the stack size in ESPhome?

ssieb, 2022-09-26:
```yaml
esphome:
  ...
  platformio_options:
    build_flags: 
      - -DCONFIG_ARDUINO_LOOP_STACK_SIZE=16384
```

0x90ChuckTessela, 2022-12-14:
Running into this same issue on a somewhat large config.  Using esphome/esphome#3577 and the above compiler flag doesn't fix the issue.

As far as I can tell, using 

> ```yaml
> esphome:
>   ...
>   platformio_options:
>     build_flags: 
>       - -DCONFIG_ARDUINO_LOOP_STACK_SIZE=16384
> ```

doesn't actually change the stack size.  the issue continues to show up, and the generated core.cpp shows the default stack size:

```cpp
extern "C" void app_main() {
  esp32::setup_preferences();
  xTaskCreate(loop_task, "loopTask", 8192, nullptr, 1, &loop_task_handle);
}
```
The flag is properly getting set for the platformio options and all, as seen in the platformio.ini file:
```cpp
build_flags =
    -DCONFIG_ARDUINO_LOOP_STACK_SIZE=16384
```

The compilation spams warnings due to the fact that the arduino libraries spam `#define CONFIG_ARDUINO_LOOP_STACK_SIZE 8192` throughout the entirety of the codebase.

Pointing esphome to use newer versions of the arduino framework breaks with the existing ESPHome ethernet component, so I'm completely at a loss.

ssieb, 2022-12-14:
Try the latest release or maybe dev.  There have been some version updates and I think that involved ethernet updates as well.