esphome/issues #3618: problems with esp32cam board

timdaman, 2022-09-25:
### The problem

With 
* A AI-Thinker ESP32-CAM 
* Using a `esp32_camera:` component
* A esp32.board of `esp32cam`

I see the following issues
1. The unit frequently disconnects from the HomeAssistant Server
2. Images are not always available
3. View the logs usually hangs before showing the configuration.

I saw this problem with two different units.

**Doesn't work**
```
...
esp32:
  board: esp32cam
...
esp32_camera:
...
```

**Works**
```
...
esp32:
  board: esp32dev
...
esp32_camera:
...
```

I looked at the board configurations for those and couldn't find any difference that could explain what I saw.

Removing the `esp32_camera:` component reliably removed all instability. Later, I discovered changing the board type also resolved the instability while still supporting the camera component.

### Which version of ESPHome has the issue?

2022.9.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.9.6

### What platform are you using?

ESP32

### Board

AI Thinker ESP32-CAM

### Component causing the issue

esp32_camera

### Example YAML snippet

```yaml
**BROKEN**

substitutions:
  # Modify variables based on your settings
  hostname: 'esp32-cam-02'

esphome:
  name: $hostname

esp32:
  board: esp32cam
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "REDACTED"

ota:
  password: "REDACTED"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

bluetooth_proxy:
esp32_ble_tracker:

esp32_camera:
  external_clock:
    pin: GPIO0
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32

  name: Camera
  max_framerate: 1 fps
  resolution: 1600x1200
  jpeg_quality: 30

#   resolution: 1600x1200
#  contrast: 2
#  saturation: 2

output:
#flash
  - platform: ledc
    pin: GPIO4
    channel: 2 # channel 1 is used for esp32_camera
    id: flash

# statuslight
  - platform: gpio
    pin:
      number: GPIO33
      inverted: True
    id: status_led

light:
#flashlight
  - platform: monochromatic
    output: flash
    name: $hostname flash
  - platform: binary
    output: status_led
    name: $hostname status led
    
    
sensor:
  - platform: wifi_signal
    name: $hostname WiFi Signal
    update_interval: 60s
  - platform: uptime
    name: $hostname Uptime
    
text_sensor:
  - platform: version
    name: $hostname ESPHome Version
#   - platform: wifi_info
#     ssid:
#       name: $hostname WiFi

switch:      
  - platform: restart
    name: $hostname restart
    
binary_sensor:
  - platform: status
    name: $hostname status
```

**Working**
```
substitutions:
  # Modify variables based on your settings
  hostname: 'esp32-cam-02'

esphome:
  name: $hostname

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "REDACTED"

ota:
  password: "REDACTED"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

bluetooth_proxy:
esp32_ble_tracker:

esp32_camera:
  external_clock:
    pin: GPIO0
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32

  name: Camera
  max_framerate: 1 fps
  resolution: 1600x1200
  jpeg_quality: 30

#   resolution: 1600x1200
#  contrast: 2
#  saturation: 2

output:
#flash
  - platform: ledc
    pin: GPIO4
    channel: 2 # channel 1 is used for esp32_camera
    id: flash

# statuslight
  - platform: gpio
    pin:
      number: GPIO33
      inverted: True
    id: status_led

light:
#flashlight
  - platform: monochromatic
    output: flash
    name: $hostname flash
  - platform: binary
    output: status_led
    name: $hostname status led
    
    
sensor:
  - platform: wifi_signal
    name: $hostname WiFi Signal
    update_interval: 60s
  - platform: uptime
    name: $hostname Uptime
    
text_sensor:
  - platform: version
    name: $hostname ESPHome Version
#   - platform: wifi_info
#     ssid:
#       name: $hostname WiFi

switch:      
  - platform: restart
    name: $hostname restart
    
binary_sensor:
  - platform: status
    name: $hostname status
```
```

### Anything in the logs that might be useful for us?

```txt
10:36.649 -> [0;32m[I][logger:243]: Log initialized[0m
17:10:36.649 -> [0;35m[C][ota:465]: There have been 3 suspected unsuccessful boot attempts.[0m
17:10:36.649 -> [0;36m[D][esp32.preferences:113]: Saving 1 preferences to flash...[0m
17:10:36.649 -> [0;36m[D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed[0m
17:10:36.684 -> [0;32m[I][app:029]: Running through setup()...[0m
17:10:36.684 -> [0;36m[D][ledc.output:041]: Calculating resolution bit-depth for frequency 1000.000000[0m
17:10:36.684 -> [0;36m[D][ledc.output:046]: Resolution calculated as 16[0m
17:10:36.684 -> [0;35m[C][light:036]: Setting up light 'esp32-cam-02 flash'...[0m
17:10:36.684 -> [0;36m[D][light:035]: 'esp32-cam-02 flash' Setting:[0m
17:10:36.684 -> [0;36m[D][light:040]:   Color mode: [0m
17:10:36.684 -> [0;36m[D][light:084]:   Transition length: 1.0s[0m
17:10:36.718 -> [0;35m[C][light:036]: Setting up light 'esp32-cam-02 status led'...[0m
17:10:36.718 -> [0;36m[D][light:035]: 'esp32-cam-02 status led' Setting:[0m
17:10:36.718 -> [0;36m[D][light:040]:   Color mode: [0m
17:10:36.785 -> [E][camera.c:1113] camera_probe(): Detected camera not supported.
17:10:36.785 -> [E][camera.c:1379] esp_camera_init(): Camera probe failed with error 0x20004
17:10:36.785 -> [1;31m[E][esp32_camera:024]: esp_camera_init failed: ERROR[0m
17:10:36.785 -> [1;31m[E][component:112]: Component esp32_camera was marked as failed.[0m
17:10:36.819 -> [0;36m[D][text_sensor:067]: 'esp32-cam-02 ESPHome Version': Sending state '2022.9.1 Sep 25 2022, 17:08:01'[0m
17:10:36.819 -> [0;36m[D][binary_sensor:034]: 'esp32-cam-02 status': Sending initial state OFF[0m
17:10:41.825 -> E (12143) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
17:10:41.825 -> E (12143) task_wdt:  - loopTask (CPU 1)
17:10:41.825 -> E (12143) task_wdt: Tasks currently running:
17:10:41.825 -> E (12143) task_wdt: CPU 0: IDLE0
17:10:41.825 -> E (12143) task_wdt: CPU 1: loopTask
17:10:41.825 -> E (12143) task_wdt: Aborting.
17:10:41.825 -> abort() was called at PC 0x401c69d8 on core 0
17:10:41.825 -> 
17:10:41.825 -> ELF file SHA256: 0000000000000000
17:10:41.858 -> 
17:10:41.858 -> Backtrace: 0x40095338:0x3ffc0f20 0x400955b1:0x3ffc0f40 0x401c69d8:0x3ffc0f60 0x400918bd:0x3ffc0f80 0x401fe2ff:0x3ffbc100 0x401c82ab:0x3ffbc120 0x40098a51:0x3ffbc140 0x400972a6:0x3ffbc160
17:10:41.858 -> 
17:10:41.858 -> Rebooting...
17:10:41.858 -> ets Jun  8 2016 00:22:57
17:10:41.858 -> 
17:10:41.858 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
17:10:41.858 -> configsip: 0, SPIWP:0xee
17:10:41.858 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
17:10:41.895 -> mode:DOUT, clock div:2
17:10:41.895 -> load:0x3fff0030,len:184
17:10:41.895 -> load:0x40078000,len:12616
17:10:41.895 -> load:0x40080400,len:2916
17:10:41.895 -> entry 0x400805c4
```

martinrevert, 2022-10-21:
The problem persists on v2022.10.0. Thanks for hinting on how to make it work @timdaman .

damndoline, 2023-02-03:
Any news on this?

I'm using 3 esp32cam modules from AI Thinker and they only half-way work when declared as esp32dev boards. However the two issues stated by @timdaman persist. Could this be due to psram not being available for esp32dev boards?