esphome/issues #3620: `Exec format error` for ESP32, running as HA Addon, in Parallels VM, on M1 Mac Mini

tader, 2022-09-26:
### The problem

Situation:

- On an M1 Mac Mini, I am running Home Assistant `generic-aarch64` in a Parallels virtual machine
- In Home Assistant, I am running ESPHome as an addon
- From ESPHome, I am trying to update my devices

Problem:

- I can not compile for ESP32-based boards
- Error (details see details): `Exec format error`
- I can, however, compile for ESP8266-based boards)

I understand that I might be running an unusual setup, and I would be very glad if someone could help me find a workaround for this problem.

### Which version of ESPHome has the issue?

2022.9.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.9.6

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

n/a

### Example YAML snippet

```yaml
esphome:
  name: esphome-web-d312b0

esp32:
  board: esp32dev
  framework:
    type: arduino

wifi:
  networks:
    - ssid: !secret wifi_ssid
      password: !secret wifi_password
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/esphome-web-d312b0.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esphome-web-d312b0 (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- WiFi @ 1.0
|-- ESPmDNS @ 1.0
Compiling /data/esphome-web-d312b0/.pioenvs/esphome-web-d312b0/src/esphome/components/esp32/core.cpp.o
sh: 1: xtensa-esp32-elf-g++: Exec format error
Compiling /data/esphome-web-d312b0/.pioenvs/esphome-web-d312b0/src/esphome/components/esp32/gpio_arduino.cpp.o
Compiling /data/esphome-web-d312b0/.pioenvs/esphome-web-d312b0/src/esphome/components/esp32/gpio_idf.cpp.o
sh: 1: xtensa-esp32-elf-g++: Exec format error
sh: 1: xtensa-esp32-elf-g++: Exec format error
*** [/data/esphome-web-d312b0/.pioenvs/esphome-web-d312b0/src/esphome/components/esp32/core.cpp.o] Error 126
*** [/data/esphome-web-d312b0/.pioenvs/esphome-web-d312b0/src/esphome/components/esp32/gpio_arduino.cpp.o] Error 126
*** [/data/esphome-web-d312b0/.pioenvs/esphome-web-d312b0/src/esphome/components/esp32/gpio_idf.cpp.o] Error 126
Compiling /data/esphome-web-d312b0/.pioenvs/esphome-web-d312b0/src/esphome/components/esp32/preferences.cpp.o
sh: 1: xtensa-esp32-elf-g++: Exec format error
*** [/data/esphome-web-d312b0/.pioenvs/esphome-web-d312b0/src/esphome/components/esp32/preferences.cpp.o] Error 126
========================== [FAILED] Took 0.31 seconds ==========================
```

beele, 2022-10-06:
I have the same issue. This seems to be because it tries to run a 32bit lib on a 64 bit os/kernel. 
Haven't found any way around this issue unfortunately.

ceinem, 2022-10-31:
Same issue for me, but running ESPhome in a docker container on an M1 MacMini.

goldbingo, 2022-11-01:
Same issue in QEMU aarch64 VM

I'm an using this image:
haos_generic-aarch64-9.3.qcow2

Here's my detail setup:
qemu-system-aarch64 \
  -serial stdio\
  -M virt,highmem=off \
  -accel hvf \
  -cpu cortex-a72 \
  -smp 4 \
  -m 2048 \
  -bios /Users/xxxxx/qemu-vm/QEMU_EFI.fd \
  -display none \
  -device qemu-xhci \
  -device intel-hda -device hda-output \
  -audiodev none,id=noaudio \
  -drive file=/Users/xxxxx/qemu-vm/disk/haos_generic-aarch64-9.3.qcow2 \
  -nic vmnet-bridged,ifname=vlan1,mac=xx:xx:xx:xx:xx:xx

ceinem, 2022-11-01:
@beele do you know which libraries are the issue? I'm just wondering, because on the 64bit image on my Raspberry Pi 3 it runs completely fine (maybe a bit slow), but on a 64bit VM it doesn't?

beele, 2022-11-01:
No Idea, have been trying to find a solution from time to time but haven't had any luck so far

Szepi, 2022-12-02:
Same problem.

tader, 2022-12-14:
I can confirm what @beele said, for me it is working now too!

closed 2022-12-14 by tader