esphome/issues #3625: device_class: volume editor says invalid

level451, 2022-09-29:
### The problem

The editor does not allow for device_class: volume - but the link for available does show volume.

sensor:

  - platform: template
    id: shower_water_used_last_event
    name: 'Water (Last Event)'
    device_class: volume
    state_class: measurement
    unit_of_measurement: 'gal'

### Which version of ESPHome has the issue?

2022.9.2

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

node

### Example YAML snippet

```yaml
- platform: template
    id: shower_water_used_last_event
    name: 'Water (Last Event)'
    device_class: volume
    state_class: measurement
    unit_of_measurement: 'gal'
```

hubikj, 2022-10-06:
Issue is still present in v2022.9.3

shakin89, 2022-10-08:
I confirm this issue still present in 2022.9.4. 
This configuration is invalid in the editor and does not compile 
`- platform: template
    name: "Velocità del Vento Km/h"
    id: sensore_vento_kmh
    unit_of_measurement: 'Km/h'
    update_interval: ${wind_update_interval}
    accuracy_decimals: 2
    icon: 'mdi:weather-windy'
    device_class: "speed"
    #state_class: "measurement"
    filters:
      - multiply: 3.6
`
throwing this error

`sensor.template: [source /config/esphome/stazione-meteo.yaml:407]
  platform: template
  name: Velocità del Vento Km/h
  id: sensore_vento_kmh
  unit_of_measurement: Km/h
  update_interval: 5s
  accuracy_decimals: 2
  icon: mdi:weather-windy
  
  Unknown value 'speed', valid options are '', 'apparent_power', 'aqi', 'battery', 'carbon_dioxide', 'carbon_monoxide', 'current', 'date', 'duration', 'energy', 'frequency', 'gas', 'humidity', 'illuminance', 'monetary', 'nitrogen_dioxide', 'nitrogen_monoxide', 'nitrous_oxide', 'ozone', 'pm1', 'pm10', 'pm25', 'power', 'power_factor', 'pressure', 'reactive_power', 'signal_strength', 'sulphur_dioxide', 'temperature', 'timestamp', 'volatile_organic_compounds', 'voltage'.`
Home Assistant 2022.10.1
Supervisor 2022.09.1
ESPHome 2022.9.4

tomvanswam, 2023-10-28:
I ran into the same problem for `water`, which is also on the list mentioned by @hubikj 
Probably a few more are missing as well, as a quick check also showns `wind_speed` is on the ha list, but not on the valid options list

tomvanswam, 2023-10-30:
I managed to "fix" it for myself. 
After looking though the code on the ESPHome repo, I coudn't find anything that could lead to this error. All current devices_classes were implemented in the constants used.
Digging a bit further I noticed I had the wrong add-on repository installed in HA, one that was stuck on version 2022.3. So where I thought the autoupdate kept me up to date and the latest code was on my machine, it obviously was not.
Installing the correct repro from the addon-store made that I could install the previously incorrect code.