esphome/issues #3627: Wemos S2 Mini - Error: This board doesn't support arduino framework!

alfwro13, 2022-09-30:
### The problem

I am trying to install esphome on Wemos S2 S2 Mini board using board type: esp32-s2-saola-1
The process fails with error: Error: This board doesn't support arduino framework!

### Which version of ESPHome has the issue?

 v2022.9.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.9.7

### What platform are you using?

ESP32

### Board

esp32-s2-saola-1

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/garden-gate-sensor.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing garden-gate-sensor (board: esp32-s2-saola-1; framework: arduino; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
Library Manager: Installing esphome/AsyncTCP-esphome @ 1.2.2
INFO Installing esphome/AsyncTCP-esphome @ 1.2.2
Unpacking  [####################################]  100%
Library Manager: AsyncTCP-esphome@1.2.2 has been installed!
INFO AsyncTCP-esphome@1.2.2 has been installed!
Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
INFO Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
Unpacking  [####################################]  100%
Library Manager: ESPAsyncWebServer-esphome@2.1.0 has been installed!
INFO ESPAsyncWebServer-esphome@2.1.0 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing ottowinter/ESPAsyncTCP-esphome
INFO Installing ottowinter/ESPAsyncTCP-esphome
Unpacking  [####################################]  100%
Library Manager: ESPAsyncTCP-esphome@1.2.3 has been installed!
INFO ESPAsyncTCP-esphome@1.2.3 has been installed!
Library Manager: Installing esphome/noise-c @ 0.1.4
INFO Installing esphome/noise-c @ 0.1.4
Unpacking  [####################################]  100%
Library Manager: noise-c@0.1.4 has been installed!
INFO noise-c@0.1.4 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10018.1
INFO Installing esphome/libsodium @ 1.10018.1
Unpacking  [####################################]  100%
Library Manager: libsodium@1.10018.1 has been installed!
INFO libsodium@1.10018.1 has been installed!
HARDWARE: ESP32S2 240MHz, 320KB RAM, 4MB Flash
Error: This board doesn't support arduino framework!
========================== [FAILED] Took 2.44 seconds ==========================
```

ssieb, 2022-09-30:
This isn't an esphome issue.  That's what platformio is saying.  However, you can make it work by using a newer platform:
```yaml
esp32:
  board: esp32-s2-saola-1
  framework:
    type: arduino
    version: 2.0.3
    platform_version: 4.4.0
```

ssieb, 2022-09-30:
Actually, if you have the S2 mini, then this is more correct:
```yaml
esp32:
  board: lolin_s2_mini
  variant: esp32s2
  framework:
    type: arduino
    version: 2.0.3
    platform_version: 4.4.0
```

fran6120, 2022-09-30:
I can't compile it using arduino framework but esp-idf compiles correctly, at least on x86-x64, on armv7 both fails.

```
esp32:
  board: lolin_s2_mini
  variant: ESP32S2
  framework:
    type: esp-idf
    platform_version: 4.4.0
```

Jason2866, 2022-10-08:
It is not a Platformio issue. esphome "just" uses a to old Arduino framework.
Platformio supports the actual Arduino core 2.0.5.