esphome/issues #3628: PID actions `id` not templatable

nagyrobi, 2022-09-30:
### The problem

I have a node with 7 PID climates on it. I'd like to expose to HA the actions `climate.pid.set_control_parameters`, `climate.pid.reset_integral_term` and `climate.pid.autotune` as [user defined services](https://esphome.io/components/api.html#user-defined-services).

Here's my config:
```yaml
api:
  reboot_timeout: 15min
  password: !secret api_password
  encryption:
    key: !secret encryption_key
  services:
    - service: pid_set_control_parameters 
      variables:
        climate_id: string
        kp: float
        ki: float
        kd: float
      then:
        - climate.pid.set_control_parameters:
            id: !lambda 'return climate_id;'
            kp: !lambda 'return kp;'
            ki: !lambda 'return ki;'
            kd: !lambda 'return kd;'
    - service: pid_reset_integral_term 
      variables:
        climate_id: string
      then:
        - climate.pid.reset_integral_term:
            id: !lambda 'return climate_id;'
    - service: pid_autotune 
      variables:
        climate_id: string
        noiseband: float
        positive_output: float
        negative_output: float
      then:
        - climate.pid.autotune:
            id: !lambda 'return climate_id;'
            noiseband: !lambda 'return noiseband;'
            positive_output: !lambda 'return positive_output;'
            negative_output: !lambda 'return negative_output;'
```
In order to avoid creating _for each climate_ 3 user defined services in HA, I'd like to create a total of 3, where the `id` of the climate is also passed as a variable from the service call.

The config above fails with the following error:
```yaml

api: [source /config/esphome/futes-vezerles-radiatorokkal.yaml:31]
  reboot_timeout: 15min
  password: !secret api_password
  encryption: 
    key: !secret encryption_key
  services: 
    - service: pid_set_control_parameters
      variables: 
        climate_id: string
        kp: float
        ki: float
        kd: float
      then: 
        - climate.pid.set_control_parameters: 
            
            This option is not templatable!.
            id: !lambda |-
              return climate_id;
            kp: !lambda |-
              return kp;
            ki: !lambda |-
              return ki;
            kd: !lambda |-
              return kd;
    - service: pid_reset_integral_term
      variables: 
        climate_id: string
      then: 
        - climate.pid.reset_integral_term: 
            id: !lambda |-
              return climate_id;
    - service: pid_autotune
      variables: 
        climate_id: string
        noiseband: float
        positive_output: float
        negative_output: float
      then: 
        - climate.pid.autotune: 
            id: !lambda |-
              return climate_id;
            noiseband: !lambda |-
              return noiseband;
            positive_output: !lambda |-
              return positive_output;
            negative_output: !lambda |-
              return negative_output;
```

### Which version of ESPHome has the issue?

v2022.9.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.9

### What platform are you using?

ESP32

### Board

WirelessTag WT32-ETH01

### Component causing the issue

pid climate

### Example YAML snippet

```yaml
See above
```

### Anything in the logs that might be useful for us?

```txt
See above
```

### Additional information

N/A

nagyrobi, 2022-09-30:
@patrickcollins12

ssieb, 2023-01-31:
This is not possible.  The `id:` is not a string, it's a reference to an internal variable name.  You could replace the action with the lambda equivalent and then you could match the string to the id yourself if you want.

closed 2023-01-31 by ssieb