esphome/issues #3629: Tuya communication stuck at init_state 3

rickyelopez, 2022-10-01:
### The problem

I have a WiFi Temperature and Humidity sensor ([this](https://www.aliexpress.us/item/3256803531872877.html?gatewayAdapt=glo2usa4itemAdapt&_randl_shipto=US) one, the wifi version) with a Tuya MCU and a CB3S. I swapped out the CB3S for an ESP12-e and loaded ESPHome onto it, but the Tuya initialization gets stuck at init_state 3. I turned on verbose logging and uart debugging and it looks like the commands sent by ESPHome make sense, but that the reply from the Tuya MCU doesn't. From some googling, it looks like the response from the MCU is that of the bluetooth mode?

ESPHome sends
`55:AA:00:00:00:00:FF`
to which the MCU replies with
`55:AA:00:00:00:01:01:01`
and it doesn't go any further than that.

From [this](https://developer.tuya.com/en/docs/iot/ble-basic-features?id=Kavijuzgxppvs) page, it looks like this has something to do with the bluetooth part of Tuya, but I really don't know what I'm looking at. Any suggestions?

### Which version of ESPHome has the issue?

2022.6.2

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Board

ESP12-E

### Component causing the issue

Tuya

### Example YAML snippet

```yaml
esphome:
  name: sensor-ambient1

esp8266:
  board: esp12e

logger:
  baud_rate: 0
  level: VERY_VERBOSE

api:
  encryption:
    key: "<key>"

ota:
  password: "<pw>"

wifi:
  ssid: "<ssid>"
  password: "<pw>"
  use_address: <ip>
  domain: <domain>
  fast_connect: on

  ap:
    ssid: "sensor-ambient1"
    password: "<pw>"

captive_portal:

status_led:
  pin:
    number: 2
    inverted: true

uart:
  rx_pin: GPIO3
  tx_pin: GPIO1
  baud_rate: 9600
  debug:

tuya:
```
```

### Anything in the logs that might be useful for us?

```txt
[17:04:14][D][uart_debug:114]: >>> 55:AA:00:00:00:00:FF
[17:04:14][VV][scheduler:195]: Running interval 'wifi' with interval=1000 last_execution=6044653 (now=6045653)
[17:04:14][V][tuya:383]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[] INIT_STATE=3
[17:04:14][D][uart_debug:114]: >>> 55:AA:00:00:00:00:FF
[17:04:14][V][tuya:383]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[] INIT_STATE=3
[17:04:14][D][uart_debug:114]: >>> 55:AA:00:00:00:00:FF
[17:04:15][V][tuya:383]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[] INIT_STATE=3
[17:04:15][D][uart_debug:114]: >>> 55:AA:00:00:00:00:FF
[17:04:15][VV][scheduler:195]: Running interval 'wifi' with interval=1000 last_execution=6045653 (now=6046655)
[17:04:15][V][tuya:383]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[] INIT_STATE=3
[17:04:15][D][uart_debug:114]: >>> 55:AA:00:00:00:00:FF
[17:04:15][E][tuya:409]: Initialization failed at init_state 3
```
```

rickyelopez, 2022-10-09:
Bump. Anybody have any ideas about what's going on, or any suggestions for me to debug further?

ssieb, 2022-10-10:
This appears to be the low-power protocol used for battery powered devices.  It's different, one indication is that it stays in protocol 0 the whole time.  That's the difference between the `55:AA:00:02:00:00:01` that it's sending compared to the `55 aa 03 02 00 00 04` (version 3) that you're expecting.  I started working on this at one point, but didn't get it finished.

leoshusar, 2022-10-14:
I have the same problem.
I have curtain motor powered from mains but it looks like they put the protocol for battery powered devices in it.
I can control it, but since ESPHome receives `55:AA:00:03:00:01:03:06` and doesn't know what to do with it, it's stuck on waiting for init therefore never runs `TuyaCover::setup()` with configuration.

I tried to debug it somehow; I tried to put some `LOGV` commands to `tuya.cpp` message handling and finally in the main loop:
```cpp
void Tuya::loop() {
  while (this->available()) {
    uint8_t c;
    this->read_byte(&c);
    this->handle_char_(c);
    ESP_LOGV(TAG, "hit");
  }
  process_command_queue_();
}
```
and for some reason I didn't get any hit logs after receiving this particular message (message logged via `uart_debug`). But it worked for all other messages like config query and heartbeats, which is strange.

Evka2k, 2023-02-10:
Same issue with BCM500DS-TYW Curtain Motor.
```[V][tuya:383]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[] INIT_STATE=0
[V][tuya:120]: Received Tuya: CMD=0x00 VERSION=0 DATA=[01] INIT_STATE=0
[V][tuya:148]: MCU Heartbeat (0x01)
[V][tuya:383]: Sending Tuya: CMD=0x01 VERSION=0 DATA=[] INIT_STATE=1
[C][tuya.cover:107]: Tuya Cover:
[C][tuya.cover:122]:    Position has datapoint ID 101
[V][tuya:120]: Received Tuya: CMD=0x01 VERSION=0 DATA=[58.53.37.36.42.59.35.51.31.75.4B.4F.36.67.6A.43.31.2E.30.2E.30 (21)] INIT_STATE=1
[V][tuya:383]: Sending Tuya: CMD=0x02 VERSION=0 DATA=[] INIT_STATE=2
[V][tuya:120]: Received Tuya: CMD=0x02 VERSION=0 DATA=[] INIT_STATE=2
[V][tuya:201]: Configured WIFI_STATE periodic send
[D][tuya:456]: Sending WiFi Status
[V][tuya:383]: Sending Tuya: CMD=0x03 VERSION=0 DATA=[03] INIT_STATE=3
[C][tuya:033]: Tuya:
[C][tuya:038]:   Configuration will be reported when setup is complete. Current init_state: 3
[C][tuya:041]:   If no further output is received, confirm that this is a supported Tuya device.
[V][tuya:383]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[] INIT_STATE=3
[V][tuya:120]: Received Tuya: CMD=0x00 VERSION=0 DATA=[01] INIT_STATE=3
[V][tuya:148]: MCU Heartbeat (0x01)
[V][tuya:383]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[] INIT_STATE=3
[V][tuya:120]: Received Tuya: CMD=0x00 VERSION=0 DATA=[01] INIT_STATE=3
[V][tuya:148]: MCU Heartbeat (0x01)
[V][tuya:383]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[] INIT_STATE=3
[V][tuya:120]: Received Tuya: CMD=0x00 VERSION=0 DATA=[01] INIT_STATE=3
[V][tuya:148]: MCU Heartbeat (0x01)
EvkaPC➜  ~ cat /tmp/blind.log | grep "tuya\|uart"
[C][uart.arduino_esp8266:059]: Setting up UART bus...
[V][tuya:383]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[] INIT_STATE=0
[D][uart_debug:158]: >>> "U\xAA\x00\x00\x00\x00\xFF"
[V][tuya:120]: Received Tuya: CMD=0x00 VERSION=0 DATA=[01] INIT_STATE=0
[V][tuya:148]: MCU Heartbeat (0x01)
[V][tuya:383]: Sending Tuya: CMD=0x01 VERSION=0 DATA=[] INIT_STATE=1
[D][uart_debug:158]: <<< "U\xAA\x00\x00\x00\x01\x01\x01"
[C][uart.arduino_esp8266:102]: UART Bus:
[C][uart.arduino_esp8266:103]:   TX Pin: GPIO15
[C][uart.arduino_esp8266:104]:   RX Pin: GPIO13
[C][uart.arduino_esp8266:106]:   RX Buffer Size: 256
[C][uart.arduino_esp8266:108]:   Baud Rate: 9600 baud
[C][uart.arduino_esp8266:109]:   Data Bits: 8
[C][uart.arduino_esp8266:110]:   Parity: NONE
[C][uart.arduino_esp8266:111]:   Stop bits: 1
[C][uart.arduino_esp8266:113]:   Using hardware serial interface.
[C][tuya.cover:107]: Tuya Cover:
[C][tuya.cover:122]:    Position has datapoint ID 101
[D][uart_debug:158]: >>> "U\xAA\x00\x01\x00\x00\x00"
[V][tuya:120]: Received Tuya: CMD=0x01 VERSION=0 DATA=[58.53.37.36.42.59.35.51.31.75.4B.4F.36.67.6A.43.31.2E.30.2E.30 (21)] INIT_STATE=1
[V][tuya:383]: Sending Tuya: CMD=0x02 VERSION=0 DATA=[] INIT_STATE=2
[D][uart_debug:158]: <<< "U\xAA\x00\x01\x00\x15XS76BY5Q1uKO6gjC1.0.0\xC5"
[D][uart_debug:158]: >>> "U\xAA\x00\x02\x00\x00\x01"
[V][tuya:120]: Received Tuya: CMD=0x02 VERSION=0 DATA=[] INIT_STATE=2
[V][tuya:201]: Configured WIFI_STATE periodic send
[D][tuya:456]: Sending WiFi Status
[V][tuya:383]: Sending Tuya: CMD=0x03 VERSION=0 DATA=[03] INIT_STATE=3
[D][uart_debug:158]: <<< "U\xAA\x00\x02\x00\x00\x01"
[C][tuya:033]: Tuya:
[C][tuya:038]:   Configuration will be reported when setup is complete. Current init_state: 3
[C][tuya:041]:   If no further output is received, confirm that this is a supported Tuya device.
[D][uart_debug:158]: >>> "U\xAA\x00\x03\x00\x01\x03\x06"
[V][tuya:383]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[] INIT_STATE=3
[D][uart_debug:158]: >>> "U\xAA\x00\x00\x00\x00\xFF"
[V][tuya:120]: Received Tuya: CMD=0x00 VERSION=0 DATA=[01] INIT_STATE=3
[V][tuya:148]: MCU Heartbeat (0x01)
[D][uart_debug:158]: <<< "U\xAA\x00\x00\x00\x01\x01\x01"
[V][tuya:383]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[] INIT_STATE=3
[D][uart_debug:158]: >>> "U\xAA\x00\x00\x00\x00\xFF"
[V][tuya:120]: Received Tuya: CMD=0x00 VERSION=0 DATA=[01] INIT_STATE=3
[V][tuya:148]: MCU Heartbeat (0x01)
[D][uart_debug:158]: <<< "U\xAA\x00\x00\x00\x01\x01\x01"
```

github-k8n, 2023-07-25:
Similar issue with BMM700D-TY01 curtain motor.

Previously I was connecting an esp in parallel to the existing chip (WBR1-IPEX)
There, it seems the MCU was initialized correctly (afaik), at least the esp received and showed the datapoints correctly.
(101 for position, 102 for control), even if the stop, up, down values are mixed up..

Once I removed the original chip and only kept the ESP however I do not receive anything from the MCU anymore, however I can still control it by sending values to the datapoints...
Debugging shows me a similar issue to the issue above:

[15:48:59][D][uart_debug:114]: >>> 55:AA:00:00:00:00:FF
[15:48:59][D][uart_debug:114]: <<< 55:AA:00:00:00:01:01:01

"Configuration will be reported when setup is complete. Current init_state: 3"

Any chance that this can get addressed at some point?

rickyelopez, 2024-11-08:
Ok so I've now largely figured out this protocol, thanks mostly to [this](https://github.com/brandond/esphome-tuya_pir/blob/master/sb1_uart.h) code from @brandond and [this](https://developer.tuya.com/en/docs/iot/tuyacloudlowpoweruniversalserialaccessprotocol?id=K95afs9h4tjjh#title-0-Overview) page from Tuya (which I linked in my original post, but somehow I didn't connect the dots back then).

I've refactored that code a bit as a super WIP POC, and posted this [gist](https://gist.github.com/rickyelopez/d78899f8569cb5aeb949013de1c78b5a) if anyone wants to take a look. All this really accomplishes is printing out the DP data, which was the primary thing I wanted to get working here.

There's a limit to how much can be done in this way (including a header file in the device's `yaml`), and I'd love to get this work integrated into esphome, either in the existing Tuya component or in a new component (`tuya_lowpower`?), but there are a few things I haven't been able to quickly find answers to:

1. Adding support for this low-power mode to the existing Tuya component seems like it would be a pretty significant refactor, which is not very appealing to me
1. Based on 1, it would be great if this could just go into a new component, but it would suck to have to duplicate all of the tuya sub-components (`sensor`, `cover`, etc.) which would all effectively be identical as far as I can tell. I don't see any examples of components having either-or dependencies, though, so I don't know if there's a way around this? Ideally, all of the sub-components would depend on either `tuya` or `tuya_lowpower` having been configured
1. This low-power mode requires the device to be able to check in with a server for any cached instructions it might have missed while offline. In the Tuya firmware, the server is Tuya's cloud. In this application, it would probably be either Home Assistant or mqtt? I haven't been able to find any examples of this functionality, though.

If anybody has any ideas for any of the above, I'd love to hear them!

github-k8n, 2025-04-17:
@szupi-ipuzs , @rickyelopez  has there been some progress regarding this?

rickyelopez, 2025-04-18:
No, I never came to a solution that I liked and I sort of lost interest. I will probably take it up again at some point but I don't have anything planned right now, sorry.