esphome/issues #3632: Unable to reuse ESP8266 D1 mini device (likely to affect other types)

Bekkie, 2022-10-01:
### The problem

After removing a ESPHome device from the ESPHome dashboard, used for some experiments, I wanted to reuse the same physical ESP8266 for a different purpose. 
After creating and installing the new functionality in the reused ESP8266, it did not appear as "new device detected" in Home Assistant Notification tab. It was accessible via the ESPHome Dashboard and web portal, but no entities appeared in HA. Also rebooting HA, ESPHome and the host did not solve the issue. After a while the notification windows reports, for the removed device: 

"_Integration requires reconfiguration
At least one of your integrations requires reconfiguration to continue functioning. [Check it out](https://waalre.duckdns.org:8123/config/integrations)._"

Pressing the "reconfigure button" and apply the new API key, does not help either

Using a "fresh" ESP8266 with the same code did appear in HA. So most likely not all data is removed (like the device code "Espressif xx:xx:xx") from the HA database. 

### Which version of ESPHome has the issue?

2022.9.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.9.7

### What platform are you using?

ESP8266

### Board

ESP8266 D1 mini

ssieb, 2022-10-01:
You have to remove the integration entry for the old device and then add the new one.  It may or may not auto-detect.

Bekkie, 2022-10-06:
I did remove the integration entry and now the new device appeared!

closed 2022-10-06 by Bekkie