esphome/issues #3634: SML Component: Publishing every value instead only if value changed

Viermusketiere, 2022-10-02:
### The problem

When using the SML Component to monitor a smart meter every newly received packet of data (Obis-Data) is published to Home Assistant. Expected behaviour would be that the value is only published when the newly read value differs from the one before (eg. as with a temperature sensor).

The occuring behaviour results in a huge pile of data written to the HA database, as most smart meters publish their data every one to two seconds.

### Which version of ESPHome has the issue?

2022.9.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.7.3

### What platform are you using?

ESP8266

### Board

NodeMCU

### Component causing the issue

sml

### Example YAML snippet

```yaml
uart:
  id: uart_bus
  rx_pin: GPIO3
  baud_rate: 9600
  data_bits: 8
  parity: NONE
  stop_bits: 1
#  debug:
#    after:
#      timeout: 100ms

sml:
  id: mysml
  uart_id: uart_bus

sensor:
  - platform: sml
    name: "Zählerstand HT + NT"
    sml_id: mysml
    obis_code: "1-0:1.8.0"
    unit_of_measurement: kWh
    accuracy_decimals: 1
    device_class: energy
    state_class: total_increasing
    force_update: False
  - platform: sml
    name: "Zählerstand HT"
    sml_id: mysml
    obis_code: "1-0:1.8.2"
    unit_of_measurement: kWh
    accuracy_decimals: 1
    device_class: energy
    state_class: total_increasing
    force_update: False
  - platform: sml
    name: "Zählerstand NT"
    sml_id: mysml
    obis_code: "1-0:1.8.1"
    unit_of_measurement: kWh
    accuracy_decimals: 1
    device_class: energy
    state_class: total_increasing
    force_update: False
  - platform: sml
    name: "Zählerstand Einspeisung HT + NT"
    sml_id: mysml
    obis_code: "1-0:2.8.0"
    unit_of_measurement: kWh
    accuracy_decimals: 1
    device_class: energy
    state_class: total_increasing
    force_update: False
  - platform: sml
    name: "Zählerstand Einspeisung HT"
    sml_id: mysml
    obis_code: "1-0:2.8.2"
    unit_of_measurement: kWh
    accuracy_decimals: 1
    device_class: energy
    state_class: total_increasing
    force_update: False
  - platform: sml
    name: "Zählerstand Einspeisung NT"
    sml_id: mysml
    obis_code: "1-0:2.8.1"
    unit_of_measurement: kWh
    accuracy_decimals: 1
    device_class: energy
    state_class: total_increasing
    force_update: False
  

text_sensor:
  - platform: sml
    name: "Zähler Hersteller"
    sml_id: mysml
    obis_code: "129-129:199.130.3"
    format: text
  - platform: sml
    name: "Zähler Public Key"
    sml_id: mysml
    obis_code: "129-129:199.130.5"
    format: text
```

### Anything in the logs that might be useful for us?

```txt
[22:41:50][D][sensor:126]: 'Zählerstand HT + NT': Sending state 43359.00000 kWh with 1 decimals of accuracy
[22:41:50][D][sensor:126]: 'Zählerstand Einspeisung HT + NT': Sending state 12.00000 kWh with 1 decimals of accuracy
[22:41:50][D][sensor:126]: 'Zählerstand NT': Sending state 39638.00000 kWh with 1 decimals of accuracy
[22:41:50][D][sensor:126]: 'Zählerstand Einspeisung NT': Sending state 11.00000 kWh with 1 decimals of accuracy
[22:41:50][D][sensor:126]: 'Zählerstand HT': Sending state 3721.00000 kWh with 1 decimals of accuracy
[22:41:51][D][sensor:126]: 'Zählerstand Einspeisung HT': Sending state 1.00000 kWh with 1 decimals of accuracy
[22:41:51][D][text_sensor:067]: 'Zähler Public Key': Sending state '__PRIVATE_CONTENT__'
[22:41:51][D][sml:065]: OBIS info:
[22:41:51][D][sml:071]:   (0901454d4800006ce696) 129-129:199.130.3 [__PRIVATE_CONTENT__]
[22:41:51][D][sml:071]:   (0901454d4800006ce696) 1-0:0.0.9 [0x0901454d4800006ce696]
[22:41:51][D][sml:071]:   (0901454d4800006ce696) 1-0:1.8.0 [0x000000a95f]
[22:41:51][D][sml:071]:   (0901454d4800006ce696) 1-0:2.8.0 [0x000000000c]
[22:41:51][D][sml:071]:   (0901454d4800006ce696) 1-0:1.8.1 [0x0000009ad6]
[22:41:51][D][sml:071]:   (0901454d4800006ce696) 1-0:2.8.1 [0x000000000b]
[22:41:51][D][sml:071]:   (0901454d4800006ce696) 1-0:1.8.2 [0x0000000e89]
[22:41:51][D][sml:071]:   (0901454d4800006ce696) 1-0:2.8.2 [0x0000000001]
[22:41:52][D][sml:071]:   (0901454d4800006ce696) 129-129:199.130.5 [__PRIVATE_CONTENT__]
[22:41:52][D][api:102]: Accepted 192.168.178.10
[22:41:52][D][api.connection:861]: Home Assistant 2022.7.3 (192.168.178.10): Connected successfully
[22:41:52][W][api.connection:080]: Home Assistant 2022.7.3 (192.168.178.10): Connection closed
[22:41:53][D][api:102]: Accepted 192.168.178.10
[22:41:53][D][api.connection:861]: Home Assistant 2022.7.3 (192.168.178.10): Connected successfully
[22:41:53][D][text_sensor:067]: 'Zähler Hersteller': Sending state 'EMH'
[22:41:53][D][sensor:126]: 'Zählerstand HT + NT': Sending state 43359.00000 kWh with 1 decimals of accuracy
[22:41:53][D][sensor:126]: 'Zählerstand Einspeisung HT + NT': Sending state 12.00000 kWh with 1 decimals of accuracy
[22:41:54][D][sensor:126]: 'Zählerstand NT': Sending state 39638.00000 kWh with 1 decimals of accuracy
[22:41:54][D][sensor:126]: 'Zählerstand Einspeisung NT': Sending state 11.00000 kWh with 1 decimals of accuracy
[22:41:54][D][sensor:126]: 'Zählerstand HT': Sending state 3721.00000 kWh with 1 decimals of accuracy
[22:41:54][D][sensor:126]: 'Zählerstand Einspeisung HT': Sending state 1.00000 kWh with 1 decimals of accuracy
[22:41:54][D][text_sensor:067]: 'Zähler Public Key': Sending state '__PRIVATE_CONTENT__'
[22:41:54][D][sml:065]: OBIS info:
[22:41:54][D][sml:071]:   (0901454d4800006ce696) 129-129:199.130.3 [__PRIVATE_CONTENT__]
[22:41:54][D][sml:071]:   (0901454d4800006ce696) 1-0:0.0.9 [0x0901454d4800006ce696]
[22:41:54][D][sml:071]:   (0901454d4800006ce696) 1-0:1.8.0 [0x000000a95f]
[22:41:54][D][sml:071]:   (0901454d4800006ce696) 1-0:2.8.0 [0x000000000c]
[22:41:54][D][sml:071]:   (0901454d4800006ce696) 1-0:1.8.1 [0x0000009ad6]
[22:41:55][D][sml:071]:   (0901454d4800006ce696) 1-0:2.8.1 [0x000000000b]
[22:41:55][D][sml:071]:   (0901454d4800006ce696) 1-0:1.8.2 [0x0000000e89]
[22:41:55][D][sml:071]:   (0901454d4800006ce696) 1-0:2.8.2 [0x0000000001]
[22:41:55][D][sml:071]:   (0901454d4800006ce696) 129-129:199.130.5 [__PRIVATE_CONTENT__]
```

### Additional information

[image]

If help debugging the issue while using the component in combination with a smart meter is needed, let me know.

jbbjarnason, 2022-10-06:
@Viermusketiere I think I can fix this, probably this weekend. Would you be able to test it? 
I don't got the needed hardware for it.

Viermusketiere, 2022-10-07:
@jbbjarnason Yeah, that won't be a problem. I'd have to check how to get another branch loaded in Home Assistant or manually compiled and flashed to the ESP.
But that should be easy to figure out.

Viermusketiere, 2022-10-17:
@jbbjarnason Did you manage to do some work on the issue?
I figured out how I could test it, when provided with the branch/needed files for it.

jbbjarnason, 2022-10-18:
I have looked at this and I think the esphome `Sensor` abstraction should be able to take care of this. Have you tried to use the `delta` filter, https://esphome.io/components/sensor/index.html#delta ?

Viermusketiere, 2022-10-24:
I tried implementing a delta filter, but even with the filter added the SML component sens data on every received data packet from the meter. Interesting is, that with the filter added the lines in the log, where it says 'Sending state' are missing (Code box below), but the data is received in Home Assistant.
```
[22:41:53][D][sensor:126]: 'Zählerstand HT + NT': Sending state 43359.00000 kWh with 1 decimals of accuracy
[22:41:53][D][sensor:126]: 'Zählerstand Einspeisung HT + NT': Sending state 12.00000 kWh with 1 decimals of accuracy
[22:41:54][D][sensor:126]: 'Zählerstand NT': Sending state 39638.00000 kWh with 1 decimals of accuracy
[22:41:54][D][sensor:126]: 'Zählerstand Einspeisung NT': Sending state 11.00000 kWh with 1 decimals of accuracy
[22:41:54][D][sensor:126]: 'Zählerstand HT': Sending state 3721.00000 kWh with 1 decimals of accuracy
[22:41:54][D][sensor:126]: 'Zählerstand Einspeisung HT': Sending state 1.00000 kWh with 1 decimals of accuracy
```

Log output now looks like this:
```
11:09:05][D][api:102]: Accepted 192.168.178.10
[11:09:05][D][api.connection:917]: Home Assistant 2022.7.3 (192.168.178.10): Connected successfully
[11:09:06][D][text_sensor:067]: 'Zähler Hersteller': Sending state 'EMH'
[11:09:06][D][text_sensor:067]: 'Zähler Public Key': Sending state '\xc9f\xf8\xb7\xcet\x9bЌ\xbe\x82\x94dG~3l\xd3\xdapf|\x92\x91\xc2\xc5O P\xd1%\xbbtr\xf4C\xfd\x89\xba\xc7\xec'
[11:09:06][D][sml:065]: OBIS info:
[11:09:06][D][sml:071]:   (0901454d4800006ce696) 129-129:199.130.3 [0x454d48]
[11:09:06][D][sml:071]:   (0901454d4800006ce696) 1-0:0.0.9 [0x0901454d4800006ce696]
[11:09:06][D][sml:071]:   (0901454d4800006ce696) 1-0:1.8.0 [0x000000adb5]
[11:09:06][D][sml:071]:   (0901454d4800006ce696) 1-0:2.8.0 [0x0000000010]
[11:09:06][D][sml:071]:   (0901454d4800006ce696) 1-0:1.8.1 [0x0000009e97]
[11:09:06][D][sml:071]:   (0901454d4800006ce696) 1-0:2.8.1 [0x000000000f]
[11:09:06][D][sml:071]:   (0901454d4800006ce696) 1-0:1.8.2 [0x0000000f1d]
[11:09:06][D][sml:071]:   (0901454d4800006ce696) 1-0:2.8.2 [0x0000000001]
[11:09:07][D][sml:071]:   (0901454d4800006ce696) 129-129:199.130.5 [0xc966f811b7ce749bd08cbe829464477e336cd3da700f667c9291c20407c54f06201350d125bb027472f443fd89bac7ec]
[11:09:07][D][api:102]: Accepted 192.168.178.10
[11:09:09][W][api.connection:081]: Home Assistant 2022.7.3 (192.168.178.10): Connection reset
[11:09:09][D][api.connection:917]: Home Assistant 2022.7.3 (192.168.178.10): Connected successfully
[11:09:09][W][api.connection:083]: Home Assistant 2022.7.3 (192.168.178.10): Connection closed
[11:09:09][D][api:102]: Accepted 192.168.178.10
[11:09:09][D][text_sensor:067]: 'Zähler Hersteller': Sending state 'EMH'
[11:09:09][D][text_sensor:067]: 'Zähler Public Key': Sending state '\xc9f\xf8\xb7\xcet\x9bЌ\xbe\x82\x94dG~3l\xd3\xdapf|\x92\x91\xc2\xc5O P\xd1%\xbbtr\xf4C\xfd\x89\xba\xc7\xec'
[11:09:09][D][sml:065]: OBIS info:
[11:09:09][D][sml:071]:   (0901454d4800006ce696) 129-129:199.130.3 [0x454d48]
[11:09:09][D][sml:071]:   (0901454d4800006ce696) 1-0:0.0.9 [0x0901454d4800006ce696]
[11:09:09][D][sml:071]:   (0901454d4800006ce696) 1-0:1.8.0 [0x000000adb5]
[11:09:09][D][sml:071]:   (0901454d4800006ce696) 1-0:2.8.0 [0x0000000010]
[11:09:09][D][sml:071]:   (0901454d4800006ce696) 1-0:1.8.1 [0x0000009e97]
[11:09:09][D][sml:071]:   (0901454d4800006ce696) 1-0:2.8.1 [0x000000000f]
[11:09:09][D][sml:071]:   (0901454d4800006ce696) 1-0:1.8.2 [0x0000000f1d]
[11:09:09][D][sml:071]:   (0901454d4800006ce696) 1-0:2.8.2 [0x0000000001]
[11:09:09][D][sml:071]:   (0901454d4800006ce696) 129-129:199.130.5 [0xc966f811b7ce749bd08cbe829464477e336cd3da700f667c9291c20407c54f06201350d125bb027472f443fd89bac7ec]
[11:09:09][D][api.connection:917]: Home Assistant 2022.7.3 (192.168.178.10): Connected successfully
[11:09:10][W][api.connection:083]: Home Assistant 2022.7.3 (192.168.178.10): Connection closed
```

The data received in Home Assistant looks like this (Diplayed is a timespan of five minutes:
[image]

I think there may be a problem with the computation of the values read by the SML component and processed by the Generic Sensor Class.

ssieb, 2022-10-25:
Only sending values when they change will cause strange looking graphs, but if that's what you want, the delta filter should work.  What is your config?

ssieb, 2022-10-25:
Can you paste some logs?  The filter will affect when the data is sent to HA, but I don't know how the classes will affect how HA displays it.

Viermusketiere, 2022-11-17:
@ssieb Sorry for the long delay.
I have checked with the event listener and Home Assistant receives a state changed event for every attribute of the SML component every second.
```
{
    "event_type": "state_changed",
    "data": {
        "entity_id": "sensor.zahlerstand_einspeisung_nt",
        "old_state": {
            "entity_id": "sensor.zahlerstand_einspeisung_nt",
            "state": "unavailable",
            "attributes": {
                "unit_of_measurement": "kWh",
                "device_class": "energy",
                "friendly_name": "Zählerstand Einspeisung NT"
            },
            "last_changed": "2022-11-17T15:16:20.408361+00:00",
            "last_updated": "2022-11-17T15:16:20.408361+00:00",
            "context": {
                "id": "01GJ32K2SR22T89J6JRATSBRTC",
                "parent_id": null,
                "user_id": null
            }
        },
        "new_state": {
            "entity_id": "sensor.zahlerstand_einspeisung_nt",
            "state": "23.0",
            "attributes": {
                "unit_of_measurement": "kWh",
                "device_class": "energy",
                "friendly_name": "Zählerstand Einspeisung NT"
            },
            "last_changed": "2022-11-17T15:16:21.246874+00:00",
            "last_updated": "2022-11-17T15:16:21.246874+00:00",
            "context": {
                "id": "01GJ32K3KYP7JBD25G0XZXRK41",
                "parent_id": null,
                "user_id": null
            }
        }
    },
    "origin": "LOCAL",
    "time_fired": "2022-11-17T15:16:21.246874+00:00",
    "context": {
        "id": "01GJ32K3KYP7JBD25G0XZXRK41",
        "parent_id": null,
        "user_id": null
    }
}
Event 80 fired 4:16 PM:
{
    "event_type": "state_changed",
    "data": {
        "entity_id": "sensor.zahlerstand_einspeisung_ht",
        "old_state": {
            "entity_id": "sensor.zahlerstand_einspeisung_ht",
            "state": "unavailable",
            "attributes": {
                "unit_of_measurement": "kWh",
                "device_class": "energy",
                "friendly_name": "Zählerstand Einspeisung HT"
            },
            "last_changed": "2022-11-17T15:16:20.408106+00:00",
            "last_updated": "2022-11-17T15:16:20.408106+00:00",
            "context": {
                "id": "01GJ32K2SRBQY7RNNMDGZS23X9",
                "parent_id": null,
                "user_id": null
            }
        },
        "new_state": {
[... 84 lines cut ...]
[22:41:53][D][api:102]: Accepted 192.168.178.10
[22:41:53][D][api.connection:861]: Home Assistant 2022.7.3 (192.168.178.10): Connected successfully
[22:41:53][D][text_sensor:067]: 'Zähler Hersteller': Sending state 'EMH'
[22:41:53][D][sensor:126]: 'Zählerstand HT + NT': Sending state 43359.00000 kWh with 1 decimals of accuracy
[22:41:53][D][sensor:126]: 'Zählerstand Einspeisung HT + NT': Sending state 12.00000 kWh with 1 decimals of accuracy
[22:41:54][D][sensor:126]: 'Zählerstand NT': Sending state 39638.00000 kWh with 1 decimals of accuracy
[22:41:54][D][sensor:126]: 'Zählerstand Einspeisung NT': Sending state 11.00000 kWh with 1 decimals of accuracy
[22:41:54][D][sensor:126]: 'Zählerstand HT': Sending state 3721.00000 kWh with 1 decimals of accuracy
[22:41:54][D][sensor:126]: 'Zählerstand Einspeisung HT': Sending state 1.00000 kWh with 1 decimals of accuracy
[22:41:54][D][text_sensor:067]: 'Zähler Public Key': Sending state '__PRIVATE_CONTENT__'
[22:41:54][D][sml:065]: OBIS info:
[22:41:54][D][sml:071]:   (0901454d4800006ce696) 129-129:199.130.3 [__PRIVATE_CONTENT__]
[22:41:54][D][sml:071]:   (0901454d4800006ce696) 1-0:0.0.9 [0x0901454d4800006ce696]
[22:41:54][D][sml:071]:   (0901454d4800006ce696) 1-0:1.8.0 [0x000000a95f]
[22:41:54][D][sml:071]:   (0901454d4800006ce696) 1-0:2.8.0 [0x000000000c]
[22:41:54][D][sml:071]:   (0901454d4800006ce696) 1-0:1.8.1 [0x0000009ad6]
[22:41:55][D][sml:071]:   (0901454d4800006ce696) 1-0:2.8.1 [0x000000000b]
[22:41:55][D][sml:071]:   (0901454d4800006ce696) 1-0:1.8.2 [0x0000000e89]
[22:41:55][D][sml:071]:   (0901454d4800006ce696) 1-0:2.8.2 [0x0000000001]
[22:41:55][D][sml:071]:   (0901454d4800006ce696) 129-129:199.130.5 [__PRIVATE_CONTENT__]
```

So I think there may be some issues with the SML component and the processing of delta filters.

Let me know if you need anymore logs.

Viermusketiere, 2022-11-18:
I have figured out the root cause of the unexpectedly reported values.
If the ESPHome configuration is set to logger level DEBUGevery value detected by the SML component is transferd to Home Asisstant. This could be solved by changing the logger level to anything higher than DEBUG eg. INFO.

This is clearly not expected behaviour because the default logger level is DEBUG.
I am not as far into the source code that i could propose a PR top solve that issue.

If further information are needed, let me know.