esphome/issues #3637: HomeAssistant sensor data not available in ESPHome

Nedevski, 2022-10-03:
### The problem

Hi, I followed the [`Time & Temperature on OLED Display tutorial`](https://esphome.io/cookbook/display_time_temp_oled.html) and I am getting very odd results.

The time component is a bit slow to update and sometimes it requires a power-cycle to sync to HA, but at least works.
However the temperature values never work and never show up. This is weird since both the time and the sensor use the HA connection.

I tried debugging using the `ESP_LOGI` method but I get a `NaN`.
I added a sensor that reads the analog pin of the ESP and that works and the data is displayed on the screen.

As suggested by someone I removed everything and tried only having a single sensor and an `interval` to debug its value - still no luck.

I have created a thread on the HA forum where we tried to debug the issue but no luck.
https://community.home-assistant.io/t/sensor-data-from-ha-is-not-received-in-esphome/466466/13

### Which version of ESPHome has the issue?

2022.9.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.9.4

### What platform are you using?

ESP32

### Board

ESP-WROOM-32 dev board

### Component causing the issue

sensor

### Example YAML snippet

```yaml
esphome:
  name: esp32-screen-testing

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
#   encryption:
#     key: "redacted"

ota:
  password: "redacted"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32-Screen-Testing"
    password: "redacted"

captive_portal:

# time:
#   - platform: homeassistant
#     id: esptime

sensor:
  - platform: homeassistant
    id: room_temp
    entity_id: sensor.temperature_livingroom
    
interval:
  - interval: 1s
    then: 
      - lambda: |
            ESP_LOGI("log", "%f", id(room_temp).state);
```

### Anything in the logs that might be useful for us?

```txt
![image](https://user-images.githubusercontent.com/7324633/193503394-86207c8e-c5e5-4547-9f83-f4034f8c9ffd.png)
```

alexdetsch, 2022-10-05:
Did you configured the integration in HA? Settings --> devices/services --> integrations tab. There you should already see a configurable ESPHome instance of your ESP32. Click on configure and enter the encryption key from the yaml in ESPHome of your specific ESP32. Look under api:encryption:key.

Nedevski, 2022-10-05:
I thought it was added automatically when creating a new ESPHome device.. I added that and it worked, yes.

closed 2022-10-05 by Nedevski