esphome/issues #3640: Wifi issues

pittbull, 2022-10-03:
I am flashing my m5stack from your webpage, and it works perfectly.

But if I take the yaml from your git and install it from my local esphome it fails to connect to the assigned wifi. I have 2 networks at home and even if I tell it to connect it to one, it connects to the other. (UDM is running the wifi).

Any insights?

nagyrobi, 2022-10-03:
How do you configure those networks in the yaml?

How is your wifi system set up?

pittbull, 2022-10-03:
I pulled the .yaml from here: https://github.com/esphome/bluetooth-proxies/

and changed the wifi part:

```
substitutions:
  name: atom-bluetooth-proxy
  wifi_ap_password: "" <--- removed this

esphome:
  name: ${name}
  name_add_mac_suffix: true
  project:
    name: esphome.bluetooth-proxy
    version: "1.0"

esp32:
  board: m5stack-atom
  framework:
    type: arduino

wifi:
  ssid: !secret wifi_ssid <----- added this
  password: !secret wifi_password <---- added this

api:
logger:
ota:
improv_serial:
captive_portal:

dashboard_import:
  package_import_url: github://esphome/bluetooth-proxies/m5stack-atom-lite.yaml@main

esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

bluetooth_proxy:

button:
- platform: safe_mode
  name: Safe Mode Boot
  entity_category: diagnostic

```
The device does connect, but it does not connect to my secondary wifi (for iot devices). I tried multiple times, and even with two other fresh devices, but they all connect to my main wifi, and not the one specified in the yaml.

I have also tried specifying the name and password in the yaml, but same results.

nagyrobi, 2022-10-03:
Do you use the same SSID and password for your main and iot devices?

pittbull, 2022-10-03:
Noted on the dashboard - removed it.

I have separate ssid's and passwords. I am retrying now and actually was successful, but when I updated the config to switch to my main wifi it now stays behind on my iot network.

I also changed the hostname in the config before reinstalling it and it seems the devices is not fully reset upon reload a new config, because the both the original wifi and hostname are still present in the logs after reflashing.

Do I actually have to reset the board before updating the config?

pittbull, 2022-10-03:
[image]

```
wifi:
  ssid: !secret wifi_ssid2
  password: !secret wifi_ssid2_password
```

Reflashed the config 10 times atleast. This wifi config points to my second iot network (different name and password). As per screen shot the device still connects to my main wifi: modern family.

bkaufx, 2022-10-05:
Do you see the line `Loaded saved wifi settings:` in logs?  If so, that would indicate that the device is loading saved credentials from flash instead of using the yaml configured credentials.

You might also check the main.cpp file in esphome/.esphome/build/atom-bluetooth-proxy/src folder, and see what credentials are being assigned there.  Search for ssid.

pittbull, 2022-10-05:
Will verify, but how can I overwritw?

nagyrobi, 2022-10-05:
https://esphome.io/components/switch/factory_reset.html

bkaufx, 2022-10-05:
The saved wifi credentials should be overwritten every time you compile a new firmware bin file.  If you are installing through the ESPHome dashboard I believe this will happen no matter what.  The credentials won't be overwritten if you just reflash the same bin file multiple times without recompiling.  If you are recompiling and the saved wifi credentials are still being loaded from before, that is really weird.

For the yaml-configured credentials, just make sure that the main.cpp file shows the values you are expecting based on the substitutions.  If the main.cpp file shows different credentials than the ones the device ends up connecting to, that is really weird too.

pittbull, 2022-10-06:
Thank you for a detailed feedback. I will retry with a few devices later and cross-check the main.cpp file.