esphome/issues #3641: platform remote_receiver - support of multiple RAW codes

erd82, 2022-10-03:
### The problem

Hello,

I am having the same problem as discussed before here: https://github.com/esphome/issues/issues/1239

When I press the same button on any IR remote several times, the IR receiver captures different raw values each time.
To catch all the different raw values and trigger the same switch in Home Assistant each time, I could simply use the following code example:

```
binary_sensor:
  - platform: remote_receiver
    name: "TV On/Off Button"
    raw:
      code: [869, -914, 869, -913, 1757, -1802, 1757, -914, 867, -915, 866, -915, 867, -915, 867, -1804, 867, -914, 1757, -914, 867]
    filters:
      - delayed_off: 200ms
  - platform: remote_receiver
    name: "TV On/Off Button"
    raw:
      code: [870, -915, 1760, -913, 869, -1805, 1783, -891, 893, -890, 893, -890, 894, -890, 869, -1803, 868, -916, 1759, -914, 894]
    filters:
      - delayed_off: 200ms
  - platform: remote_receiver
    name: "TV On/Off Button"
    raw:
      code: [912, -890, 1785, -889, 893, -1779, 1785, -890, 893, -890, 894, -890, 893, -890, 893, -1783, 890, -890, 1784, -889, 894]
    filters:
      - delayed_off: 200ms
```
This always toggles the same switch on and off in Home Assistant independently from the received raw code.

It would be great if Esphome could support multiple raw values in one binary sensor - something like:
```
binary_sensor:
  - platform: remote_receiver
    name: "TV On/Off Button"
    raw:
      code: [869, -914, 869, -913, 1757, -1802, 1757, -914, 867, -915, 866, -915, 867, -915, 867, -1804, 867, -914, 1757, -914, 867]
      code: [870, -915, 1760, -913, 869, -1805, 1783, -891, 893, -890, 893, -890, 894, -890, 869, -1803, 868, -916, 1759, -914, 894]
      code: [912, -890, 1785, -889, 893, -1779, 1785, -890, 893, -890, 894, -890, 893, -890, 893, -1783, 890, -890, 1784, -889, 894]
    filters:
      - delayed_off: 200ms
```

### Which version of ESPHome has the issue?

2022.3.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.9.7

### What platform are you using?

ESP8266

### Board

esp01_1m

### Component causing the issue

remote_receiver

### Example YAML snippet

```yaml
remote_receiver:
  pin:
    number: GPIO02
    inverted: true
    mode: INPUT_PULLUP
  dump: all
  idle: 25ms

binary_sensor:
  - platform: remote_receiver
    name: "TV On/Off Button"
    raw:
      code: [869, -914, 869, -913, 1757, -1802, 1757, -914, 867, -915, 866, -915, 867, -915, 867, -1804, 867, -914, 1757, -914, 867]
    filters:
      - delayed_off: 200ms
  - platform: remote_receiver
    name: "TV On/Off Button"
    raw:
      code: [870, -915, 1760, -913, 869, -1805, 1783, -891, 893, -890, 893, -890, 894, -890, 869, -1803, 868, -916, 1759, -914, 894]
    filters:
      - delayed_off: 200ms
  - platform: remote_receiver
    name: "TV On/Off Button"
    raw:
      code: [912, -890, 1785, -889, 893, -1779, 1785, -890, 893, -890, 894, -890, 893, -890, 893, -1783, 890, -890, 1784, -889, 894]
    filters:
      - delayed_off: 200ms
```

erd82, 2023-02-01:
Is this something that will be improved / supported in the future?

ssieb, 2023-02-02:
Probably not, `raw` is pretty much last resort and is generally not very useful.  You can't get any protocol to decode that signal?  What is the remote?  If you could get a `pronto` decode, that could help you, since the matching is allowed to be slightly fuzzy.
Also, that decode seems quite short.  Are you sure you're getting it all?

fantasmisiciliani, 2023-02-05:
You're receiving consistent data, binary code is always the same: negative values equal to 0 (low signal), positive ones to 1 (high signal) and the absolute value is the signal lenght (duration in microseconds).
Your problem is about the high deviation in signal length. You can solve using only the second or the third raw code and set an higher tolerance in Remote Receiver than the default one (25%), for example 60%.
Please refer to the [Remote Receiver Component documentation](https://esphome.io/components/remote_receiver.html) for more info about settings.
Best approach when using raw codes is dumping several transmissions, checking the consistency, getting an average for every signal lenght and finally setting a tolerance that can cover all the higher and the lower values.