esphome/issues #3644: Waking up VL53L1X from HW standby with XSHUT

OkhammahkO, 2022-10-04:
### The problem

I seem to be able to put my single VL53L0X sensor into hardware standby by using the  XSHUT / enable_pin.
I cannot seem to bring the VL53L0X  back out of  HW standby though.

The issue seems similar to this one, and could possibly be related?
https://forum.arduino.cc/t/waking-up-vl53l1x-from-hw-standby-with-xshut/966021/4 

### Which version of ESPHome has the issue?

2022.9.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.9.7

### What platform are you using?

ESP32

### Board

esp32doit-devkit-v1

### Component causing the issue

VL53L1X

### Example YAML snippet

```yaml
- platform: vl53l0x
    id: tof
    name: "ToF Sensor"
    address: 0x29 #Have also tried other adresses
    update_interval: 2s  #Have also tried other intervals
    enable_pin: GPIO17 #Have also tried other pins
    accuracy_decimals: 1 
    filters:
    #Convert to cm
      - multiply: 100

# Example configuration entry
switch:
  - platform: gpio
    pin:
      number: 17
      #mode: INPUT #Have also tried a bunch of other modes 
    name: "ToF Switch"
    id: tof_switch
    restore_mode: ALWAYS_ON  #Have also tried a bunch of other restore modes
```

### Anything in the logs that might be useful for us?

```txt
$$$ Sensor is working ok here $$$$$$$$$$$
[19:10:00][D][sensor:127]: 'ToF Sensor': Sending state 53.20000 m with 1 decimals of accuracy
[19:10:01][D][switch:017]: 'ToF Switch' Turning OFF.
[19:10:01][D][switch:037]: 'ToF Switch': Sending state OFF

$$$ I turned the switch off, and the sensor now sends nan, which I guess could be expected.$$$$$$$$$$$

[19:10:04][D][sensor:127]: 'ToF Sensor': Sending state nan m with 1 decimals of accuracy
[19:10:04][W][vl53l0x:266]: ToF Sensor - update called before prior reading complete - initiated:0 waiting_for_interrupt:1
[19:10:06][D][sensor:127]: 'ToF Sensor': Sending state nan m with 1 decimals of accuracy
[19:10:06][W][vl53l0x:266]: ToF Sensor - update called before prior reading complete - initiated:0 waiting_for_interrupt:1
[19:10:08][D][sensor:127]: 'ToF Sensor': Sending state nan m with 1 decimals of accuracy
[19:10:08][W][vl53l0x:266]: ToF Sensor - update called before prior reading complete - initiated:0 waiting_for_interrupt:1
[19:10:08][D][switch:013]: 'ToF Switch' Turning ON.
[19:10:08][D][switch:037]: 'ToF Switch': Sending state ON

$$$ I turned the switch back on, and the sensor sends still nan, I thought values would appear again.$$$$$$$$$$$

[19:10:10][D][sensor:127]: 'ToF Sensor': Sending state nan m with 1 decimals of accuracy
[19:10:10][W][vl53l0x:266]: ToF Sensor - update called before prior reading complete - initiated:0 waiting_for_interrupt:1
[19:10:12][D][sensor:127]: 'ToF Sensor': Sending state nan m with 1 decimals of accuracy
[19:10:12][W][vl53l0x:266]: ToF Sensor - update called before prior reading complete - initiated:0 waiting_for_interrupt:1
```

### Additional information

Tried:
 - using a few different pins
 - trying other VL53L1X sensors
 - powering sensor via 3.3v & 5v

I haven't been able to find confirmation on the ESPHome Discord or HA forums that people have successfully used this feature. 

Perhaps someone with a sensor can first confirm if the feature works for them and confirm the config required (and how to use etc)?  

There's some more info on this Discord thread where I was intially seeking support.
https://discord.com/channels/429907082951524364/1026044937646657586

Hilarious83, 2022-10-07:
same problem here, did you manage to fix this?

OkhammahkO, 2022-10-08:
Nope. I found an alternative solution which involves switching off the power to all of my 5v sensors via my solar power manager (which has that feature).

But I'm still interested in using the XSHUT feature.

smarthomedesign, 2024-03-16:
hello all, any news about this issue ? did anyone manage to fix it ?