esphome/issues #3646: Communication with HTU21D failed!

Tonguc-Endem, 2022-10-05:
### The problem

I know the same issue was submitted before but I'm 100% sure that this is a new issue.

This started happening on two separate devices after updating to the latest firmware.
They were functioning perfectly fine until I hit update on each of these devices which have been showing "UPDATE AVAILABE" for some time.

I have tried reversing GPIO2 and GPIO0, tried 100KHz frequency, tried a different HTU21D module, tried a different power supply, checked connections, basicaly everything you can think of.

The important thing is that they worked until the update.

**The following is from an ESP01:**
```
[16:57:36][C][logger:275]: Logger:
[16:57:36][C][logger:276]:   Level: DEBUG
[16:57:36][C][logger:277]:   Log Baud Rate: 0
[16:57:36][C][logger:278]:   Hardware UART: UART0
[16:57:36][C][i2c.arduino:038]: I2C Bus:
[16:57:36][C][i2c.arduino:039]:   SDA Pin: GPIO0
[16:57:36][C][i2c.arduino:040]:   SCL Pin: GPIO2
[16:57:36][C][i2c.arduino:041]:   Frequency: 50000 Hz
[16:57:36][C][i2c.arduino:044]:   Recovery: bus successfully recovered
[16:57:36][I][i2c.arduino:054]: Results from i2c bus scan:
[16:57:36][I][i2c.arduino:056]: Found no i2c devices!
[16:57:36][C][htu21d:028]: HTU21D:
[16:57:36][C][htu21d:029]:   Address: 0x40
[16:57:36][E][htu21d:031]: Communication with HTU21D failed!
...
```
**And the following is from an ESP32:**
```
[17:08:22][C][i2c.arduino:038]: I2C Bus:
[17:08:22][C][i2c.arduino:039]:   SDA Pin: GPIO21
[17:08:22][C][i2c.arduino:040]:   SCL Pin: GPIO22
[17:08:22][C][i2c.arduino:041]:   Frequency: 50000 Hz
[17:08:22][C][i2c.arduino:044]:   Recovery: bus successfully recovered
[17:08:22][I][i2c.arduino:054]: Results from i2c bus scan:
[17:08:22][I][i2c.arduino:060]: Found i2c device at address 0x40
[17:08:22][I][i2c.arduino:060]: Found i2c device at address 0x41
...
[17:08:22][C][htu21d:028]: HTU21D:
[17:08:22][C][htu21d:029]:   Address: 0x40
[17:08:22][E][htu21d:031]: Communication with HTU21D failed!
[17:08:22][C][htu21d:033]:   Update Interval: 60.0s
```

### Which version of ESPHome has the issue?

ESPHome Current version: 2022.9.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant 2022.9.7

### What platform are you using?

ESP8266

### Board

ESP01, ESP32

### Component causing the issue

i2c

### Example YAML snippet

```yaml
esphome:
  name: htu21d-01
  comment: Corridor

esp8266:
  board: esp01_1m

# Enable logging
logger:
  baud_rate: 0 #disable logging over uart
  
# Enable Home Assistant API
api:

ota:
  password: "..."

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    ssid: "HTU21D-01 Fallback Hotspot"
    password: "..."

captive_portal:

# Configuration for HTU21D Temperature & Humidity sensor
i2c:
  sda: GPIO0
  scl: GPIO2
  scan: true

sensor:
  - platform: htu21d
    temperature:
      name: "HTU21D-01 Temperature"
      accuracy_decimals: 1
    humidity:
      name: "HTU21D-01 Humidity"
      accuracy_decimals: 1
    update_interval: 60s
```

### Anything in the logs that might be useful for us?

```txt
The following is from an ESP01:

[16:57:36][C][logger:275]: Logger:
[16:57:36][C][logger:276]:   Level: DEBUG
[16:57:36][C][logger:277]:   Log Baud Rate: 0
[16:57:36][C][logger:278]:   Hardware UART: UART0
[16:57:36][C][i2c.arduino:038]: I2C Bus:
[16:57:36][C][i2c.arduino:039]:   SDA Pin: GPIO0
[16:57:36][C][i2c.arduino:040]:   SCL Pin: GPIO2
[16:57:36][C][i2c.arduino:041]:   Frequency: 50000 Hz
[16:57:36][C][i2c.arduino:044]:   Recovery: bus successfully recovered
[16:57:36][I][i2c.arduino:054]: Results from i2c bus scan:
[16:57:36][I][i2c.arduino:056]: Found no i2c devices!
[16:57:36][C][htu21d:028]: HTU21D:
[16:57:36][C][htu21d:029]:   Address: 0x40
[16:57:36][E][htu21d:031]: Communication with HTU21D failed!
...
And the following is from an ESP32:

[17:08:22][C][i2c.arduino:038]: I2C Bus:
[17:08:22][C][i2c.arduino:039]:   SDA Pin: GPIO21
[17:08:22][C][i2c.arduino:040]:   SCL Pin: GPIO22
[17:08:22][C][i2c.arduino:041]:   Frequency: 50000 Hz
[17:08:22][C][i2c.arduino:044]:   Recovery: bus successfully recovered
[17:08:22][I][i2c.arduino:054]: Results from i2c bus scan:
[17:08:22][I][i2c.arduino:060]: Found i2c device at address 0x40
[17:08:22][I][i2c.arduino:060]: Found i2c device at address 0x41
...
[17:08:22][C][htu21d:028]: HTU21D:
[17:08:22][C][htu21d:029]:   Address: 0x40
[17:08:22][E][htu21d:031]: Communication with HTU21D failed!
[17:08:22][C][htu21d:033]:   Update Interval: 60.0s
```

1danielsan, 2022-11-08:
same issue to me.... :(
esp32 doit module + htu2x i2c bus on pin 21&22.... working since last update...

klegerj, 2022-11-08:
same issue to me.... :(
esp32 doit module + htu2x i2c bus on pin 21&22.... working since last update...

mjbogusz, 2022-11-18:
While debugging a similar issue using multiple sensors (Si7021==HTU21D and BME280) on a single I2C bus I've noticed that the first sensor defined in the config fails every time while next ones work without issues, regardless of their types, adresses etc.
I've worked around it by defining a dummy 'sacrificial' sensor as the first one - could you try this?

1danielsan, 2022-11-19:
> 

Hello, could you help me  with the code? 
How do you define a dummy sensor as the first one? could you please write me an example?
thank you very much!

DoerakkerHA, 2022-11-19:
> While debugging a similar issue using multiple sensors (Si7021==HTU21D and BME280) on a single I2C bus I've noticed that the first sensor defined in the config fails every time while next ones work [...]

I will give it a try but would appreciate your example scrificial lamb

mjbogusz, 2022-11-19:
> Hello, could you help me with the code? How do you define a dummy sensor as the first one? could you please write me an example? thank you very much!

I've simply added an additional sensor with another address, something like this:
```yaml
sensor:
  - platform: htu21d
    address: 0x39
    temperature:
      name: ${upper_devicename} dummy temp
    humidity:
      name: ${upper_devicename} dummy humidity
  - platform: htu21d
    address: 0x40
    temperature:
      name: ${upper_devicename} valid temp
    humidity:
      name: ${upper_devicename} valid humidity
```
---
However, in case of DoerakkerHA, the part with
>  [09:41:51][I][i2c.arduino:070]: Found no i2c devices!

tells me that you most probably have a connectivity problem - either swapped pins, some broken wire or possibly too weak pullups; for me the automatic I2C scan always showed valid addresses if connected correctly.
Additionally, I see you're explicitly configuring I2C ID for the sensors - what's your I2C bus config?

DoerakkerHA, 2022-11-19:
i2c:
  i2c_component

The ONLY change made was to change OS from Tasmota to Esphome. The actually heater where this is physically installed was not touched or changed in any other way

mjbogusz, 2022-11-19:
it should be something like:
```yaml
i2c:
  - id: i2c_component
```
([reference](https://esphome.io/components/i2c.html))

Also try setting the pins (`sda` and `scl`) in the i2c config manually/explicitly - it is possible that the board config you've selected defaults to other values than your Tasmota config.
For testing purposes you can lower the frequency to 10kHz as well, it may help if the line is noisy (again, Tasmota may default to lower frequencies and/or have better error handling).

DoerakkerHA, 2022-11-19:
Thanks I will try it later. Just to rule out I didn't loose my marbles I just OTA Tasmota sensors back on it:
[image]

DoerakkerHA, 2022-11-20:
> it should be something like:
>
> ```yaml
>
> i2c:
>
>   - id: i2c_component
>
> ```
>
> ([reference](https://esphome.io/components/i2c.html))
>
>
>
> Also try setting the pins (`sda` and `scl`) in the i2c config [...]

Stumbled across some posts relating to this issue. It turns out it is not a i2c device but a semi MCU -> 1wire solution from sonoff to make the Si7021 compatible with their TH10/16 series and AM2031 sensor package

Now I know that the discovery has been (via the TH21d.h package) looked into. I guess that was work done for Tasmota

So it is not a i2c issue but from here I have no idea where to look next

1danielsan, 2022-12-06:
Today I Updated esphome to the latest 2022.11.5 version. recompiled and upload the code. 
SAME ISSUE. 
no sensor found....

NikonNLG, 2022-12-06:
Same for me for a years.
HTU21D works correctly on ESP8266, but failed on ESP32 very often without any reasons.
It doesn't depend from count of connected devices, pins (I tried to choose different).
In verbose mode it looks like this:
```
[23:04:13][VV][i2c.arduino:140]: 0x40 TX F3
[23:04:13][VV][i2c.arduino:116]: 0x40 RX 6850
[23:04:13][VV][i2c.arduino:140]: 0x40 TX F5
[23:04:13][VV][i2c.arduino:164]: TX failed: not acknowledged
[23:04:13][V][component:200]: Component htu21d.sensor took a long time for an operation (0.06 s).
[23:04:13][V][component:201]: Components should block for at most 20-30ms.
```
But SOMETIMES it start to work, without touching any wires or something like that. I'm confused how to reproduce bug

ChangappaMA, 2022-12-07:
Even I'm facing the same issue with connecting HTU21D with Esp01.

Logs:

INFO Reading configuration /config/esphome/testtempsense.yaml...
INFO Starting log output from testtempsense.local using esphome API
WARNING Can't connect to ESPHome API for testtempsense.local: Timeout while connecting to ('192.168.0.135', 6053)
INFO Trying to reconnect to testtempsense.local in the background
INFO Successfully connected to testtempsense.local
[13:00:25][I][app:102]: ESPHome version 2022.11.5 compiled on Dec  7 2022, 12:47:58
[13:00:27][C][wifi:504]: WiFi:
[13:00:27][C][wifi:362]:   Local MAC: E8:68:E7:CB:CC:D3
[13:00:27][C][wifi:363]:   SSID: [redacted]
[13:00:27][C][wifi:364]:   IP Address: 192.168.0.135
[13:00:27][C][wifi:365]:   BSSID: [redacted]
[13:00:27][C][wifi:367]:   Hostname: 'testtempsense'
[13:00:27][C][wifi:369]:   Signal strength: -54 dB ▂▄▆█
[13:00:27][C][wifi:373]:   Channel: 2
[13:00:27][C][wifi:374]:   Subnet: 255.255.255.0
[13:00:27][C][wifi:375]:   Gateway: 192.168.0.1
[13:00:27][C][wifi:376]:   DNS1: 192.168.0.1
[13:00:27][C][wifi:377]:   DNS2: 0.0.0.0
[13:00:27][C][logger:293]: Logger:
[13:00:27][C][logger:294]:   Level: DEBUG
[13:00:27][C][logger:295]:   Log Baud Rate: 115200
[13:00:27][C][logger:296]:   Hardware UART: UART0
[13:00:27][C][i2c.arduino:052]: I2C Bus:
[13:00:27][C][i2c.arduino:053]:   SDA Pin: GPIO0
[13:00:27][C][i2c.arduino:054]:   SCL Pin: GPIO2
[13:00:32][C][i2c.arduino:055]:   Frequency: 50000 Hz
[13:00:32][C][i2c.arduino:058]:   Recovery: bus successfully recovered
[13:00:32][I][i2c.arduino:068]: Results from i2c bus scan:
[13:00:32][I][i2c.arduino:074]: Found i2c device at address 0x13
[13:00:32][E][i2c.arduino:076]: Unknown error at address 0x1F
[13:00:32][E][i2c.arduino:076]: Unknown error at address 0x21
[13:00:32][E][i2c.arduino:076]: Unknown error at address 0x26
[13:00:32][E][i2c.arduino:076]: Unknown error at address 0x2B
[13:00:32][E][i2c.arduino:076]: Unknown error at address 0x2E
[13:00:32][I][i2c.arduino:074]: Found i2c device at address 0x33
[13:00:32][E][i2c.arduino:076]: Unknown error at address 0x45
[13:00:32][E][i2c.arduino:076]: Unknown error at address 0x4A
[13:00:32][E][i2c.arduino:076]: Unknown error at address 0x4E
[13:00:32][E][i2c.arduino:076]: Unknown error at address 0x57
[13:00:32][I][i2c.arduino:074]: Found i2c device at address 0x58
[13:00:32][I][i2c.arduino:074]: Found i2c device at address 0x68
INFO testtempsense.local: Ping timed out!
INFO Disconnected from ESPHome API for testtempsense.local
WARNING Disconnected from API
INFO Successfully connected to testtempsense.local
[13:01:24][D][api:102]: Accepted 192.168.0.176
[13:01:24][W][api.connection:071]: 192.168.0.176: Socket operation failed: CONNECTION_CLOSED errno=11
INFO testtempsense.local: Ping timed out!
INFO Disconnected from ESPHome API for testtempsense.local
WARNING Disconnected from API

ChangappaMA, 2022-12-07:
INFO Reading configuration /config/esphome/testtempsense.yaml...
INFO Starting log output from testtempsense.local using esphome API
INFO Successfully connected to testtempsense.local
[14:18:08][I][app:102]: ESPHome version 2022.11.5 compiled on Dec  7 2022, 14:01:45
[14:18:09][C][wifi:504]: WiFi:
[14:18:09][C][wifi:362]:   Local MAC: 
[14:18:09][C][wifi:363]:   SSID: [redacted]
[14:18:09][C][wifi:364]:   IP Address: 192.168.0.135
[14:18:09][C][wifi:365]:   BSSID: [redacted]
[14:18:09][C][wifi:367]:   Hostname: 'testtempsense'
[14:18:09][C][wifi:369]:   Signal strength: -63 dB ▂▄▆█
[14:18:09][C][wifi:373]:   Channel: 2
[14:18:09][C][wifi:374]:   Subnet: 255.255.255.0
[14:18:09][C][wifi:375]:   Gateway: 192.168.0.1
[14:18:09][C][wifi:376]:   DNS1: 192.168.0.1
[14:18:09][C][wifi:377]:   DNS2: 0.0.0.0
[14:18:09][C][logger:293]: Logger:
[14:18:09][C][logger:294]:   Level: DEBUG
[14:18:09][C][logger:295]:   Log Baud Rate: 115200
[14:18:09][C][logger:296]:   Hardware UART: UART0
[14:18:09][C][i2c.arduino:052]: I2C Bus:
[14:18:09][C][i2c.arduino:053]:   SDA Pin: GPIO0
[14:18:09][C][i2c.arduino:054]:   SCL Pin: GPIO2
[14:18:09][C][i2c.arduino:055]:   Frequency: 50000 Hz
[14:18:09][C][i2c.arduino:058]:   Recovery: bus successfully recovered
[14:18:09][I][i2c.arduino:068]: Results from i2c bus scan:
[14:18:09][I][i2c.arduino:070]: Found no i2c devices!
[14:18:09][C][htu21d:028]: HTU21D:
[14:18:09][C][htu21d:029]:   Address: 0x40
[14:18:09][E][htu21d:031]: Communication with HTU21D failed!
[14:18:09][C][htu21d:033]:   Update Interval: 60.0s
[14:18:09][C][htu21d:034]:   Temperature '${upper_devicename} dummy temp'
[14:18:09][C][htu21d:034]:     Device Class: 'temperature'
[14:18:09][C][htu21d:034]:     State Class: 'measurement'
[14:18:09][C][htu21d:034]:     Unit of Measurement: '°C'
[14:18:09][C][htu21d:034]:     Accuracy Decimals: 1
[14:18:09][C][htu21d:035]:   Humidity '${upper_devicename} dummy humidity'
[14:18:09][C][htu21d:035]:     Device Class: 'humidity'
[14:18:09][C][htu21d:035]:     State Class: 'measurement'
[14:18:09][C][htu21d:035]:     Unit of Measurement: '%'
[14:18:09][C][htu21d:035]:     Accuracy Decimals: 1
[14:18:09][C][htu21d:028]: HTU21D:
[14:18:09][C][htu21d:029]:   Address: 0x40
[14:18:09][E][htu21d:031]: Communication with HTU21D failed!
[14:18:09][C][htu21d:033]:   Update Interval: 60.0s
[14:18:09][C][htu21d:034]:   Temperature '${upper_devicename} valid temp'
[14:18:09][C][htu21d:034]:     Device Class: 'temperature'
[14:18:09][C][htu21d:034]:     State Class: 'measurement'
[14:18:09][C][htu21d:034]:     Unit of Measurement: '°C'
[14:18:09][C][htu21d:034]:     Accuracy Decimals: 1
[14:18:09][C][htu21d:035]:   Humidity '${upper_devicename} valid humidity'
[14:18:09][C][htu21d:035]:     Device Class: 'humidity'
[14:18:09][C][htu21d:035]:     State Class: 'measurement'
[14:18:09][C][htu21d:035]:     Unit of Measurement: '%'
[14:18:09][C][htu21d:035]:     Accuracy Decimals: 1
[14:18:09][C][captive_portal:088]: Captive Portal:
[14:18:09][C][mdns:103]: mDNS:
[14:18:09][C][mdns:104]:   Hostname: testtempsense
[14:18:09][C][ota:093]: Over-The-Air Updates:
[14:18:09][C][ota:094]:   Address: testtempsense.local:8266
[14:18:09][C][ota:097]:   Using Password.
[14:18:09][C][api:138]: API Server:
[14:18:09][C][api:139]:   Address: testtempsense.local:6053
[14:18:09][C][api:141]:   Using noise encryption: YES

NikonNLG, 2022-12-27:
If someone have same issue:
It start work with 2.2kOhm resistors on SDA/SCL.
Still not sure why HTU21D work on ESP8266 without resistors.

ChangappaMA, 2022-12-27:
No Luck 😞 

> If someone have same issue: It start work with 2.2kOhm resistors on SDA/SCL. Still not sure why HTU21D work on ESP8266 without resistors.