esphome/issues #3647: raspiaudio muse proto board no audio

MiAutomations, 2022-10-05:
### The problem

Hello to all. 

I just need some help with the ESP - Raspiaudio-muse-proto board, I had successfully added the device/Board to my Home assistant, I'm able also to control the switch and the light on this board over the ESP integration but I not able listening anything... I try radio browser, and also the Text to Speech without success. And I also tested the same Board with a different firmware that use Bluetooth connection and I'm able to listening the music, I presume from this point that the problem is not on the board...
I already try contact the support of the of the Raspiaudio that point me to this site also 

[image]

[image]

I already removed the device and the ESP integration starting again all the installation without success 

And the device still online with all the updates installed 

 I don't have any further idea how to solve this issue...  Is there any log to check this?

Thank you for your support and best regards

### Which version of ESPHome has the issue?

11.1.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.10.0b6

### What platform are you using?

ESP32

nielsnl68, 2022-10-05:
Please show your yaml file otherwise we cant help you.

MiAutomations, 2022-10-05:
> Please show your yaml file otherwise we cant help you.

What YAML file do you need?

nielsnl68, 2022-10-05:
the one you used to setup the esphome device to enable the media-player

MiAutomations, 2022-10-06:
> the one you used to setup the esphome device to enable the media-player

Sorry but I'm not using any YAML file to configure the ESPhome to enable the media-player.

When I restart the HA the ESP appears as a new device available to be added, I added this device and for the UI I try to play radio browser

MiAutomations, 2022-10-07:
Hello Yes I'm new in the ESP world 

I follow all the steps and I'm receiving the following erro: 

[image]

nielsnl68, 2022-10-09:
As i wrote in the script you should have changed the `device: mediaplayer' line where `mediaplayer` should be replaced with the name this device was given on your.

MiAutomations, 2022-10-11:
> As i wrote in the script you should have changed the `device: mediaplayer' line where `mediaplayer` should be replaced with the name this device was given on your.

Ok sorry, now is working I successfully install the device, but now I'm with 2 devices in a ESP list ... Is this normal? 

[image]

And I'm not able to see the device in available in the media -> radio Browser even with another name the protoboard is not appear in the list ...

[image]

Thank you for your support and best regards

nielsnl68, 2022-10-11:
the first issue will go away when you restart the esphome add-on

The second one, did you added the protoboard device in the home-assistant integrations list?

MiAutomations, 2022-10-11:
Ok after the addon restart all are available now, thank you

But I still without sound on the protoboard or in the speaker connected :(