esphome/issues #3648: cc1plus not found

fhempy, 2022-10-05:
### The problem

Unfortunately I can't compile any of my yamls any more. I'm using version 2022.9.2. See the error below:
```
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- ESPAsyncTCP-esphome @ 1.2.3
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- ESPAsyncTCP-esphome @ 1.2.3
|   |-- Hash @ 1.0
|   |-- ESP8266WiFi @ 1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- AsyncMqttClient-esphome @ 0.8.6
|   |-- ESPAsyncTCP-esphome @ 1.2.3
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/airpurifier/src/esphome/components/adc/adc_sensor.cpp.o
xtensa-lx106-elf-g++: fatal error: cannot execute 'cc1plus': execvp: No such file or directory
compilation terminated.
Compiling .pioenvs/airpurifier/src/esphome/components/binary_sensor/automation.cpp.o
xtensa-lx106-elf-g++: fatal error: cannot execute 'cc1plus': execvp: No such file or directory
compilation terminated.
Compiling .pioenvs/airpurifier/src/esphome/components/binary_sensor/binary_sensor.cpp.o
xtensa-lx106-elf-g++: fatal error: cannot execute 'cc1plus': execvp: No such file or directory
compilation terminated.
*** [.pioenvs/airpurifier/src/esphome/components/adc/adc_sensor.cpp.o] Error 1
*** [.pioenvs/airpurifier/src/esphome/components/binary_sensor/automation.cpp.o] Error 1
*** [.pioenvs/airpurifier/src/esphome/components/binary_sensor/binary_sensor.cpp.o] Error 1
Compiling .pioenvs/airpurifier/src/esphome/components/binary_sensor/filter.cpp.o
xtensa-lx106-elf-g++: fatal error: cannot execute 'cc1plus': execvp: No such file or directory
compilation terminated.
*** [.pioenvs/airpurifier/src/esphome/components/binary_sensor/filter.cpp.o] Error 1
========================== [FAILED] Took 3.14 seconds ==========================
```

### Which version of ESPHome has the issue?

2022.9.2

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

none

### What platform are you using?

ESP8266

### Board

d1_mini

### Anything in the logs that might be useful for us?

```txt
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- ESPAsyncTCP-esphome @ 1.2.3
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- ESPAsyncTCP-esphome @ 1.2.3
|   |-- Hash @ 1.0
|   |-- ESP8266WiFi @ 1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- AsyncMqttClient-esphome @ 0.8.6
|   |-- ESPAsyncTCP-esphome @ 1.2.3
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/airpurifier/src/esphome/components/adc/adc_sensor.cpp.o
xtensa-lx106-elf-g++: fatal error: cannot execute 'cc1plus': execvp: No such file or directory
compilation terminated.
Compiling .pioenvs/airpurifier/src/esphome/components/binary_sensor/automation.cpp.o
xtensa-lx106-elf-g++: fatal error: cannot execute 'cc1plus': execvp: No such file or directory
compilation terminated.
Compiling .pioenvs/airpurifier/src/esphome/components/binary_sensor/binary_sensor.cpp.o
xtensa-lx106-elf-g++: fatal error: cannot execute 'cc1plus': execvp: No such file or directory
compilation terminated.
*** [.pioenvs/airpurifier/src/esphome/components/adc/adc_sensor.cpp.o] Error 1
*** [.pioenvs/airpurifier/src/esphome/components/binary_sensor/automation.cpp.o] Error 1
*** [.pioenvs/airpurifier/src/esphome/components/binary_sensor/binary_sensor.cpp.o] Error 1
Compiling .pioenvs/airpurifier/src/esphome/components/binary_sensor/filter.cpp.o
xtensa-lx106-elf-g++: fatal error: cannot execute 'cc1plus': execvp: No such file or directory
compilation terminated.
*** [.pioenvs/airpurifier/src/esphome/components/binary_sensor/filter.cpp.o] Error 1
========================== [FAILED] Took 3.14 seconds ==========================
```

ssieb, 2022-10-05:
That `g++` is not the one used for compiling esphome.  platformio downloads the cross-compiler that is required.  See that it's `xtensa-lx106-elf-g++` that is getting called.  Are you running a 32-bit OS on the pi?  Try deleting the `.platformio` directory and see what buildtools platformio will download.

fhempy, 2022-10-05:
Yes, it's a 32-bit OS on the pi.

Many thanks for your fast support, deleting `.platformio` fixed the issue 👍

closed 2022-10-05 by fhempy