esphome/issues #3650: ESP-IDF with display component `graph`

webash, 2022-10-05:
### The problem

On linking when using a `graph` component on a `st7735` display:

```
/home/ashley/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/agnh-esph-env0/src/esphome/components/display/display_buffer.o:(.literal._ZN7esphome7display13DisplayBuffer5graphEiiPNS_5graph5GraphENS_5ColorE+0x0): undefined reference to `esphome::graph::Graph::draw(esphome::display::DisplayBuffer*, unsigned short, unsigned short, esphome::Color)'
/home/ashley/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/agnh-esph-env0/src/esphome/components/display/display_buffer.o: in function `esphome::display::DisplayBuffer::graph(int, int, esphome::graph::Graph*, esphome::Color)':
/home/ashley/esphome/.esphome/build/agnh-esph-env0/src/esphome/components/display/display_buffer.cpp:256: undefined reference to `esphome::graph::Graph::draw(esphome::display::DisplayBuffer*, unsigned short, unsigned short, esphome::Color)'
/home/ashley/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/agnh-esph-env0/src/main.o:(.literal._Z5setupv+0x130): undefined reference to `vtable for esphome::graph::Graph'
```

### Which version of ESPHome has the issue?

2022.9.3

### What type of installation are you using?

pip

### What platform are you using?

ESP32-IDF

### Component causing the issue

az-delivery-devkit-v4

### Example YAML snippet

```yaml
color:
  - id: graph_red
    red: 100%
    green: 0%
    blue: 0%
  - id: graph_green
    red: 0%
    green: 100%
    blue: 0%
  - id: graph_blue
    red: 0%
    green: 0%
    blue: 100%
  - id: white
    red: 100%
    green: 100%
    blue: 100%
    

graph:
  - id: particulates_simple_graph
    sensor: pms5003_10um
    duration: 8h
    width: 150
    height: 50
  - id: particulates_graph
    duration: 8h
    x_grid: 1h
    y_grid: 1.0     # degC/div
    width: 150
    height: 60
    traces:
      - sensor: pms5003_1um
        line_type: DOTTED
        line_thickness: 2
        color: graph_green
      - sensor: pms5003_2dot5um
        line_type: DASHED
        line_thickness: 2
        color: graph_red
      - sensor: pms5003_10um
        line_type: SOLID
        line_thickness: 2
        color: graph_blue

display:
#https://esphome.io/components/display/st7735.html
  - platform: st7735
    model: "INITR_MINI160X80"
    cs_pin: GPIO5
    dc_pin: GPIO19
    device_width: 80
    device_height: 160
    rotation: 90
    col_start: 0
    row_start: 0
    eight_bit_color: true
    update_interval: 30s
    lambda: |-
      auto red = Color(255, 0, 0);
      auto green = Color(0, 255, 0);
      auto blue = Color(0, 0, 255);
      auto white = Color(255, 255, 255);
      auto black = Color(0, 0, 0);
      it.fill(black);
      it.graph(5, 20, id(particulates_simple_graph));
```

### Anything in the logs that might be useful for us?

```txt
/home/ashley/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/agnh-esph-env0/src/esphome/components/display/display_buffer.o:(.literal._ZN7esphome7display13DisplayBuffer5graphEiiPNS_5graph5GraphENS_5ColorE+0x0): undefined reference to `esphome::graph::Graph::draw(esphome::display::DisplayBuffer*, unsigned short, unsigned short, esphome::Color)'
/home/ashley/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/agnh-esph-env0/src/esphome/components/display/display_buffer.o: in function `esphome::display::DisplayBuffer::graph(int, int, esphome::graph::Graph*, esphome::Color)':
/home/ashley/esphome/.esphome/build/agnh-esph-env0/src/esphome/components/display/display_buffer.cpp:256: undefined reference to `esphome::graph::Graph::draw(esphome::display::DisplayBuffer*, unsigned short, unsigned short, esphome::Color)'
/home/ashley/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/agnh-esph-env0/src/main.o:(.literal._Z5setupv+0x130): undefined reference to `vtable for esphome::graph::Graph'
```

webash, 2023-02-03:
Still a problem

webash, 2023-03-15:
Anything further that can be done here? Like is there a way I could get my hands dirty and help with it to submit a PR? I just don't really know the space to know where to start.

linkedupbits, 2024-01-10:
Is this still an issue? I have  an ST7735 working now so could try to reproduce it?

linkedupbits, 2024-01-10:
Sorry - I can't reproduce it:
``` yaml
esphome:
  name: t-relay
  friendly_name: t-relay

esp32:
  board: esp32dev
  framework:
    type: esp-idf
    

# Enable logging
# Logger must be at least debug (default)
logger:
  level: DEBUG

# Enable Home Assistant API
api:
  encryption: 
    key: !secret grandviewcontrol_encryption_key

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

spi:
  - id: spi_bus0
    clk_pin: GPIO14 # purple
    mosi_pin: GPIO27 # blue
    #miso_pin: GPIO19 # optional
output:
  # Setup a pin to control the backlight
  - platform: ledc
    pin: GPIO13 # grey => LED
    id: backlight_pwm

light:
  # Define a monochromatic, dimmable light for the backlight
  - platform: monochromatic
    output: backlight_pwm
    name: "Display Backlight"
    id: back_light
    restore_mode: ALWAYS_ON

sensor:
  - platform: homeassistant
    id: inside_temperature
    entity_id: sensor.office_aqara_temperature
    internal: true

graph:
  - id: particulates_simple_graph
    sensor: inside_temperature
    duration: 5minutes
    width: 100
    height: 50

display:
  - platform: st7735
    model: "INITR_REDTAB"
    reset_pin: GPIO22 # yellow
    cs_pin: GPIO26 # orange
    dc_pin: GPIO23 # green (rs = register select)
    rotation: 0
    device_width: 128
    device_height: 128
    col_start: 0
    row_start: 0
    eight_bit_color: False
    update_interval: 500ms
    lambda: |-
      auto red = Color(255, 0, 0);
      auto green = Color(0, 255, 0);
      auto blue = Color(0, 0, 255);
      auto white = Color(255, 255, 255);
      auto black = Color(0, 0, 0);
      it.fill(black);
      it.graph(5, 20, id(particulates_simple_graph));
      
```

```
INFO ESPHome 2024.1.0-dev
INFO Reading configuration config/t-relay-githubissue.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing t-relay (board: esp32dev; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40405.230623 (4.4.5) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.4
Compiling .pioenvs/t-relay/src/main.o
Linking .pioenvs/t-relay/firmware.elf
RAM:   [=         ]  10.5% (used 34520 bytes from 327680 bytes)
Flash: [======    ]  55.7% (used 1021705 bytes from 1835008 bytes)
Building .pioenvs/t-relay/firmware.bin
Creating esp32 image...
Successfully created esp32 image.
esp32_create_combined_bin([".pioenvs/t-relay/firmware.bin"], [".pioenvs/t-relay/firmware.elf"])
Wrote 0x10ad80 bytes to file /workspaces/esphome_tft_espi/config/.esphome/build/t-relay/.pioenvs/t-relay/firmware-factory.bin, ready to flash to offset 0x0
========================= [SUCCESS] Took 28.26 seconds =========================
```

menglert2, 2024-02-02:
I hit this with a SSD1306 drawing a graph on esp-idf.  Cleaning the build files looks to have resolved it for me.