esphome/issues #3653: Out of memory while trying to pair with active Bluetooth device

heytcass, 2022-10-06:
### The problem

Trying to get my Bluetooth proxy devices to work with a First Alert Onelink Smart Smoke + CO Alarm, as documented here: https://github.com/home-assistant/core/issues/77926

After updating yesterday with some encouraging news from Nick, I am getting an out of memory error on my NodeMCU-32s using a fresh flash direct from the proxy installer site.

There are additional issues with using an Olimex, as well, also documented in the above issue.

Error on Home Assistant frontend is attached in a screenshot:
[image]

### Which version of ESPHome has the issue?

2022.9.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.10.0

### What platform are you using?

ESP32-IDF

### Board

nodemcu-32s

### Component causing the issue

bluetooth_proxy

### Example YAML snippet

```yaml
substitutions:
  name: esp32-bluetooth-proxy-fe0114
packages:
  esphome.bluetooth-proxy: github://esphome/bluetooth-proxies/esp32-generic.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

### Anything in the logs that might be useful for us?

```txt
[D][api.connection:917]: Home Assistant 2022.10.0 (::FFFF:A00:20): Connected successfully
[I][ota:109]: Boot seems successful, resetting boot loop counter.
[D][esp32.preferences:113]: Saving 1 preferences to flash...
[D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[D][esp32_ble_tracker:264]: Starting scan...
[I][esp32_ble_client:058]: Attempting BLE connection to a0:e6:f8:2a:4e:fe
[D][esp32_ble_client:099]: [a0:e6:f8:2a:4e:fe] Unexpected conn_id in CONNECT_EVT: param conn=0, open conn=22136
[I][esp32_ble_client:142]: Service UUID: 0x1800
[I][esp32_ble_client:143]:   start_handle: 0x1  end_handle: 0x7
[I][esp32_ble_client.service:057]:  characteristic 0x2A00, handle 0x3, properties 0x2
[I][esp32_ble_client.service:057]:  characteristic 0x2A01, handle 0x5, properties 0x2
[I][esp32_ble_client.service:057]:  characteristic 0x2A04, handle 0x7, properties 0x2
[I][esp32_ble_client:142]: Service UUID: 0x1801
[I][esp32_ble_client:143]:   start_handle: 0x8  end_handle: 0x8
[I][esp32_ble_client:142]: Service UUID: 0x180A
[I][esp32_ble_client:143]:   start_handle: 0x9  end_handle: 0x1b
[... 63 lines cut ...]
[I][esp32_ble_client.service:057]:  characteristic E604E95D-A759-4817-87D3-AA005083A0D1, handle 0x99, properties 0x2
[I][esp32_ble_client.service:057]:  characteristic 0000000F-18E4-11E5-B60B-1697F925EC7B, handle 0x9b, properties 0xa
[I][esp32_ble_client.service:057]:  characteristic 0000000E-18E4-11E5-B60B-1697F925EC7B, handle 0x9e, properties 0xa
[I][esp32_ble_client.service:057]:  characteristic 00000010-18E4-11E5-B60B-1697F925EC7B, handle 0xa1, properties 0xa
[I][esp32_ble_client.service:057]:  characteristic 00000023-0000-1000-8000-0026BB765291, handle 0xa4, properties 0xa
[I][esp32_ble_client:142]: Service UUID: 00000096-0000-1000-8000-0026BB765291
[I][esp32_ble_client:143]:   start_handle: 0xa6  end_handle: 0xffff
[I][esp32_ble_client.service:057]:  characteristic E604E95D-A759-4817-87D3-AA005083A0D1, handle 0xa8, properties 0x2
[I][esp32_ble_client.service:057]:  characteristic 00000068-0000-1000-8000-0026BB765291, handle 0xaa, properties 0x2a
[I][esp32_ble_client.service:057]:  characteristic 0000008F-0000-1000-8000-0026BB765291, handle 0xae, properties 0x2a
[I][esp32_ble_client.service:057]:  characteristic 00000079-0000-1000-8000-0026BB765291, handle 0xb2, properties 0x2a
[I][esp32_ble_client.service:057]:  characteristic 00000023-0000-1000-8000-0026BB765291, handle 0xb6, properties 0xa
[D][esp32_ble_tracker:264]: Starting scan...
[D][esp-idf:000]: E (358645) MDNS_Networking: Cannot allocate memory (line: 123, free heap: 660 bytes)

[I][bluetooth_proxy:206]: [a0:e6:f8:2a:4e:fe] Disconnecting.
[W][bluetooth_proxy:247]: Cannot write GATT characteristic, not connected.
[W][bluetooth_proxy:222]: Cannot read GATT characteristic, not connected.
[W][bluetooth_proxy:247]: Cannot write GATT characteristic, not connected.
[W][bluetooth_proxy:222]: Cannot read GATT characteristic, not connected.
[W][bluetooth_proxy:247]: Cannot write GATT characteristic, not connected.
[W][bluetooth_proxy:222]: Cannot read GATT characteristic, not connected.
[W][bluetooth_proxy:247]: Cannot write GATT characteristic, not connected.
[W][bluetooth_proxy:222]: Cannot read GATT characteristic, not connected.
```

heytcass, 2022-10-06:
https://github.com/home-assistant/core/issues/77926#issuecomment-1270432414

Just tried again with a different ESP device, the Olimex. Again, freshly flashed. Different error this time:

13:10:45][I][esp32_ble_client:058]: Attempting BLE connection to a0:e6:f8:5d:3c:2e
[13:10:46][W][esp32_ble_client:089]: connect to 00:00:00:00:00:00 failed, status=133
[13:10:46][D][esp32_ble_tracker:264]: Starting scan...
[13:10:46][I][esp32_ble_client:058]: Attempting BLE connection to a0:e6:f8:5d:3c:2e
[13:10:51][I][esp32_ble_client:142]: Service UUID: 0x1800
[13:10:51][I][esp32_ble_client:143]:   start_handle: 0x1  end_handle: 0x7
[13:10:51][I][esp32_ble_client.service:057]:  characteristic 0x2A00, handle 0x3, properties 0x2
[13:10:51][I][esp32_ble_client.service:057]:  characteristic 0x2A01, handle 0x5, properties 0x2
[13:10:51][I][esp32_ble_client.service:057]:  characteristic 0x2A04, handle 0x7, properties 0x2
[13:10:51][I][esp32_ble_client:142]: Service UUID: 0x1801
[13:10:51][I][esp32_ble_client:143]:   start_handle: 0x8  end_handle: 0x8
[13:10:52][I][esp32_ble_client:142]: Service UUID: 0x180A
[13:10:52][I][esp32_ble_client:143]:   start_handle: 0x9  end_handle: 0x1b
[13:10:52][I][esp32_ble_client.service:057]:  characteristic 0x2A23, handle 0xb, properties 0x2
[13:10:52][I][esp32_ble_client.service:057]:  characteristic 0x2A24, handle 0xd, properties 0x2
[13:10:52][I][esp32_ble_client.service:057]:  characteristic 0x2A25, handle 0xf, properties 0x2
[13:10:52][I][esp32_ble_client.service:057]:  characteristic 0x2A26, handle 0x11, properties 0x2
[13:10:52][I][esp32_ble_client.service:057]:  characteristic 0x2A27, handle 0x13, properties 0x2
[13:10:52][I][esp32_ble_client.service:057]:  characteristic 0x2A28, handle 0x15, properties 0x2
[13:10:52][I][esp32_ble_client.service:057]:  characteristic 0x2A29, handle 0x17, properties 0x2
[13:10:52][I][esp32_ble_client.service:057]:  characteristic 0x2A2A, handle 0x19, properties 0x2
[13:10:52][I][esp32_ble_client.service:057]:  characteristic 0x2A50, handle 0x1b, properties 0x2
[13:10:52][I][esp32_ble_client:142]: Service UUID: 0000003E-0000-1000-8000-0026BB765291
[13:10:52][I][esp32_ble_client:143]:   start_handle: 0x1c  end_handle: 0x34
[13:10:52][I][esp32_ble_client.service:057]:  characteristic E604E95D-A759-4817-87D3-AA005083A0D1, handle 0x1e, properties 0x2
[13:10:52][I][esp32_ble_client.service:057]:  characteristic 00000014-0000-1000-8000-0026BB765291, handle 0x20, properties 0xa
[13:10:52][I][esp32_ble_client.service:057]:  characteristic 00000020-0000-1000-8000-0026BB765291, handle 0x23, properties 0xa
[13:10:52][I][esp32_ble_client.service:057]:  characteristic 00000021-0000-1000-8000-0026BB765291, handle 0x26, properties 0xa
[13:10:52][I][esp32_ble_client.service:057]:  characteristic 00000023-0000-1000-8000-0026BB765291, handle 0x29, properties 0xa
[13:10:52][I][esp32_ble_client.service:057]:  characteristic 00000030-0000-1000-8000-0026BB765291, handle 0x2c, properties 0xa
[13:10:52][I][esp32_ble_client.service:057]:  characteristic 00000052-0000-1000-8000-0026BB765291, handle 0x2f, properties 0xa
[13:10:52][I][esp32_ble_client.service:057]:  characteristic 000000A6-0000-1000-8000-0026BB765291, handle 0x32, properties 0x2a
[13:10:52][I][esp32_ble_client:142]: Service UUID: 000000A2-0000-1000-8000-0026BB765291
[13:10:52][I][esp32_ble_client:143]:   start_handle: 0x35  end_handle: 0x3a
[13:10:52][I][esp32_ble_client.service:057]:  characteristic E604E95D-A759-4817-87D3-AA005083A0D1, handle 0x37, properties 0x2
[13:10:52][I][esp32_ble_client.service:057]:  characteristic 00000037-0000-1000-8000-0026BB765291, handle 0x39, properties 0x2a
[13:10:52][I][esp32_ble_client:142]: Service UUID: 00000055-0000-1000-8000-0026BB765291
[13:10:52][I][esp32_ble_client:143]:   start_handle: 0x3b  end_handle: 0x49
[13:10:52][I][esp32_ble_client.service:057]:  characteristic E604E95D-A759-4817-87D3-AA005083A0D1, handle 0x3d, properties 0x2
[13:10:52][I][esp32_ble_client.service:057]:  characteristic 0000004C-0000-1000-8000-0026BB765291, handle 0x3f, properties 0xa
[13:10:52][I][esp32_ble_client.service:057]:  characteristic 0000004E-0000-1000-8000-0026BB765291, handle 0x42, properties 0xa
[13:10:52][I][esp32_ble_client.service:057]:  characteristic 0000004F-0000-1000-8000-0026BB765291, handle 0x45, properties 0xa
[13:10:52][I][esp32_ble_client.service:057]:  characteristic 00000050-0000-1000-8000-0026BB765291, handle 0x48, properties 0xa
[13:10:52][I][esp32_ble_client:142]: Service UUID: 00000087-0000-1000-8000-0026BB765291
[13:10:52][I][esp32_ble_client:143]:   start_handle: 0x4a  end_handle: 0x62
[13:10:52][I][esp32_ble_client.service:057]:  characteristic E604E95D-A759-4817-87D3-AA005083A0D1, handle 0x4c, properties 0x2
[13:10:52][I][esp32_ble_client.service:057]:  characteristic 0000000C-18E4-11E5-B60B-1697F925EC7B, handle 0x4e, properties 0xa
[13:10:52][I][esp32_ble_client.service:057]:  characteristic 00000005-18E4-11E5-B60B-1697F925EC7B, handle 0x51, properties 0xa
[13:10:52][I][esp32_ble_client.service:057]:  characteristic 00000007-18E4-11E5-B60B-1697F925EC7B, handle 0x54, properties 0xa
[13:10:52][I][esp32_ble_client.service:057]:  characteristic 00000006-18E4-11E5-B60B-1697F925EC7B, handle 0x57, properties 0xa
[13:10:52][I][esp32_ble_client.service:057]:  characteristic 00000008-18E4-11E5-B60B-1697F925EC7B, handle 0x5a, properties 0xa
[... cut ...]
INFO olimex-bluetooth-proxy-d10694.local: Error while reading incoming messages: Error while reading data: [Errno 104] Connection reset by peer
INFO olimex-bluetooth-proxy-d10694.local: Ping Failed: Error while reading data: [Errno 104] Connection reset by peer
INFO Disconnected from ESPHome API for olimex-bluetooth-proxy-d10694.local
WARNING Disconnected from API
INFO Successfully connected to olimex-bluetooth-proxy-d10694.local
And on Home Assistant frontend:
Screen Shot 2022-10-06 at 1 16 22 PM

nagyrobi, 2022-10-13:
Add to your config, and flash again:

```
debug:

sensor:
- platform: template
  name: Free HEAP
  lambda: return heap_caps_get_free_size(MALLOC_CAP_INTERNAL);
  icon: "mdi:memory"
  entity_category: diagnostic
  state_class: measurement
  unit_of_measurement: "b"
  update_interval: 60s
- platform: uptime
  name: Uptime
- platform: debug
  free:
    name: Free MEM
```
Wait for some time and look at the graphs (without trying to pair, just wait).

heytcass, 2022-10-14:
@nagyrobi Did as you said and found the attached four graphs. Very interesting, and based on the uptime graphs, it seems like the spikes are coming from the ESPs running themselves out of memory and rebooting themselves. It definitely doesn't come from me rebooting.

[image]
[image]
[image]
[image]

nagyrobi, 2022-10-14:
A memory leak has been identified causing this. The fix will be available in the next beta release.

heytcass, 2022-10-19:
Updated to 2020.10.0 this morning. Still having the same issue.

nagyrobi, 2022-10-20:
Note that Bluetooth Proxy currently only supports active connection with one single Bluetooth device.

heytcass, 2022-10-21:
Still having the same memory graphs show up on my Olimex Bluetooth Proxy:

[image]

nagyrobi, 2022-10-21:
Please post your _entire_ config

heytcass, 2022-10-21:
> Please post your _entire_ config

About as simple as it gets. Was initially installed from the Bluetooth Proxy web flasher, then just added what you suggested.

```
---
substitutions:
  name: olimex-bluetooth-proxy
packages:
  esphome.bluetooth-proxy: github://esphome/bluetooth-proxies/olimex-esp32-poe-iso.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false

button:
  - platform: restart
    name: ${name}" Restart"

debug:

sensor:
- platform: template
  name: Free HEAP
  lambda: return heap_caps_get_free_size(MALLOC_CAP_INTERNAL);
  icon: "mdi:memory"
  entity_category: diagnostic
  state_class: measurement
  unit_of_measurement: "b"
  update_interval: 60s
- platform: uptime
  name: Uptime
- platform: debug
  free:
    name: Free MEM
```

nagyrobi, 2022-10-21:
It must be related to the active device you're trying to use it with.
I use my Olimex with 10 Xiaomi BLE sensors transmitting BTHome packets, work stable now since latest release.