esphome/issues #3654: Crash when using Software UART

dentra, 2022-10-06:
### The problem

I have a factory made usb-stick with ESP8266EX which is flashed with ESPHome and connected to Air Purifier. I request (send)  and then receive binary data every 5 seconds. It may work from minutes to hours but always lead to crash when reading Software UART data in `esphome::uart::ESP8266SoftwareSerial::read_bit_`.  In my code I am always wait for enough data size is available and only then read array of bytes (there is a very lot of code but the interesting part is less than [20 lines](https://github.com/dentra/esphome-tion/blob/master/components/tion-api/tion-api-uart.cpp#L103))

Are there any restrictions on reading/writing to the Software UART, maybe I can change my code so as to try to avoid such a situation. Or maybe someone will direct me so that I can make a fix to the Software UART component. Decoded stack trace are included.

Unfortunately switching to Hardware UART is not possible because of fixed device and usb-stick pins.

### Which version of ESPHome has the issue?

2022.9.2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.9

### What platform are you using?

ESP8266

### Board

esp8266ex

### Component causing the issue

uart

### Example YAML snippet

```yaml
uart:
  id: uart_bus
  tx_pin: RX
  rx_pin: TX
  baud_rate: 9600
```

### Anything in the logs that might be useful for us?

```txt
0x40250290: pm_keep_active_enable at ??:?
0x40256279: ets_timer_handler_isr at ??:?
0x402562be: ets_timer_handler_isr at ??:?
0x40000f49: ?? ??:0
0x40000f49: ?? ??:0
0x40000f49: ?? ??:0
0x40000e19: ?? ??:0
0x401059f3: call_user_start_local at ??:?
0x4010112c: umm_init at ??:?
0x4010000d: call_user_start at ??:?
0x401000ac: app_entry_redefinable at ??:?
0x402464a9: cont_continue at cont.S.o:?
0x4010028d: esphome::uart::ESP8266SoftwareSerial::read_bit_(unsigned int*, unsigned int const&) at ??:?
0x40100302: esphome::uart::ESP8266SoftwareSerial::gpio_intr(esphome::uart::ESP8266SoftwareSerial*) at ??:?
0x40100302: esphome::uart::ESP8266SoftwareSerial::gpio_intr(esphome::uart::ESP8266SoftwareSerial*) at ??:?
0x40100a00: interrupt_handler at ??:?
0x4010028d: esphome::uart::ESP8266SoftwareSerial::read_bit_(unsigned int*, unsigned int const&) at ??:?
0x40100302: esphome::uart::ESP8266SoftwareSerial::gpio_intr(esphome::uart::ESP8266SoftwareSerial*) at ??:?
0x40100a1c: interrupt_handler at ??:?
0x40229a0d: _dtoa_r at /workdir/repo/newlib/newlib/libc/stdlib/dtoa.c:426
0x4010093c: interrupt_handler at ??:?
0x4000050c: ?? ??:0
0x4022ae1a: _Balloc at /workdir/repo/newlib/newlib/libc/stdlib/mprec.c:104 (discriminator 5)
0x40102654: pp_post at ??:?
0x401058eb: lmacRxDone at ??:?
0x40106185: ets_timer_disarm at ??:?
0x401031df: rcReachRetryLimit at ??:?
0x401006f0: ets_post at ??:?
0x40103876: wDev_ProcessFiq at ??:?
0x40106185: ets_timer_disarm at ??:?
0x48b17119: ?? ??:0

and another decoded stacktrace:
0x401006f0: ets_post at ??:?
0x402507fc: pm_keep_active_enable at ??:?
0x402210a8: delay_end at ??:?
0x402567e5: ets_timer_handler_isr at ??:?
0x4025682a: ets_timer_handler_isr at ??:?
0x402209a2: loop_task(ETSEventTag*) at core_esp8266_main.cpp:?
0x40000f49: ?? ??:0
0x40000f49: ?? ??:0
0x40000f49: ?? ??:0
0x40000e19: ?? ??:0
0x401059ed: call_user_start_local at ??:?
0x401059f3: call_user_start_local at ??:?
0x4010112c: umm_init at ??:?
0x4010000d: call_user_start at ??:?
0x401000ac: app_entry_redefinable at ??:?
0x40246a64: cont_ret at cont.S.o:?
0x40246a15: cont_continue at cont.S.o:?
0x4010028d: esphome::uart::ESP8266SoftwareSerial::read_bit_(unsigned int*, unsigned int const&) at ??:?
0x401006f0: ets_post at ??:?
0x40102654: pp_post at ??:?
```

hauntedhill, 2023-02-20:
i have a simular problem, does anyone find a solution or fix?

dentra, 2023-02-20:
Unfortunately I haven't yet :(

hauntedhill, 2023-02-20:
In my case i could switch to hardware UART. Does hardware UART solve the problem? Anybody know?

ssieb, 2023-02-20:
The stack trace is good, but you didn't include the actual error.  Have you disabled serial logging?

ssieb, 2023-02-20:
A hardware uart wouldn't have this problem.

dentra, 2023-02-21:
@ssieb as I remember it was WDT, I will grab new one and paste here full details. Yes, the serial logging is disabled by baud_rate: 0, I need this port for data exchange.

ssieb, 2023-02-21:
But why would someone make the board with the RX and TX backwards?

dentra, 2023-02-22:
Both devices are proprietary and it seems to me that this is intentional obfuscation as part of the protection

hamster65, 2023-10-01:
I have this, too. Hardware is a wemos d1 mini clone, ESP12-F. As soon as I define a software uart (without even using it), the ESP restarts randomly. Sometimes afer 200 Seconds, sometimes after 5000.

uart:
  id: uart_bus
  rx_pin:
    number: GPIO4 #D2
    mode:
      input: true
      pullup: true
  tx_pin: GPIO5 #D1
  baud_rate: 9600
  data_bits: 8
  parity: NONE
  stop_bits: 1

When I use the HW uart Pins, it runs stable.

closed 2026-10-11 by jesserockz