esphome/issues #3655: Reconfigure intergration spam

jLynx, 2022-10-06:
### The problem

I woke up this morning to HOmeAssistant spamming me with my same esphome device telling me I need to reconfigure it. This device has been working untouched for 6+ months. Now ass of a sudden I am getting these errors telling me I need to enter in the encryption key. The device was originally made without a key (When that was the default), but when I try and press ok on the modal without a key it says it requires one.
[image]
[image]

### Which version of ESPHome has the issue?

2022.9.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.09.1

### What platform are you using?

ESP8266

### Board

D1

### Example YAML snippet

```yaml
esphome:
  name: door-bell
  platform: ESP8266
  board: d1_mini

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "adb951e565fe0ee841f39800695aab67"

wifi:
  ssid: "Mothership"
  password: "AAAAAAAAAAAAAA"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Door-Bell Fallback Hotspot"
    password: "VRBq7lqrAvon"

captive_portal:

binary_sensor:
  - platform: gpio
    name: "Door bell pressed"
    pin:
      number: D2
      mode: INPUT_PULLUP
    filters:
      - delayed_off: 1000ms
```

jLynx, 2022-10-06:
Have just tried adding an encryption key. It uploaded successfully, I then went to try add the key to HomeAssistant, but it said the key was incorrect. So I ended up removing the key and flashing again to put it back to how it was. Still haven't been able to fix HomeAssistant yet

jLynx, 2022-10-07:
Now my HomeAssistant mobile app is freezing as there are 1000's showing up. They seem to keep adding. Only way to temporarily fix it is to restart HomeAssistant. I'm going to try remove it and re add it and see if that fixes it

jLynx, 2022-10-07:
Deleting it and re adding it seems to have solved it. Will see if it comes back

hannemann, 2022-10-16:
Had the same Issue today with one of my devices. Deleting and re-adding solved it.

kiklhorn, 2022-12-05:
I seen this behavior after router replacement on my network. Suspected swapped IP addresses between esphome devices.
Is there any way to show internal bindings (IP, name, whatever) between esphome devices and HA?