esphome/issues #3659: Error: RX Buffer can't be resized when Serial is already running.

spali, 2022-10-06:
### The problem

If watch the logs over serial during boot:
```shell
esphome.exe logs --device COM4 .\core2-test.yaml
```
I get the error bellow.

### Which version of ESPHome has the issue?

2022.9.3

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

m5stack-core-esp32

### Component causing the issue

uart

### Example YAML snippet

```yaml
esp32:
  board: m5stack-core-esp32
  framework:
    type: arduino
    version: 2.0.5
    platform_version: 5.2.0
```

### Anything in the logs that might be useful for us?

```txt
[22:54:20]configsip: 0, SPIWP:0xee
[22:54:20]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[22:54:20]mode:DIO, clock div:1
[22:54:20]load:0x3fff0030,len:1344
[22:54:20]load:0x40078000,len:13864
[22:54:20]load:0x40080400,len:3608
[22:54:20]entry 0x400805f0
[22:54:21][I][logger:243]: Log initialized
[22:54:21][C][ota:465]: There have been 1 suspected unsuccessful boot attempts.
[22:54:21][D][esp32.preferences:113]: Saving 1 preferences to flash...
[22:54:21][D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[22:54:21][I][app:029]: Running through setup()...
[22:54:21][I][i2c.arduino:161]: Performing I2C bus recovery
[22:54:21][C][spi:023]: Setting up SPI bus...
[22:54:21][C][uart.arduino_esp32:077]: Setting up UART...
[22:54:21][    75][E][HardwareSerial.cpp:525] setRxBufferSize(): RX Buffer can't be resized when Serial is already running.
[22:54:21]
```

### Additional information

This didn't happen before I updated framework and platform (used to use default before):

```
    version: 2.0.5
    platform_version: 5.2.0
```

I assume this line is the caller for the error:
https://github.com/esphome/esphome/blob/80d03a631e3bccc055d76cc642a6f8993f141df4/esphome/components/uart/uart_component_esp32_arduino.cpp#L103
esphome/esphome/blob/80d03a631e3bccc055d76cc642a6f8993f141df4/esphome/components/uart/uart_component_esp32_arduino.cpp#L103
And Here the code that throws the error:
https://github.com/espressif/arduino-esp32/blob/a5f03a86512b270be1080d7e7cb82e4fd71c9a1a/cores/esp32/HardwareSerial.cpp#L525

I'm not an expert, but looks like a switch of the `begin` and `setRxBufferSize` call could solve this?

formatBCE, 2023-03-02:
Not stale

kingozone, 2023-03-04:
Not stale

playya, 2023-04-13:
I had the same issue, but it's already fixec in dev: https://github.com/esphome/esphome/commit/9922eb83e2851f23992d021ae59adba5d56d73d7
Double checking if this works now ...

maxsarges, 2023-04-21:
Hi,

i have checked newest version of ESPHome and the Problem still exists.