esphome/issues #3663: ESP32 Safe Mode Boot became unavailable

tomi19921702, 2022-10-07:
### The problem

Every few minutes Safe Mode Boot became unavailable and don't get data of Bluetooth Devices.

### Which version of ESPHome has the issue?

2022.9.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.10

### What platform are you using?

ESP32

tomi19921702, 2022-10-11:
[image]

[image]
Attached a couple screenshots of the esphome logs when its helping

nagyrobi, 2022-10-13:
Add to your config, and flash again:

```
debug:

sensor:
- platform: template
  name: Free HEAP
  lambda: return heap_caps_get_free_size(MALLOC_CAP_INTERNAL);
  icon: "mdi:memory"
  entity_category: diagnostic
  state_class: measurement
  unit_of_measurement: "b"
  update_interval: 60s
- platform: uptime
  name: Uptime
- platform: debug
  free:
    name: Free MEM
```
Wait for some time and look at the graphs.

tomi19921702, 2022-10-13:
[image]

closed 2023-03-23 by github-actions[bot]