esphome/issues #3664: m5stick cplus - cannot have both wifi and ble tracker at the same time

evrardjp, 2022-10-07:
### The problem

If I am adding esp32_ble_tracker on my config (regardless of the scan_parameters), the wifi doesn't connect.

### Which version of ESPHome has the issue?

2022.9.2 (running from docker image sha256:791c8f1475d64389332702d50205175ade9b55a6a562ad2e81d)

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

Not using HA

### What platform are you using?

ESP32

### Board

m5stick-c

### Component causing the issue

unknown. maybe esp32_ble_tracker, maybe wifi, maybe other

### Example YAML snippet

```yaml
See https://github.com/evrardjp/m5stick-cplus-esphome/blob/main/config.yaml .
Removing the comment https://github.com/evrardjp/m5stick-cplus-esphome/blob/8540a472babf256420a08fa184232abc93234d3c/config.yaml#L100 will make it fail.
```

### Anything in the logs that might be useful for us?

```txt
See log here: https://gist.github.com/evrardjp/41102f19964d507c45cd5cec85d2c976
I am curious on why esp-idf is logged here, I never asked for esp-idf, but I am new to all of this :)
```

evrardjp, 2022-10-07:
Oh also, I am naturally inclined to say it might be a memory issue. If you need extra logs, please feel free to contact me.

evrardjp, 2022-10-09:
How can I investigate this further? Any help is welcomed.

arthurkomatsu, 2022-10-13:
@evrardjp https://github.com/esphome/issues/issues/2112

kdorff, 2022-11-08:
Running

* esp32 30-pin devkit
* esphome latest (2022.10.2?)

I'm playing with pretty much the same app on two different displays. What I find is

ESPHome + Wifi + bluetooth_proxy + SSD1306 display : works fine
ESPHome + Wifi + bluetooth_proxy + ili9341display : works fine
ESPHome + Wifi + bluetooth_proxy + ili9341display + **xpt2046 touchscreeen** : boot loop showing the following message repeating over and over

```
[D][esp32_ble_tracker:809]: Found device E6:FC:87:6F:1B:E0 RSSI=-81
[D][esp32_ble_tracker:830]:   Address Type: RANDOM
[D][esp32_ble_tracker:832]:   Name: 'N06Q9'
[I][wifi:255]: WiFi Connecting to 'MY WIFI SSID'...
[E][WiFiGeneric.cpp:137] wifiLowLevelInit(): esp_wifi_init 257
[W][wifi_esp32:057]: Setting WiFi mode failed!
[E][wifi:301]: wifi_sta_connect_ failed!
```

evrardjp, 2022-11-10:
Might be a RAM issue, I don't know, I am lost at the debugging bit ;)
Moved to another esp32 board without issues, so I am leaving this open-ed until I can debug this further, but FYI my priorities have changed.

kdorff, 2022-11-10:
> Might be a RAM issue, I don't know, I am lost at the debugging bit ;) Moved to another esp32 board without issues, so I am leaving this open-ed until I can debug this further, but FYI my priorities have changed.

Should you decide to come back to it, I discovered it was a RAM issue by connecting the device over USB, asking ESPHome for logs over USB, and clicking "Reset Device". This enabled me to the logs right from startup, allowing me to see where it was Wifi initialization was complaining about RAM.

I don't have this RAM issue when I use the much lower resolution 2.42" OLED (128x64) SSD1306/1309 - I can use the bluetooth proxy on those devices that are otherwise doing about the same work (maybe fewer `font`s). But the touchscreen and increased resolution of the ili9341 is nice.