esphome/issues #3665: ESP8266EX WiFi Connection Issues ('Probe Request Unsuccessful')

supermopf, 2022-10-07:
### The problem

Hey there, 
I just bought some 8266EX from [Amazon](https://www.amazon.de/dp/B08BTH77F3) and flashed them with ESPhome. I got some weird connection issue going on: 

The ESP Boots up and tries to connect to my WiFi and after a lot of failed 'Probe Request Unsuccessful' Messages it sometimes connects to my WiFi and after a short while it fails of with a 'Beacon Timeout'.

What I already tried:
- Removed Sensors from Pins
- Restarted Router
- Erase Flash with esptool
- Enabled 802.11b+g+n in Router
- Disabled AP and Captive Portal
- Set power_save_mode: NONE
- Switched from DHCP to Static
- Switched board from esp01_1m to nodemcuv2

My Config:

```
esphome:
  name: door

esp8266:
  board: nodemcuv2
  framework:
    version: recommended

# Enable logging
logger:
  level: VERY_VERBOSE
# Enable Home Assistant API
api:
  encryption:
    key: "<apipassword>"

ota:
  password: "<ota password>"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: NONE
  manual_ip:
    static_ip: 192.168.178.201
    gateway: 192.168.178.1
    subnet: 255.255.255.0

# Enable fallback hotspot (captive portal) in case wifi connection fails
#  ap:
#    ssid: "Door Fallback Hotspot"

#captive_portal:

switch:
  - platform: gpio
    pin: 2
    inverted: true
    name: "Door Opener"
    id: door_relay
    icon: "mdi:door"
    on_turn_on:
    - delay: 3000ms
    - switch.turn_off: door_relay
```

'Probe Request Unsuccessful' Log:
```
[W][wifi:583]: Restarting WiFi adapter...
[V][wifi_esp8266:060]: Disabling STA.
del if0
usl
mode : null
[V][wifi_esp8266:534]: Event: Changed Mode old=STA new=OFF
wifi evt: 8
[D][wifi:384]: Starting scan...
[V][wifi_esp8266:058]: Enabling STA.
mode : sta(4c:75:25:34:a5:7c)
ip:192.168.178.201,mask:255.255.255.0,gw:192.168.178.1
add if0
[V][wifi_esp8266:534]: Event: Changed Mode old=OFF new=STA
wifi evt: 8
scandone
[D][wifi:399]: Found networks:
[I][wifi:442]: - 'FRITZ!Box 7490' [redacted]▂▄▆█
[D][wifi:444]:     Channel: 1
[D][wifi:445]:     RSSI: -62 dB
[D][wifi:447]: - [redacted] [redacted]▂▄▆█
[D][wifi:447]: - [redacted] [redacted]▂▄▆█
[D][wifi:447]: - [redacted] [redacted]▂▄▆█
[D][wifi:447]: - [redacted] [redacted]▂▄▆█
[I][wifi:255]: WiFi Connecting to 'FRITZ!Box 7490'...
[V][wifi:257]: Connection Params:
[V][wifi:258]:   SSID: 'FRITZ!Box 7490'
[V][wifi:261]:   BSSID: E0:28:6D:25:CC:F7
[V][wifi:281]:   Password: [redacted]
[V][wifi:286]:   Channel: 1
[V][wifi:292]:   Manual IP: Static IP=192.168.178.201 Gateway=192.168.178.1 Subnet=255.255.255.0 DNS1=0.0.0.0 DNS2=0.0.0.0
[V][wifi:297]:   Hidden: NO
scandone
[W][wifi_esp8266:479]: Event: Disconnected ssid='FRITZ!Box 7490' reason='Probe Request Unsuccessful'
wifi evt: 1
STA disconnect: 201
[W][wifi:547]: Error while connecting to network.
[W][wifi:583]: Restarting WiFi adapter...
```

When the connection suddenly works:

```
[W][wifi:583]: Restarting WiFi adapter...
[V][wifi_esp8266:060]: Disabling STA.
del if0
usl
mode : null
[V][wifi_esp8266:534]: Event: Changed Mode old=STA new=OFF
wifi evt: 8
[D][wifi:384]: Starting scan...
[V][wifi_esp8266:058]: Enabling STA.
mode : sta(4c:75:25:34:a5:7c)
ip:192.168.178.201,mask:255.255.255.0,gw:192.168.178.1
add if0
[V][wifi_esp8266:534]: Event: Changed Mode old=OFF new=STA
wifi evt: 8
scandone
[D][wifi:399]: Found networks:
[I][wifi:442]: - 'FRITZ!Box 7490' [redacted]▂▄▆█
[D][wifi:444]:     Channel: 1
[D][wifi:445]:     RSSI: -55 dB
[D][wifi:447]: - [redacted] [redacted]▂▄▆█
[D][wifi:447]: - [redacted] [redacted]▂▄▆█
[D][wifi:447]: - [redacted] [redacted]▂▄▆█
[D][wifi:447]: - [redacted] [redacted]▂▄▆█
[D][wifi:447]: - [redacted] [redacted]▂▄▆█
[D][wifi:447]: - [redacted] [redacted]▂▄▆█
[I][wifi:255]: WiFi Connecting to 'FRITZ!Box 7490'...
[V][wifi:257]: Connection Params:
[V][wifi:258]:   SSID: 'FRITZ!Box 7490'
[V][wifi:261]:   BSSID: E0:28:6D:25:CC:F7
[V][wifi:281]:   Password: [redacted]
[V][wifi:286]:   Channel: 1
[V][wifi:292]:   Manual IP: Static IP=192.168.178.201 Gateway=192.168.178.1 Subnet=255.255.255.0 DNS1=0.0.0.0 DNS2=0.0.0.0
[V][wifi:297]:   Hidden: NO
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 5
cnt 

connected with FRITZ!Box 7490, channel 1
ip:192.168.178.201,mask:255.255.255.0,gw:192.168.178.1
[V][wifi_esp8266:468]: Event: Connected ssid='FRITZ!Box 7490' bssid=E0:28:6D:25:CC:F7 channel=1
wifi evt: 0
[V][wifi_esp8266:507]: Event: Got IP static_ip=192.168.178.201 gateway=192.168.178.1 netmask=255.255.255.0
wifi evt: 3
[I][wifi:516]: WiFi Connected!
[C][wifi:360]:   Local MAC: 4C:75:25:34:A5:7C
[C][wifi:361]:   SSID: [redacted]
[C][wifi:362]:   IP Address: 192.168.178.201
[C][wifi:363]:   BSSID: [redacted]
[C][wifi:365]:   Hostname: 'door'
[C][wifi:367]:   Signal strength: -62 dB ▂▄▆█
[V][wifi:369]:   Priority: -39.0
[C][wifi:371]:   Channel: 1
[C][wifi:372]:   Subnet: 255.255.255.0
[C][wifi:373]:   Gateway: 192.168.178.1
[C][wifi:374]:   DNS1: 0.0.0.0
[C][wifi:375]:   DNS2: 0.0.0.0
[C][ota:089]: Over-The-Air Updates:
[C][ota:090]:   Address: 192.168.178.201:8266
[C][ota:093]:   Using Password.
[C][api:025]: Setting up Home Assistant API server...
[I][app:062]: setup() finished successfully!
[I][ota:109]: Boot seems successful, resetting boot loop counter.
[I][app:102]: ESPHome version 2022.9.4 compiled on Oct  7 2022, 13:48:19
[C][wifi:502]: WiFi:
[C][wifi:360]:   Local MAC: 4C:75:25:34:A5:7C
[C][wifi:361]:   SSID: [redacted]
[C][wifi:362]:   IP Address: 192.168.178.201
[C][wifi:363]:   BSSID: [redacted]
[C][wifi:365]:   Hostname: 'door'
[C][wifi:367]:   Signal strength: -62 dB ▂▄▆█
[V][wifi:369]:   Priority: -39.0
[C][wifi:371]:   Channel: 1
[C][wifi:372]:   Subnet: 255.255.255.0
[C][wifi:373]:   Gateway: 192.168.178.1
[C][wifi:374]:   DNS1: 0.0.0.0
[C][wifi:375]:   DNS2: 0.0.0.0
[C][logger:275]: Logger:
[C][logger:276]:   Level: VERY_VERBOSE
[C][logger:277]:   Log Baud Rate: 115200
[C][logger:278]:   Hardware UART: UART0
[C][switch.gpio:050]: GPIO Switch 'Door Opener'
[C][switch.gpio:050]:   Icon: 'mdi:door'
[C][switch.gpio:050]:   Inverted: YES
[C][switch.gpio:051]:   Pin: GPIO2
[C][switch.gpio:073]:   Restore Mode: Restore (Defaults to OFF)
[C][mdns:100]: mDNS:
[C][mdns:101]:   Hostname: door
[V][mdns:102]:   Services:
[V][mdns:104]:   - _esphomelib, _tcp, 6053
[V][mdns:106]:     TXT: version = 2022.9.4
[V][mdns:106]:     TXT: mac = 4c752534a57c
[V][mdns:106]:     TXT: platform = ESP8266
[V][mdns:106]:     TXT: board = nodemcuv2
[V][mdns:106]:     TXT: network = wifi
[C][ota:089]: Over-The-Air Updates:
[C][ota:090]:   Address: 192.168.178.201:8266
[C][ota:093]:   Using Password.
[C][api:138]: API Server:
[C][api:139]:   Address: 192.168.178.201:6053
[C][api:141]:   Using noise encryption: YES
fpm close 3 
pm open,type:0 0
bcn_timout,ap_probe_send_start
[D][api:102]: Accepted 192.168.178.59
[VV][api.socket:693]: 192.168.178.59: Handshake complete!
[V][component:199]: Component api took a long time for an operation (0.35 s).
[V][component:200]: Components should block for at most 20-30ms.
[VV][api.service:410]: on_hello_request: HelloRequest {
  client_info: 'Home Assistant 2022.10.1'
  api_version_major: 1
  api_version_minor: 7
}
[V][api.connection:898]: Hello from client: 'Home Assistant 2022.10.1 (192.168.178.59)' | API Version 1.7
[VV][api.service:013]: send_hello_response: HelloResponse {
  api_version_major: 1
  api_version_minor: 7
  server_info: 'door (esphome v2022.9.4)'
  name: 'door'
}
[VV][api.service:419]: on_connect_request: ConnectRequest {
  password: ''
}
[D][api.connection:917]: Home Assistant 2022.10.1 (192.168.178.59): Connected successfully
```

And then after a while the 'Beacon Timeout' (Probably not an issue, but throws me back to the beginning) :

```
[VV][api.service:446]: on_ping_request: PingRequest {}
[VV][api.service:043]: send_ping_response: PingResponse {}
[VV][api.service:446]: on_ping_request: PingRequest {}
[VV][api.service:043]: send_ping_response: PingResponse {}
[VV][api.service:446]: on_ping_request: PingRequest {}
[VV][api.service:043]: send_ping_response: PingResponse {}
[VV][scheduler:195]: Running interval '' with interval=60000 last_execution=1176012 (now=1236017)
[VV][api.service:446]: on_ping_request: PingRequest {}
[VV][api.service:043]: send_ping_response: PingResponse {}
bcn_timout,ap_probe_send_start
ap_probe_send over, rest wifi status to disassoc
state: 5 -> 0 (1)
rm 0
pm close 7
[W][wifi_esp8266:482]: Event: Disconnected ssid='FRITZ!Box 7490' bssid=[redacted] reason='Beacon Timeout'
wifi evt: 1
STA disconnect: 200
[W][wifi:123]: WiFi Connection lost... Reconnecting...
[W][wifi:583]: Restarting WiFi adapter...
```

Do you have any further ideas, how to fix the connection process?

### Which version of ESPHome has the issue?

2022.9.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.10.1

### What platform are you using?

ESP8266

### Board

ESP8266EX

vobo70, 2022-10-08:
Hi,
It's not first time when I upgrade some sonoff's basic (same board type as yours - esp01_1m) my wifi fails to connect.
For now I found then commenting out:
# captive_portal:
works for me.
Regards,
Maciek

supermopf, 2022-10-13:
Hey there,

I already tried comenting out `# captive_portal` as I already stated above (also visibile in my posted config)

Any other ideas what could be the reason for this issue?

-Victor

telegnom, 2022-10-28:
Yes, disabling the captive portal brings the device back to life.

It's working fine - with enabled captive portal - with arduino framwork version <= 2.7.4

supermopf, 2022-11-04:
Hey, 

can not confirm - I downgraded to version 2.7.4 and still have the same issue.

```
esp8266:
  board: nodemcuv2
  framework:
    version: 2.7.4
```

supermopf, 2022-11-21:
Okay, I resolved this issue. It's a D1 Mini Board and with this config it works:

```
esphome:
  name: door
  platform: ESP8266
  board: d1_mini
```

closed 2022-11-21 by supermopf