esphome/issues #3667: ESP32 bluetooth-active connection keeps crashing

Salvora, 2022-10-07:
### The problem

I have painfully connected a Fibaro FGBHS-213 with homekit controller using ble active proxy of my ESP32 device.

I tried to test the device a couple of times to check the reliability. However, there are connection issues, and toggling switch a couple of times from the HA frontend crashes the esp device.

### Which version of ESPHome has the issue?

2022.9.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.10.1

### What platform are you using?

ESP32

### Board

devkitv1

### Component causing the issue

Bluetooth Proxy

### Additional information

Logs are attached.

nagyrobi, 2022-10-13:
Add to your config, and flash again:

```
debug:

sensor:
- platform: template
  name: Free HEAP
  lambda: return heap_caps_get_free_size(MALLOC_CAP_INTERNAL);
  icon: "mdi:memory"
  entity_category: diagnostic
  state_class: measurement
  unit_of_measurement: "b"
  update_interval: 60s
- platform: uptime
  name: Uptime
- platform: debug
  free:
    name: Free MEM
```
Wait for some time and look at the graphs.

Salvora, 2022-10-13:
This time, couldn't manage to connect to device at all. I tried to force it to connect by reloading the homekit controller integration for this device.

Graphs are below.
[image]
[image]

this is the uptime:
[image]

Full log is attached.
[logs_emre-oda-esp32s-devkitv1_upload.txt](https://github.com/esphome/issues/files/9781207/logs_emre-oda-esp32s-devkitv1_upload.txt)

Jarikk, 2022-10-15:
Have the same issue - with enabled active BLE trackers + active Bluetooth proxy my ESP32 board suddenly reboots. It can happen after 12 hours, 8 hours or even every 1 hour. It looks like low memory is not an issue, because it is always free ~60k. I have 7-8 BLE device in the range of the ESP32 board. 
[image]

nagyrobi, 2022-10-15:
A memory leak has been identified causing this. A potential fix is available in the current beta release, please try that.

Jarikk, 2022-10-15:
@nagyrobi I updated to the latest beta 2022.10.0b2 and updated my ESP32 with new firmware, but nothing changed.
It still reboots. How can I help with bug?
[image]

nagyrobi, 2022-10-15:
Please post your entire config.

Jarikk, 2022-10-16:
I'm using this hardware - [ZigStar LAN Gateway](https://zig-star.com/projects/zigbee-gw-lan/) and trying to activate BLE trackers and Bluetooth proxy on it. Here is the configuration:
```
substitutions:
  device_name: zigstar-livingroom
  friendly_name: Zigstar Livingroom
  ota_pw: !secret ota_password

esphome:
  name: ${device_name}
  comment: ${friendly_name}
  project:
    name: zigstar.livingroom
    version: "1.0"

esp32:
  board: esp32doit-devkit-v1

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_IN
  phy_addr: 1
  power_pin: GPIO16

  manual_ip: 
    static_ip: 192.168.0.22
    gateway: 192.168.0.1
    subnet: 255.255.255.0

logger:
  level: VERBOSE
  baud_rate: 115200
  hardware_uart: UART0

api:
  reboot_timeout: 0s

ota:
  password: ${ota_pw}

dashboard_import:
  package_import_url: github://esphome/bluetooth-proxies/wt32-eth01.yaml@main

esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

bluetooth_proxy:
  active: true

web_server:
  port: 80

time:
  - platform: homeassistant
    id: homeassistant_time

#External component Stream Server
external_components:
  - source: github://oxan/esphome-stream-server

# Extra sensor to keep track of gateway uptime
text_sensor:
  - platform: template
    name: "${friendly_name} Uptime"
    lambda: |-
      uint32_t dur = id(uptime_s).state;
      int dys = 0;
      int hrs = 0;
      int mnts = 0;
      if (dur > 86399) {
        dys = trunc(dur / 86400);
        dur = dur - (dys * 86400);
      }
      if (dur > 3599) {
        hrs = trunc(dur / 3600);
        dur = dur - (hrs * 3600);
      }
      if (dur > 59) {
        mnts = trunc(dur / 60);
        dur = dur - (mnts * 60);
      }
      char buffer[17];
      sprintf(buffer, "%ud %02uh %02um %02us", dys, hrs, mnts, dur);
      return {buffer};
    icon: mdi:clock-start
    update_interval: 60s

debug:

sensor:
  - platform: uptime
    name: "${friendly_name} Uptime Sec"
    id: uptime_s
    update_interval: 60s
  - platform: template
    name: "${friendly_name} Free HEAP"
    lambda: return heap_caps_get_free_size(MALLOC_CAP_INTERNAL);
    icon: "mdi:memory"
    entity_category: diagnostic
    state_class: measurement
    unit_of_measurement: "b"
    update_interval: 60s
  - platform: debug
    free:
      name: "${friendly_name} Free MEM"

# Device functions:
switch:
  - platform: restart
    name: "${friendly_name} Gateway Restart"
    disabled_by_default: true

  - platform: gpio
    pin: 33
    id: zRST_gpio
    inverted: yes
    restore_mode: ALWAYS_OFF
  - platform: template
    name: "${friendly_name} Zigbee RST"
    icon: mdi:toggle-switch
    disabled_by_default: true
    id: zRST
    turn_on_action:
      - switch.turn_on: zRST_gpio
      - delay: 15ms
      - switch.turn_off: zRST_gpio
      
  - platform: gpio
    pin: 32
    name: "Zigbee BSL"
    icon: mdi:toggle-switch
    id: zBSL
    inverted: yes
    restore_mode: ALWAYS_OFF
    internal: true
    
  - platform: template
    name: "${friendly_name} Firmware Update"
    icon: mdi:cellphone-arrow-down
    disabled_by_default: true
    turn_on_action:
      - script.execute: fw_update_mode
    turn_off_action:
      - switch.toggle: zRST

#Update CC2652P/CC2652RB Firmware
script:
  - id: fw_update_mode
    then:
      - switch.turn_on: zBSL
      - delay: 1s
      - switch.turn_on: zRST_gpio
      - delay: 1s
      - switch.turn_off: zRST_gpio
      - logger.log: "Delaying ~10 seconds for TI chip to be ready"
      - delay: 10s
      - switch.turn_off: zBSL
      - logger.log: "Update with cc2538-bsl tool now!"
      - logger.log: "Usage: cc2538-bsl.py -p socket://ip_or_hostname:6638 -evw firmware.hex"

#UART Settings
uart:
  - id: uart_bus
    rx_pin: GPIO5
    tx_pin: GPIO17
    baud_rate: 115200

#Serial Bridge Settings,uncomment #port to change default 6638 TCP port
stream_server:
  uart_id: uart_bus
#  port: 1234
```

I attached the log with startup info. And here is the error, that caused the board to restart:
```
[V][bluetooth_proxy:026]: Proxying packet from  - 29:9D:F2:92:37:B3. RSSI: -78 dB
[V][bluetooth_proxy:026]: Proxying packet from  - 80:47:86:89:D8:B3. RSSI: -98 dB
[V][bluetooth_proxy:026]: Proxying packet from  - 29:9D:F2:92:37:B3. RSSI: -78 dB
[V][bluetooth_proxy:026]: Proxying packet from  - 80:47:86:89:D8:B3. RSSI: -100 dB
[V][bluetooth_proxy:026]: Proxying packet from  - 29:9D:F2:92:37:B3. RSSI: -77 dB
[V][bluetooth_proxy:026]: Proxying packet from MiKetv9 - A4:C1:38:81:39:82. RSSI: -92 dB
[V][api.connection:986]: Cannot send message because of TCP buffer space
abort() was called at PC 0x401e5fa3 on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x4008f938:0x3ffd57f0 0x4008fbb5:0x3ffd5810 0x401e5fa3:0x3ffd5830 0x401e5fea:0x3ffd5850 0x401e5905:0x3ffd5870 0x401e59dc:0x3ffd5890 0x400dc02d:0x3ffd58b0 0x4013aec5:0x3ffd58d0 0x40135fe6:0x3ffd58f0 0x40090c2a:0x3ffd5920

Rebooting...

```
[StartupLog.txt](https://github.com/esphome/issues/files/9793765/StartupLog.txt)

nagyrobi, 2022-10-16:
You are overloading the node.
Remove the web server, that's a huge mamoot.
I'd also temporarily remove the external component too, and see after that. It might be very well the case that you wanna run too many things at once on the same node.

Jarikk, 2022-10-17:
@nagyrobi Ok, thank you for your comments.
I enabled VERBOSE logging level in the hope of getting more debug information at the moment of crash, reboots happened even with default logging level.
I'm interested, does the web server consume a lot of resources if no clients are connected to it? Does it constantly use network? I'm asking about network because it looks like there is enough free memory(60k) on the ESP32 at the moment of crash.
Unfortunately, I can't disable external stream-server component as it proxies a TCP connection from zigbee2Mqtt to the zigbee board via UART.

It's clear that the problem is because several components use ethernet port and TCP buffer gets overflowed. I thought there should be some options to regulate this somehow or increase that buffer. 

But, it is not related to the topic. Probably, the problem with bluetooth-active connections crashing has already been solved.

Salvora, 2022-10-17:
> A memory leak has been identified causing this. A potential fix is available in the current beta release, please try that.

I will wait for the actual release and try that.
Dunno if you also need my config file? But, I don't have a webserver.

lolorc, 2022-10-18:
2022.10.0b2 fixes it for me.
but as I also said elsewhere, I decided to remove webserver, and I keep only one connection to the api.
As @nagyrobi said, I'm also dedicating esp32 for bluetooth and use another one for sensors & switches (ble + i2c + multiple sensors has not been working well)

closed 2022-10-18 by nagyrobi